esphome/issues #4652: Sensors from PZEM integration don't go to unavailable state when no power.

JaroslawPrzybyl, 2023-07-04:
### The problem

PZEM module is powered from AC side. So, when measured AC is lost, it just stop sending new values and in result, esphome still remembers last measured value, instead of "unavailable" or just zero.
In simple application, where esphome is powered from the same AC it doesn't matter, when no power, esphome is also gone, but I found this issue in a project, where PZEM measures only one AC output in the bigger system and it is really annoying to have no way for simple detection that power on PZEM is lost. Sensors should be set to unavailable when no answer from module after time set as "update_interval"

### Which version of ESPHome has the issue?

2023.6.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

NA

### What platform are you using?

ESP32

### Board

NA

### Component causing the issue

PZEM-004Tv3

ssieb, 2023-07-06:
There's no way to set a sensor to "unavailable".  That is only set on the HA side to indicate the esphome device is not connected.  You could use a timer script to change the value if there hasn't been a new value for a certain period of time.  If you need help with that, then come ask on the esphome discord server or one of the forums.

closed 2023-07-06 by ssieb

JaroslawPrzybyl, 2023-07-06:
"unavailable" was an example only, can be zero or anything else. Definitely keeping last measured value is wrong way and it should be fixed by integration, not by additional scripts.

ssieb, 2023-07-06:
Keeping the last value is correct and consistent with all sensors.  Who decides what the stale value should be and who decides how long too long is?  You do, so if that matters to you, then you can add a little automation to handle it.  You could also request a new filter to handle those details, but that would go in the feature requests section.