esphome/issues #4656: MQTT and Traefik issue

Adesfire, 2023-07-04:
### The problem

Hi everyone,
trying to setup my ESPHome to connect directly through Internet to a remote mosquitto service hosted behind a TCP proxy made with Traefik.

When using Android MyMQTT client over the air, I can successfully connect to mydomain.com:443 using SSL security and see the connection being handle on the Mosquitto logs.

However, on my ESPHome logs I can only see this line every x seconds:  
`[I][mqtt:184]: Connecting to MQTT...`

Probably my configuration is not OK, but options here seems very limited:
```
mqtt:
  broker: !secret public_broker
  username: !mqtt_username
  password: !secret mqtt_password
  port: 443
```

I will try the SSL Fingerprints part of the doc, but if I properly understood, that will require me to update this information every time Letencrypt will change my Traefik certificate... it should be another way right?

Any help would be appreciated.

### Which version of ESPHome has the issue?

2023.5.5

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

MQTT

### Anything in the logs that might be useful for us?

```txt
[D][mqtt:121]: Resolving MQTT broker IP address...
[D][mqtt:154]: Resolved broker IP address to xxx.xxx.xxx.xxx
[I][mqtt:184]: Connecting to MQTT...
```

Adesfire, 2023-07-04:
Well, the SSL fingerprint part is not working due to an issue with the SSL lib: https://github.com/esphome/issues/issues/2660

closed 2023-12-04 by github-actions[bot]

mtudan, 2024-03-08:
@Adesfire could you explain how to:
> ...allow to connect without checking the certificate...

Thanks!

Adesfire, 2024-03-10:
> @Adesfire could you explain how to:
> 
> > ...allow to connect without checking the certificate...
> 
> Thanks!

It was a suggestion to the dev team, unfortunately so far you cannot do that. I am still waiting for a solution.