esphome/issues #4661: Using Remote Transmitter in home assistant will cause Connection reset by peer

lijiazhiren, 2023-07-05:
### The problem

in home assistant service
This code works fine with

[9005, -4584, 460, -1758, 491, -583, 536, -610, 512, -609, 509, -632, 492, -624, 495, -627, 493, -626, 492, -646, 475, -1757, 491, -1756, 496, -1737, 511, -1794, 457, -1757, 493, -1755, 494, -1757, 496, -1794, 454, -1775, 476, -644, 475, -683, 436, -669, 454, -643, 476, -644, 477, -606, 514, -624, 494, -646, 474, -1794, 456, -1756, 493, -1790, 462, -1794, 457, -1773, 476, -1757, 523]

This code is not working properly

[9134, -4368,  742, -1566,  770, -436,  768, -436,  770, -1538,  770, -1538,  768, -436,  770, -436,  770, -436,  770, -1538,  770, -436,  770, -436,  768, -1538,  770, -436,  770, -436,  744, -462,  768, -436,  770, -436,  770, -436,  768, -436,  770, -436,  768, -436,  768, -1540,  770, -436,  770, -436,  768, -436,  768, -436,  768, -438,  768, -436,  770, -1536,  770, -436,  770, -1538,  770, -436,  768, -436,  770, -1538,  770, -436,  768, -19846,  792, -438,  744, -462,  744, -460,  744, -462,  770, -436,  768, -438,  744, -462,  742, -462,  746, -460,  768, -436,  746, -460,  768, -436,  768, -438,  768, -1538,  770, -436,  768, -436,  744, -462,  770, -436,  768, -436,  770, -436,  768, -436,  770, -436,  768, -436,  768, -436,  768, -436,  770, -436,  768, -436,  770, -436,  770, -436,  770, -1538,  770, -1536,  770, -1536,  770, -39854,  9136, -4368,  770, -1538,  770, -436,  770, -436,  770, -1538,  770, -1536,  770, -436,  770, -436,  768, -436,  770, -1538,  770, -436,  770, -436,  768, -1538,  770, -436,  770, -436,  770, -436,  770, -436,  770, -436,  768, -436,  770, -436,  768, -436,  770, -436,  770, -1538,  770, -436,  770, -436,  770, -436,  770, -436,  768, -436,  768, -436,  768, -1538,  770, -1536,  768, -1538,  770, -436,  770, -436,  770, -1538,  770, -434,  770, -19844,  794, -436,  768, -436,  770, -436,  768, -436,  768, -436,  770, -436,  768, -436,  768, -436,  770, -436,  768, -438,  768, -436,  768, -436,  770, -436,  770, -436,  770, -436,  768, -436,  770, -436,  770, -436,  768, -436,  770, -436,  768, -1538,  770, -434,  770, -436,  770, -436,  768, -436,  770, -436,  768, -436,  770, -436,  768, -1538,  770, -436,  770, -1536,  770, -1536,  770]

Here is a tip from esphome

WARNING rf-text.local: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for rf-text.local
WARNING Disconnected from API
INFO Successfully connected to rf-text.local

The code that cannot be used normally is my air conditioner code. There are many. I use a component called (smartir)

But when I delete the switch component and button component, the air conditioner code can work normally in the home assistant

So this is caused by the insufficient performance of esp8266 or the problem of esphome?

### Which version of ESPHome has the issue?

2023.6.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2023.6.3 Supervisor 2023.06.4 Operating System 10.3

### What platform are you using?

ESP8266

### Board

nodemcuv2

### Component causing the issue

api:   services:

### Example YAML snippet

```yaml
api:
  services:
    - service: send_ir_command
      variables:
        command: int[]
      then:
        - remote_transmitter.transmit_raw:
            code: !lambda 'return command;'
            carrier_frequency: 38kHz
            transmitter_id: hongwai_fashe

remote_transmitter:
  - id: hongwai_fashe
    pin: GPIO14
    carrier_duty_percent: 50%

  - id: Rf433_transmitter
    pin: GPIO5
    carrier_duty_percent: 100%

  - id: Rf315_transmitter
    pin: GPIO13
    carrier_duty_percent: 100%

remote_receiver:
  - id: IR_receiver
    pin:
      number: GPIO12
      inverted: True
    dump:
    - raw
    - nec

  - id: RF433_receiver
    pin:
      number: GPIO3
      inverted: True
    dump:
      - rc_switch
    tolerance: 60%
    filter: 50us
    idle: 3ms

  - id: RF315_receiver
    pin:
      number: GPIO4
    dump:
      - rc_switch
    tolerance: 60%
    filter: 50us
    idle: 3ms

switch:

button:
```

### Anything in the logs that might be useful for us?

```txt
WARNING rf-text.local: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for rf-text.local
WARNING Disconnected from API
INFO Successfully connected to rf-text.local
```

ssieb, 2023-07-05:
Watch the serial logs to see what's happening.  Most likely the array is too long.

lijiazhiren, 2023-07-06:
> Watch the serial logs to see what's happening. Most likely the array is too long. I expect that it's crashing and rebooting.

But when I delete the switch component and button component, the air conditioner code can work normally in the home assistant

When there are only remote_transmitter and remote_receiver, the air conditioner code can be used normally in the home assistant service

lijiazhiren, 2023-07-06:
This is the serial logs

15:39:11.526 -> 
15:39:11.526 -> User exception (panic/abort/assert)
15:39:11.560 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
15:39:11.560 -> 
15:39:11.560 -> Unhandled C++ exception: OOM
15:39:11.560 -> 
15:39:11.560 -> >>>stack>>>
15:39:11.560 -> 
15:39:11.560 -> ctx: cont
15:39:11.560 -> 3ffffcd0:  3fff956c 3ffffd70 00000306 40204a99  
15:39:11.560 -> 3ffffce0:  000000fe 00000000 00000000 00000000  
15:39:11.560 -> 3ffffcf0:  00000000 00000000 00000000 00000000  
15:39:11.560 -> 3ffffd00:  3fffefd0 000000fe 3fff0c40 00000001  
15:39:11.560 -> 3ffffd10:  00000000 00000000 00000020 3fff8030  
15:39:11.592 -> 3ffffd20:  3ffffda0 40211f64 0000045c 4022376e  
15:39:11.592 -> 3ffffd30:  3ffffda0 40211f64 00000020 4022378d  
15:39:11.592 -> 3ffffd40:  3fff837c 00000000 0000045c 40222dcc  
15:39:11.592 -> 3ffffd50:  3fff9564 0000045c 3ffffd70 40211f64  
15:39:11.592 -> 3ffffd60:  40215274 3ffffda0 3fff3ffc 40212299  
15:39:11.592 -> 3ffffd70:  00000000 00000000 00000000 40222db7  
15:39:11.592 -> 3ffffd80:  3ffffd88 0000045c 3ffffda0 40211f64  
15:39:11.624 -> 3ffffd90:  40211f98 40212278 3fff3ffc 40211fd8  
15:39:11.624 -> 3ffffda0:  3fffbaac 3fffbf08 3fffbf08 3fffb64c  
15:39:11.624 -> 3ffffdb0:  3fffbaa8 3fffbaa8 3fffb1ec 3fffb648  
15:39:11.624 -> 3ffffdc0:  3fffb648 3fffad8c 3fffb1e8 3fffb1e8  
15:39:11.624 -> 3ffffdd0:  40211f98 3fff189c 3fff7fcc 40212235  
15:39:11.624 -> 3ffffde0:  3fffa92c 3fffad88 3fffad88 00000000  
15:39:11.624 -> 3ffffdf0:  00000000 00000352 00000008 402059e1  
15:39:11.656 -> 3ffffe00:  3fff83e0 00000020 0000034a 00000000  
15:39:11.656 -> 3ffffe20:  0000034a 00000352 3fff879c 3fff70f0  
15:39:11.656 -> 3ffffe30:  3fff14e8 3ffffe50 3fff14e8 40225c18  
15:39:11.656 -> 3ffffe40:  3fff70f0 40228050 3fff8494 4020520e  
15:39:11.656 -> 3ffffe50:  40228050 b6a80451 3fff837c 3fff83e0  
15:39:11.656 -> 3ffffe60:  3fff83e0 00000000 00000352 00000001  
15:39:11.688 -> 3ffffe70:  00000000 00000000 00000000 3fff70f0  
15:39:11.688 -> 3ffffe80:  00000000 0003281b a0000000 03a65ff1  
15:39:11.688 -> 3ffffe90:  00000000 00000000 00000000 3fff70f0  
15:39:11.688 -> 3ffffea0:  3fff70f0 00000000 3fff8494 402016c8  
15:39:11.688 -> 3ffffeb0:  00000000 3fffc6fc 00000000 3ffefdf0  
15:39:11.688 -> 3ffffec0:  00000002 000327fb 3fff1834 00000030  
15:39:11.688 -> 3ffffed0:  401004e1 00000030 00000010 ffffffff  
15:39:11.720 -> 3ffffee0:  00000000 0003281a fdb22d0e 3ffefdf0  
15:39:11.720 -> 3ffffef0:  3fffff5c 3fff9564 3fff98b6 3fff98b6  
15:39:11.720 -> 3fffff00:  3fff002a 00000000 00000352 3fff1834  
15:39:11.720 -> 3fffff10:  3fff70f0 3fff70f0 3fff1834 40205810  
15:39:11.720 -> 3fffff20:  4025aaae 3fff15c4 0000001c 00000000  
15:39:11.720 -> 3fffff30:  4025bc9b 00000001 3fff1a14 402269da  
15:39:11.720 -> 3fffff40:  00000000 0003281b 5a1cac08 03a65fa0  
15:39:11.752 -> 3fffff50:  00000000 0003281b 00000000 0000001c  
15:39:11.752 -> 3fffff60:  3fff833c 3fff1834 3fffff80 3fff1834  
15:39:11.752 -> 3fffff70:  3fff833c 00000002 3ffefdf0 4020f222  
15:39:11.752 -> 3fffff80:  0003281b 3fff1834 3fff7000 3fff03cc  
15:39:11.752 -> 3fffff90:  3fff8340 3fff235c 3ffefe24 3fff03cc  
15:39:11.752 -> 3fffffa0:  3fffdad0 00000000 3fff03b8 3fff03cc  
15:39:11.752 -> 3fffffb0:  3fffdad0 00000000 3fff03b8 40223298  
15:39:11.785 -> 3fffffc0:  feefeffe feefeffe 3ffe85d8 401004e1  
15:39:11.785 -> <<<stack<<<
15:39:11.785 -> 
15:39:11.785 -> last failed alloc call: 40211F64(1116)
15:39:11.785 -> 
15:39:11.785 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
15:39:11.785 -> 
15:39:11.785 -> last failed alloc caller: 0x40211f64
15:39:11.785 -> 
15:39:11.785 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,7)
15:39:11.818 -> 
15:39:11.818 -> load 0x4010f000, len 3460, room 16 
15:39:11.818 -> tail 4
15:39:11.818 -> chksum 0xcc
15:39:11.818 -> load 0x3fff20b8, len 40, room 4 
15:39:11.860 -> tail 4
15:39:11.860 -> chksum 0xc9
15:39:11.860 -> csum 0xc9
15:39:11.860 -> v0006c630
15:39:11.860 -> ~ld
15:39:11.894 -> [0;32m[I][logger:262]: Log initialized[0m
15:39:11.894 -> [0;35m[C][ota:469]: There have been 9 suspected unsuccessful boot attempts.[0m
15:39:11.927 -> [0;32m[I][app:029]: Running through setup()...[0m
15:39:11.927 -> [0;35m[C][remote_receiver.esp8266:034]: Setting up Remote Receiver...[0m
15:39:11.927 -> [0;35m[C][remote_receiver.esp8266:034]: Setting up Remote Receiver...[0m
15:39:11.927 -> [0;35m[C][remote_receiver.esp8266:034]: Setting up Remote Receiver...[0m
15:39:11.927 -> [0;35m[C][wifi:038]: Setting up WiFi...[0m
15:39:11.927 -> [0;35m[C][wifi:048]: Starting WiFi...[0m
15:39:11.968 -> [0;35m[C][wifi:049]:   Local MAC: A4:CF:12:ED:7F:12[0m
15:39:11.968 -> [0;36m[D][wifi:425]: Starting scan...[0m
15:39:17.887 -> [0;36m[D][wifi:440]: Found networks:[0m
15:39:17.887 -> [0;32m[I][wifi:483]: - 'NEWIFI32' [5m(20:76:93:4F:58:D2) [6m[0;32m▂▄▆█[0m[0m
15:39:17.887 -> [0;36m[D][wifi:485]:     Channel: 8[0m
15:39:17.887 -> [0;36m[D][wifi:486]:     RSSI: -44 dB[0m
15:39:17.887 -> [0;36m[D][wifi:488]: - [5m'NEWIFI'[6m [5m(20:76:93:28:6D:44) [6m[0;33m▂▄[0;37m▆█[0m[0m
15:39:17.931 -> [0;32m[I][wifi:274]: WiFi Connecting to 'NEWIFI32'...[0m
15:39:24.663 -> [0;36m[D][api:102]: Accepted 192.168.99.205[0m
15:39:24.663 -> [0;36m[D][api.connection:1030]: Home Assistant 2023.6.3 (192.168.99.205): Connected successfully[0m

rdlaner, 2023-10-21:
@lijiazhiren I'm running into the same issue. Did you have any luck solving this?

ssieb, 2023-10-21:
I see that I missed replying to this.  `Unhandled C++ exception: OOM`  It ran out of memory.
@rdlaner can you get serial logs from yours?

rdlaner, 2023-10-21:
Yeah, it looks like I'm hitting the same thing, likely I have too much going on w/ this esp8266:

```
User exception (panic/abort/assert)
                                   --------------- CUT HERE FOR EXCEPTION DECODER ---------------

Unhandled C++ exception: OOM

>>>stack>>>

ctx: cont
sp: 3ffffca0 end: 3fffffd0 offset: 0000
3ffffca0:  3fff67e0 3ffffd40 00000604 402129ed
3ffffcb0:  000000fe 00000000 00000000 00000000
3ffffcc0:  00000000 00000000 00000000 00000000
3ffffcd0:  3fffefd0 000000fe 3fff0f60 00000001
3ffffce0:  00000000 00000000 00000020 3fff5758
3ffffcf0:  3ffffd70 40229023 0000091c 402379de
3ffffd00:  3ffffd70 40229023 00000020 402379fd
3ffffd10:  3fff54cc 00000000 0000091c 4023703c
3ffffd20:  3fff67d8 0000091c 3ffffd40 40229023
3ffffd30:  402294a0 3ffffd70 3fff24cc 40229349
3ffffd40:  00000000 00000000 00000000 40237027
3ffffd50:  3ffffd58 0000091c 3ffffd70 40229034
3ffffd60:  40229044 40229328 3fff24cc 40229084
3ffffd70:  3fffb47c 3fffbd98 3fffbd98 3fffab5c
3ffffd80:  3fffb478 3fffb478 3fffa23c 3fffab58
3ffffd90:  3fffab58 3fff991c 3fffa238 3fffa238
3ffffda0:  40229044 3fff1cc4 3fff56f4 402292e1
3ffffdb0:  3fff8ffc 3fff9918 3fff9918 00000000
3ffffdc0:  00000000 000006e0 00000008 40213d9d
3ffffdd0:  3fff5530 00000000 000006d8 00000000
3ffffde0:  3fff67e0 000006d8 00000002 40202da9
3ffffdf0:  000006d8 00000020 3fff56f4 3fff46a8
3ffffe00:  3fff1808 3ffffe20 3fff1808 40239e84
3ffffe10:  3fff46a8 4023c7f8 3fff5374 40213156
3ffffe20:  4023c7f8 acddc8e2 3fff54cc 3fff5530
3ffffe30:  3fff5530 00000000 00000000 00000000
3ffffe40:  00000000 00000000 00000000 3fff53f4
3ffffe50:  00000000 00000000 00000000 4020c1a1
3ffffe60:  3ffffef4 00000000 00000000 40210136
3ffffe70:  3fff67d4 000006e4 000006f4 00000000
3ffffe80:  00000000 0002480b ec083126 02a34dc8
3ffffe90:  3fff46a8 3fff46a8 3fff53f4 3fff46a8
3ffffea0:  3fff46a8 00000000 3fff5374 4020d628
3ffffeb0:  00000002 000247e7 3fff1d64 00000030
3ffffec0:  00000000 00000000 00000001 00000001
3ffffed0:  00000000 00000000 00000000 fffffffe
3ffffee0:  00000000 00024807 48b43958 3fff0258
3ffffef0:  3fffff5c 3fff67d4 3fff6ec8 3fff6ec8
3fffff00:  3fff002a 00000004 000006e0 3fff1c24
3fffff10:  3fff46a8 3fff46a8 3fff1c24 4021395c
3fffff20:  4026fe72 3fff190c 0000001c 00000000
3fffff30:  00000000 00024807 947ae147 02a348c3
3fffff40:  00000000 00024807 95c28f5c 02a348c5
3fffff50:  3fff4d08 00000002 00000000 0000001c
3fffff60:  3fff4d0c 3fff1c24 3fffff80 3fff1c24
3fffff70:  3fff4d0c 00000002 3fff0258 40223d32
3fffff80:  00024807 3fff1c24 3fff3a00 3fff06f0
3fffff90:  3fff4d14 3fff2620 3fff028c 3fff06f0
3fffffa0:  3fffdad0 00000000 3fff06dc 3fff06f0
3fffffb0:  3fffdad0 00000000 3fff06dc 4023750c
3fffffc0:  feefeffe feefeffe 3ffe8614 40100501
<<<stack<<<

last failed alloc call: 40229023(2332)

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

ssieb, 2023-10-21:
Ok, so this isn't an esphome bug.
That's a long stack.  I suggest you watch the logs directly from esphome to see what's going on because that way it can decode the stack.  If you need help, come see us on discord.

closed 2023-10-21 by ssieb