esphome/issues #4663: RP2040 stops working when hardware SPI in use

piotrva, 2023-07-05:
### The problem

After the latest update to 2023.6.4 (incorporating #4478) with usage of:
* display/waveshare_epaper/2.90inv2
* spi-bus

The module with configuration as included, after the update did not connect again.
The same result was when forcing Pico into UF2 bootloader and uploading the same firmware.

Finally forcing the SPI to software by using `force_sw: true` provided a workaround and device started working normally.

### Which version of ESPHome has the issue?

2023.6.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.6.3

### What platform are you using?

RP2040

### Board

Raspberry Pi Pico

### Component causing the issue

spi, display/waveshare_epaper

### Example YAML snippet

```yaml
esphome:
  name: display-paper
  friendly_name: display-paper
  on_boot:
    priority: 600     #  Defaults to 600 DOESN'T WAIT FOR WIFI    
           # -100 waits till ALL IS INITIALIZED but NO EXECUTION HERE if no WIFI
    then:
      - display.page.show: boot
      - wait_until:
          api.connected:   #  REMEMBER THAT  LOGGER  COUNTS AS AN API CLIENT 
      - delay: 5s
      - display.page.show: datas
      - component.update: disp

rp2040:
  board: rpipicow
  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "XXX"
  reboot_timeout: 0s

font:
  - file: 'fonts/COMIC.ttf'
    id: font1
    size: 25

sensor:
  - platform: homeassistant
    id: t_out
    entity_id: sensor.temperatura_pole
  - platform: homeassistant
    id: t_in
    entity_id: sensor.temperatura_sypialnia
  - platform: homeassistant
    id: humidity
    entity_id: sensor.humidity
  - platform: homeassistant
    id: press
    entity_id: sensor.pressure
  - platform: uptime
    name: EPaper Uptime Sensor

text_sensor:
  - platform: template
    id: reset_reason
    internal: False
    name: "Reset Reason Epaper"
    update_interval: 15min
    # https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2040/hardware_regs/include/hardware/regs/vreg_and_chip_reset.h
    lambda: |-
      io_rw_32 *reset_reason = (io_rw_32 *) (VREG_AND_CHIP_RESET_BASE + 0x8);
      if ((*reset_reason & 0x00000100)) {
        return {"POR_RESET"};
      }
      if ((*reset_reason & 0x00010000)) {
        return {"RUN_RESET"};
      }
      return {"UNKNOWN_RESET"};

spi:
  clk_pin: GPIO10
  mosi_pin: GPIO11
  miso_pin: GPIO14

graph:
  # Show bare-minimum auto-ranged graph
  - id: t_in_graph
    # sensor: t_in
    duration: 6h
    width: 148
    height: 70
    x_grid: 1h
    y_grid: 1.0
    traces:
      - sensor: t_in
        id: t_in_id
  - id: t_out_graph
    duration: 6h
    width: 148
    height: 70
    x_grid: 1h
    y_grid: 1.0
    traces:
      - sensor: t_out
        id: t_out_id

display:
  - platform: waveshare_epaper
    id: disp
    cs_pin: GPIO9
    dc_pin: GPIO8
    busy_pin: GPIO13
    reset_pin: GPIO12
    model: 2.90inv2
    full_update_every: 1
    update_interval: 5min
    rotation: 90
    pages:
      - id: datas
        lambda: |-
          int sc_w = it.get_width();
          int sc_h = it.get_height();

          id(t_in_graph).set_max_value(id(t_in_id).get_tracedata()->get_recent_max() + 0.75);
          id(t_in_graph).set_min_value(id(t_in_id).get_tracedata()->get_recent_min() - 0.75);

          id(t_out_graph).set_max_value(id(t_out_id).get_tracedata()->get_recent_max() + 0.75);
          id(t_out_graph).set_min_value(id(t_out_id).get_tracedata()->get_recent_min() - 0.75);

          it.printf(sc_w/4*1,  0, id(font1), TextAlign::TOP_CENTER, "In %.1f°C",   id(t_in).state);
          it.printf(sc_w/4*3,  0, id(font1), TextAlign::TOP_CENTER, "Out %.1f°C",   id(t_out).state);
          it.printf(sc_w/4*1, 95, id(font1), TextAlign::TOP_CENTER, "%.1fhPa",   id(press).state);
          it.printf(sc_w/4*3, 95, id(font1), TextAlign::TOP_CENTER, "%.1f%%",   id(humidity).state);
          it.graph(0, 32, t_in_graph);
          it.graph(148, 32, t_out_graph);
      - id: boot
        lambda: |-
          int sc_w = it.get_width();
          int sc_h = it.get_height();
          it.print(sc_w/2, sc_h/2, id(font1), TextAlign::CENTER, "Booting...");
ota:
  password: "XXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  reboot_timeout: 0s

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Display-Paper Fallback Hotspot"
    password: !secret wifi_ap_passwd
```

ssieb, 2023-07-06:
I'm surprised that works at all.  Your `on_boot` priority is higher than the display, so it won't be initialized yet.

piotrva, 2023-11-14:
Well, this was working without any issues with software SPI. So only point of change is hardware SPI.

ferrix, 2024-12-18:
I can reproduce this with a Touchsy RP2040-based device. Configuring an SPI interface isn't enough to crash the device, but enabling the ILI9341 on board causes a boot loop and safe mode without exception.