esphome/issues #4664: [question] kalman combinator component

aura6174, 2023-07-05:
### The problem

what was the range for error: x [0.01,1.0] or [1,100]
```
sensor:
  - platform: kalman_combinator
    name: "Temperature"
    unit_of_measurement: °C
    process_std_dev: 0.001
    sources:
      - source: temperature_sensor_1
        error: x
```

### Which version of ESPHome has the issue?

2023.6.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

aura6174, 2023-07-05:
How i can use this, example for kalman combinator *std_dev (Optional, [Sensor](https://esphome.io/components/sensor/index.html#config-sensor)): A sensor that publishes the current standard deviation of the state with each update.*

ssieb, 2023-07-05:
Github issues are not for support requests.  Ask on the esphome discord server or one of the forums.

closed 2023-07-05 by ssieb