esphome/issues #4667: ble_rrsi sensor causing errors in HA

adorobis, 2023-07-06:
### The problem

I've set up `ble_rssi` platform sensor to monitor my Xiaomi Miscale which recently stopped working properly (mostly esphome is not able to read the measurements anymore). Now I'm receiving lots of errors in HA log.

### Which version of ESPHome has the issue?

2023.6.4

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.7.0

### What platform are you using?

ESP32

### Board

ESP32 DEVKITV1

### Component causing the issue

ble_rssi

### Example YAML snippet

```yaml
sensor:
  - platform: ble_rssi
    mac_address: 0C:95:41:7E:E8:17
    name: "Xiaomi Scale RSSI"
esp32_ble_tracker:
  scan_parameters:
    interval: 60s
    window: 20s
```

### Anything in the logs that might be useful for us?

```txt
2023-07-06 09:10:50.385 ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised when updating state of sensor.xiaomi_scale_rssi, topic: 'bttracker2/sensor/xiaomi_scale_rssi/state' with payload: b'nan'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 590, in state
    numerical_value = int(value)
                      ^^^^^^^^^^
ValueError: invalid literal for int() with base 10: 'nan'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 270, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 742, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 809, in _async_write_ha_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 748, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 594, in state
    raise ValueError(
ValueError: Sensor sensor.xiaomi_scale_rssi has device class 'signal_strength', state class 'measurement' unit 'dBm' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: 'nan' (<class 'str'>)
```

### Additional information

I've changed the `scan_interval` parameters to 60s and 20s accordingly as with default parameters the esphome has not been receiving any data from the scale.

luciotorelli, 2023-07-06:
Can reproduce on OS installation too. Using the IBeacon provided by the phone integration to track phones and smartwatches, it was working prior to Home Assistant 2023.7 and ESPhome 2023.6.4.

adorobis, 2023-07-06:
Yes, this i s esphome publishing on mqtt. I've also explained the reason why I'm using mqtt integration for this device rather than the native integration. Anyway both are supported and both should work properly. Esphome should publish `none` or `unavailable` (or even simply nothing e.g. clear the topic) rather than `nan` - this generates the error on HA side.

luciotorelli, 2023-07-06:
> The exception you posted above seems to indicate the sensor is coming from mqtt and not esphome
> 
> Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 270, in process_write_state_requests entity.async_write_ha_state()

If that helps I am not using mqtt but rather the sensor and binary_sensor from within ESP and have a similar issue.

```
binary_sensor:
  # Presence based on IBeacon address
  - platform: ble_presence
    ibeacon_uuid: 'UUIEXAMPLE001'
    name: "Watch Tracker"
  - platform: ble_presence
    ibeacon_uuid: 'UUIEXAMPLE002'
    name: "Phone Tracker"

sensor:
  - platform: ble_rssi
    ibeacon_uuid: 'UUIEXAMPLE001'
    name: "Watch RSSI"
  - platform: ble_rssi
    ibeacon_uuid: 'UUIEXAMPLE002'
    name: "Phone RSSI"
```

ESP Logs:

```
[15:07:36][I][bluetooth_proxy:272]: [0] [MACEXAMPLE] Connecting v3 with cache
[15:07:36][D][esp32_ble_tracker:213]: Pausing scan to make connection...
[15:07:36][I][esp32_ble_client:064]: [0] [MACEXAMPLE] 0x01 Attempting BLE connection
[15:07:37][I][esp32_ble_client:144]: [0] [MACEXAMPLE] Connected
[15:07:37][D][esp32_ble_tracker:245]: Starting scan...
[15:07:37][D][binary_sensor:036]: 'Watch Tracker': Sending state OFF
[15:07:37][D][binary_sensor:036]: 'Phone Tracker': Sending state OFF
[15:07:37][D][sensor:094]: 'Watch RSSI': Sending state nan dBm with 0 decimals of accuracy
[15:07:37][D][sensor:094]: 'Phone RSSI': Sending state nan dBm with 0 decimals of accuracy
[15:07:37][I][esp32_ble_client:265]: [0] [MACEXAMPLE] auth complete. remote BD_ADDR: MACEXAMPLE
```

I have 3 ESP boards and since update all of them having the same issue. Not sure if required for this but they are all allowed to make Home Assistant service calls.

bdraco, 2023-07-06:
@luciotorelli If you have a different trace back from HA, that might be helpful here

adorobis, 2023-07-06:
> Maybe #4641

Bingo! That was it for my problem with the Xiaomi Scale. After removing the bluetooth proxy lines it works again.
Now we just need to fix it and fix the ble_rssi platform :)

bdraco, 2023-07-06:
Probably going to have to wait for Jesse to take a look at this one as it's outside of my area of expertise

adorobis, 2023-07-10:
Sorry, but this issue is about the error related to `nan` value being sent for `ble_rssi` sensor to HA. And it is not solved by above fix. Error is still there:
```
2023-07-10 13:42:47.366 ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised when updating state of sensor.xiaomi_scale_rssi, topic: 'bttracker2/sensor/xiaomi_scale_rssi/state' with payload: b'nan'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 590, in state
    numerical_value = int(value)
                      ^^^^^^^^^^
ValueError: invalid literal for int() with base 10: 'nan'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 270, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 742, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 810, in _async_write_ha_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 748, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 594, in state
    raise ValueError(
ValueError: Sensor sensor.xiaomi_scale_rssi has device class 'signal_strength', state class 'measurement' unit 'dBm' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: 'nan' (<class 'str'>)
```

bdraco, 2023-07-11:
I'm surprised that removing the proxy lines in https://github.com/esphome/issues/issues/4667#issuecomment-1623864422 fixed the issue for you in the first place but the new code doesn't fix the issue with the scale.

Can you provide new logs with VERBOSE logger after flashing ESPHome 2023.6.5?

closed 2023-11-30 by github-actions[bot]