esphome/issues #4674: RGBWW light on ESP32 not restoring properly after reboot

PaoloTK, 2023-07-07:
### The problem

When I restore the light I observe the following:
- The RGB color and brightness are restored correctly
- The cold and warm white channels aren't restored correctly: the cold white channel is set to 100%, the warm white channel is set to 0%.

No matter what I have the cold and warm white channels set to, if they're at 0%, 100% or 32%, they always get restored CW at 100% and WW at 0% after reboot.

### Which version of ESPHome has the issue?

2023.6.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.7

### What platform are you using?

ESP32

### Board

Quinled-Esp32

### Component causing the issue

RGBWW Light

### Example YAML snippet

```yaml
light:
  - platform: rgbww
    name: ${friendly_name}
    restore_mode: RESTORE_DEFAULT_OFF 
    id: rgbcct
    icon: "mdi:led-strip"
    red: pwm_r
    green: pwm_g
    blue: pwm_b
    cold_white: pwm_cw
    warm_white: pwm_ww
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2200 K
```

anonyme22, 2023-07-22:
Samething with ESPHome 2023.7.0

anonyme22, 2023-07-23:
This issues happen on PWM and SM2135 light.
`light:
  - platform: rgbww
    restore_mode: RESTORE_DEFAULT_OFF
    # color_mode: RGB_COLD_WARM_WHITE
    default_transition_length: 0s
    name: $name
    red: output_red
    green: output_green
    blue: output_blue
    cold_white: output_cwhite
    warm_white: output_wwhite
    cold_white_color_temperature: 6536 K
    warm_white_color_temperature: 2000 K

# output entry
# GPIO4 - Red
# GPIO12 - Green
# GPIO14 - Blue
# GPIO15 - Cold White
# GPIO13 - Warm White
output:
  - platform: esp8266_pwm
    id: output_red
    pin: GPIO4
    max_power: 80%
    
  - platform: esp8266_pwm
    id: output_green
    pin: GPIO12
    max_power: 80%
    
  - platform: esp8266_pwm
    id: output_blue
    pin: GPIO14
    max_power: 80%
    
  - platform: esp8266_pwm
    id: output_cwhite
    pin: GPIO5
    max_power: 80%
    
  - platform: esp8266_pwm
    id: output_wwhite
    pin: GPIO13
    max_power: 80%`

And
`light:
  - platform: rgbww
    restore_mode: RESTORE_DEFAULT_OFF
    # color_mode: RGB_COLD_WARM_WHITE
    default_transition_length: 0s
    name: $name
    red: output_red 
    green: output_green 
    blue: output_blue 
    cold_white: output_white 
    warm_white: output_warmwhite 
    cold_white_color_temperature: 6536 K
    warm_white_color_temperature: 2000 K

sm2135:
  data_pin: GPIO14
  clock_pin: GPIO4

# Individual outputs
output:
  - platform: sm2135
    id: output_red
    channel: 2
    max_power: 0.8
  - platform: sm2135
    id: output_green
    channel: 0
    max_power: 0.8
  - platform: sm2135
    id: output_blue
    channel: 1
    max_power: 0.8
  - platform: sm2135
    id: output_white
    channel: 3
    max_power: 0.8
  - platform: sm2135
    id: output_warmwhite
    channel: 4
    max_power: 0.8
`

PaoloTK, 2023-08-12:
Any news on this?

PaoloTK, 2023-12-14:
Won't be able to test at the moment, not sure if but has been fixed. Any chance you can confirm @anonyme22 ?

Legot, 2024-03-13:
Still seeing this issue on ESPHome 2024.2.2, using the bp5758d platform.

brotaxt, 2024-08-27:
> ```
> preferences:
>   flash_write_interval: 1min #This is required for the cb2l to restore the light state after rebooting.
> ```

This fixed the problems for me too. Thank you very much! 

I am a little surprised though, as this value seems to be the default anyway!? 

> Adjusting flash writes
> # Example configuration entry
> preferences:
>   flash_write_interval: 1min
> flash_write_interval (Optional, [Time](https://esphome.io/guides/configuration-types#config-time)): Customize the frequency in which data is flushed to the flash. This setting helps to prevent rapid changes to a component from being quickly written to the flash and wearing it out. Defaults to 1min.

source: https://esphome.io/components/esphome.html