esphome/issues #4676: Since .5 Git doesn't trust any github external components

JohnGalt1717, 2023-07-08:
### The problem

Since the 2023.5.x releases for docker external components aren't trusted, so update all fails, as does all updates unless you exec into the docker container and add the commands manually.

I.e.: git config --global --add safe.directory /config/.esphome/external_components/35bd8a5b

### Which version of ESPHome has the issue?

2023.5.x+

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Example YAML snippet

```yaml
external_components:
  - source: github://syssi/esphome-jk-bms@main
    refresh: 0s
```

### Additional information

Please consider adding the ability to use environment variables to list multiple sources that should be trusted so that we can put it in the docker compose.

Or an environment variable that automatically trusts everything on github.com (or the above with a wildcard function so that we can trust all of them on github.)

Running the commands every time there is a container update is annoying and I forget, and then update all fails and then I have to do them manually from there because it redoes all of the ones already updated when you click update all even if there is no update required.

ssieb, 2023-07-09:
You didn't explain exactly what the problem is.  No one else has mentioned a problem with external components.  There shouldn't be a problem, since there are no specific restrictions.

JohnGalt1717, 2023-07-10:
Any external component using the docker container has the same problem after every update: The git commands to white list the directory has to be run.

HeMan, 2023-07-19:
Is it solved by https://github.com/esphome/esphome/pull/5102?

JohnGalt1717, 2023-07-19:
Looks like it probably should... Is it in release yet?

HeMan, 2023-07-19:
I was pretty sure it was, but no, it's not part of any release as of now.

jesserockz, 2023-07-19:
> Looks like it probably should... Is it in release yet?

Its merged into dev and an issue when developing so it doesnt need to go into any specific release.

closed 2023-07-19 by jesserockz

JohnGalt1717, 2023-07-21:
Ok. Appears to not be in .7.