esphome/issues #4679: PWM Output on RP2040 does not support higher frequencies

MarkuBu, 2023-07-09:
### The problem

I want to output a PWM frequency of 25kHz for a PWM PC fan to control speed. It is no problem with MicroPython

2000 Hz work well, 2500 Hz result in 41Hz

Is this a bug or how can I configure a PWM output of 25kHz?

### Which version of ESPHome has the issue?

2023.6.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.7.1

### What platform are you using?

RP2040

### Board

Raspberry Py Pico W

### Example YAML snippet

```yaml
output:
  - platform: rp2040_pwm
    pin: 28
    frequency: 5000 Hz
    id: fanhub_pwm
fan:
  - platform: speed
    output: fanhub_pwm
    name: "PWM Fan"
```

### Anything in the logs that might be useful for us?

```txt
[16:23:56][C][rp2040_pwm:039]: RP2040 PWM:
[16:23:56][C][rp2040_pwm:040]:   Pin: GPIO28
[16:23:56][C][rp2040_pwm:041]:   Frequency: 5000.0 Hz
[16:23:56][C][speed.fan:016]: Speed Fan 'PWM Fan'
[16:23:56][C][speed.fan:151]:   Speed: YES
[16:23:56][C][speed.fan:152]:   Speed count: 100
```

matemaciek, 2023-08-05:
May be related: https://github.com/esphome/esphome/pull/4294

matemaciek, 2023-08-05:
OK, I know the issue. ~2.1kHz is where clock divider drops below 1, and according to [the documentation of `pwm_config_set_clkdiv`](https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#gga0a00197703c8606641488cfff3f660e6acd3efbc3feaca58b87930008fdd8486b) it should be greater than or equal to 1. I'll try fix it (-:

closed 2023-08-06 by jesserockz