esphome/issues #4681: SelecEM2M Power Factor unit of measurement

hellcry37, 2023-07-10:
### The problem

Warning in logs and statistics sensors

I see this warning in logs all the time. I do not know from what version this happens.

### Anything in the logs that might be useful for us?

### Which version of ESPHome has the issue?

2023.6.4

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.7.1

### What platform are you using?

ESP32

### Board

esp32devkit

### Component causing the issue

...

### Example YAML snippet

```yaml
esphome:

substitutions:
  name: selec-em2m-energy-monitor
  friendly_name: "Selec-Em2M-Energy-Monitor"
  ssid_name: "${friendly_name}"

esphome:
  name: "${name}"

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret home_assistant_encryption_key

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: .home
  manual_ip:
    static_ip: 192.168.1.xx
    gateway: 192.168.1.x
    subnet: 255.255.255.0
    dns1: 192.168.1.x
    dns2: 192.168.1.x
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${ssid_name}"
    password: !secret default_fallback_ap_pass

captive_portal:

# Selec Em2M Energy Monitor configuration
uart:
  id: mod_bus
  rx_pin: 16
  tx_pin: 17
  baud_rate: 9600
  stop_bits: 1

sensor:
  - platform: selec_meter
    total_active_energy:
      name: "SelecEM2M Total Active Energy"
    import_active_energy:
      name: "SelecEM2M Import Active Energy"
    export_active_energy:
      name: "SelecEM2M Export Active Energy"
    total_reactive_energy:
      name: "SelecEM2M Total Reactive Energy"
    import_reactive_energy:
      name: "SelecEM2M Import Reactive Energy"
    export_reactive_energy:
      name: "SelecEM2M Export Reactive Energy"
    apparent_energy:
      name: "SelecEM2M Apparent Energy"
    active_power:
      name: "SelecEM2M Active Power"
      accuracy_decimals: 1
    reactive_power:
      name: "SelecEM2M Reactive Power"
      accuracy_decimals: 1
    apparent_power:
      name: "SelecEM2M Apparent Power"
      accuracy_decimals: 1
    voltage:
      name: "SelecEM2M Voltage"
    current:
      name: "SelecEM2M Current"
    power_factor:
      name: "SelecEM2M Power Factor"
    frequency:
      name: "SelecEM2M Frequency"
    maximum_demand_active_power:
      name: "SelecEM2M Maximum Demand Active Power"
    maximum_demand_reactive_power:
      name: "SelecEM2M Maximum Demand Reactive Power"
    maximum_demand_apparent_power:
      name: "SelecEM2M Maximum Demand Apparent Power"
```

### Anything in the logs that might be useful for us?

```txt
Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:659
Integration: Sensor (documentation, issues)
First occurred: 20:48:47 (1 occurrences)
Last logged: 20:48:47

Entity sensor.selecem2m_power_factor (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement '' which is not a valid unit for the device class ('power_factor') it is using; expected one of ['%', 'no unit of measurement']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22
```

### Additional information

This started in the last 2 or 3 esphome relases i think, I am not 100% sure.

This was posted here also:

https://github.com/home-assistant/core/issues/96169

But I was guided here.

jossiee, 2023-07-11:
Same issue with the pzemac platform

Entity sensor.airco_power_factor (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement '' which is not a valid unit for the device class ('power_factor') it is using; expected one of ['%', 'no unit of measurement']

DAVe3283, 2023-07-11:
It appears Home Assistant is receiving native_unit_of_measurement as an empty string instead of a None, and throwing this error. I am not sure this problem can be fixed on this side of the API, unless the API has some way of indicating a null/none vs. a 0-length string.

closed 2023-10-11 by hellcry37