esphome/issues #4685: ESPHome device unavailable after HA Reboot or update. Reinstall necessary.

nlyaple, 2023-07-10:
### The problem

I have multiple ESPHome devices but two that use Deep Sleep in the yaml file report unavailable after HA is rebooted or updated. One of the devices is an ESP32 and the other is an ESP8266.   The only thing that these devices have in common are the Deep Sleep entry as far as I can tell.  The devices both show up correctly in the ESPHome tab but the integration shows them as unavailable.  Doing an Install or Update corrects the problem.

### Which version of ESPHome has the issue?

2023.6.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.6 and later

### What platform are you using?

ESP8266

### Board

NodeMCU and ESP32 T-Relay (Wroom) based board

### Component causing the issue

Device unavailable.

### Example YAML snippet

```yaml
esphome:
  name: solarshade
    
esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

ota:
  password: "xxxxxxxxxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Solarshade Fallback Hotspot"
    password: "xxxxxxxxxxxx"

captive_portal:

deep_sleep:
  id: deep_sleep_1

sensor:
  - platform: wifi_signal
    name: "Solar Shade Signal Strength"
    update_interval: 60s

  - platform: adc
    name: "Battery voltage"
    pin: A0
    accuracy_decimals: 2
    update_interval: 300s
    filters:
      - multiply: 86.00  # The voltage divider requires us to multiply

# Reed switch between GPIO12 and GND
  - platform: pulse_counter
    name: "Rotation"
    pin:
      number: 12
      inverted: true
      mode:
        input: true
        pullup: true
    update_interval: 60s

binary_sensor:        
  - platform: gpio
    name: "Up Limit"
    id: "upper_Limit"
    pin: 
      number: 14
      inverted: true
      mode:
        input: true
        pullup: true
    on_press:
      then:
        - switch.turn_off: Raise_Shade

switch:
  - platform: gpio
    id: Enable_Shade
    name: "Enable"
    pin: 13
    restore_mode: ALWAYS_ON
  - platform: gpio
    name: "Down"
    id: "Lower_Shade"
    pin: 5
    interlock: [Raise_Shade]
  - platform: gpio
    name: "Up"
    id: "Raise_Shade"
    pin: 4
    interlock: [Lower_Shade]
  - platform: gpio
    name: "Blue LED"
    inverted: true
    pin: 2

  - platform: template
    name: "Sleep"
    turn_on_action:
      then:
      - deep_sleep.enter:
          id: deep_sleep_1
          sleep_duration: 12h
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.6.5
INFO Reading configuration /config/esphome/solarshade.yaml...
INFO Starting log output from solarshade.local using esphome API
INFO Successfully connected to solarshade.local
[12:53:04][I][app:102]: ESPHome version 2023.6.5 compiled on Jul 10 2023, 07:07:37
[12:53:04][C][wifi:543]: WiFi:
[12:53:04][C][wifi:379]:   Local MAC: A0:20:A6:17:45:49
[12:53:04][C][wifi:380]:   SSID: [redacted]
[12:53:04][C][wifi:381]:   IP Address: 192.168.0.154
[12:53:04][C][wifi:382]:   BSSID: [redacted]
[12:53:04][C][wifi:384]:   Hostname: 'solarshade'
[12:53:04][C][wifi:386]:   Signal strength: -57 dB ▂▄▆█
[12:53:04][C][wifi:390]:   Channel: 6
[12:53:04][C][wifi:391]:   Subnet: 255.255.255.0
[12:53:04][C][wifi:392]:   Gateway: 192.168.0.1
[12:53:04][C][wifi:393]:   DNS1: 192.168.0.1
[12:53:04][C][wifi:394]:   DNS2: 192.168.0.1
[12:53:04][C][logger:301]: Logger:
[12:53:04][C][logger:302]:   Level: DEBUG
[12:53:04][C][logger:303]:   Log Baud Rate: 115200
[12:53:04][C][logger:305]:   Hardware UART: UART0
[12:53:04][C][switch.gpio:068]: GPIO Switch 'Enable'
[12:53:04][C][switch.gpio:090]:   Restore Mode: always ON
[12:53:04][C][switch.gpio:031]:   Pin: GPIO13
[12:53:04][C][switch.gpio:068]: GPIO Switch 'Down'
[12:53:04][C][switch.gpio:090]:   Restore Mode: always OFF
[12:53:04][C][switch.gpio:031]:   Pin: GPIO5
[12:53:04][C][switch.gpio:033]:   Interlocks:
[12:53:04][C][switch.gpio:037]:     Up
[12:53:04][C][switch.gpio:068]: GPIO Switch 'Up'
[12:53:04][C][switch.gpio:090]:   Restore Mode: always OFF
[12:53:04][C][switch.gpio:031]:   Pin: GPIO4
[12:53:04][C][switch.gpio:033]:   Interlocks:
[12:53:04][C][switch.gpio:037]:     Down
[12:53:04][C][switch.gpio:068]: GPIO Switch 'Blue LED'
[12:53:04][C][switch.gpio:076]:   Inverted: YES
[12:53:04][C][switch.gpio:090]:   Restore Mode: always OFF
[12:53:04][C][switch.gpio:031]:   Pin: GPIO2
[12:53:04][C][template.switch:068]: Template Switch 'Sleep'
[12:53:04][C][template.switch:090]:   Restore Mode: always OFF
[12:53:04][C][template.switch:060]:   Restore State: NO
[12:53:04][C][template.switch:061]:   Optimistic: NO
[12:53:04][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Up Limit'
[12:53:04][C][gpio.binary_sensor:016]:   Pin: GPIO14
[12:53:04][C][adc:087]: ADC Sensor 'Battery voltage'
[12:53:04][C][adc:087]:   Device Class: 'voltage'
[12:53:04][C][adc:087]:   State Class: 'measurement'
[12:53:04][C][adc:087]:   Unit of Measurement: 'V'
[12:53:04][C][adc:087]:   Accuracy Decimals: 2
[12:53:04][C][adc:092]:   Pin: GPIO17
[12:53:04][C][adc:126]:   Update Interval: 300.0s
[12:53:04][C][pulse_counter:160]: Pulse Counter 'Rotation'
[12:53:04][C][pulse_counter:160]:   State Class: 'measurement'
[12:53:04][C][pulse_counter:160]:   Unit of Measurement: 'pulses/min'
[12:53:04][C][pulse_counter:160]:   Accuracy Decimals: 2
[12:53:04][C][pulse_counter:160]:   Icon: 'mdi:pulse'
[12:53:04][C][pulse_counter:161]:   Pin: GPIO12
[12:53:04][C][pulse_counter:162]:   Rising Edge: INCREMENT
[12:53:04][C][pulse_counter:163]:   Falling Edge: DISABLE
[12:53:04][C][pulse_counter:164]:   Filtering pulses shorter than 13 µs
[12:53:04][C][pulse_counter:165]:   Update Interval: 60.0s
[12:53:04][C][captive_portal:088]: Captive Portal:
[12:53:04][C][mdns:112]: mDNS:
[12:53:04][C][mdns:113]:   Hostname: solarshade
[12:53:04][C][ota:093]: Over-The-Air Updates:
[12:53:04][C][ota:094]:   Address: solarshade.local:8266
[12:53:04][C][ota:097]:   Using Password.
[12:53:04][C][api:138]: API Server:
[12:53:04][C][api:139]:   Address: solarshade.local:6053
[12:53:04][C][api:141]:   Using noise encryption: YES
[12:53:04][C][wifi_signal.sensor:009]: WiFi Signal 'Solar Shade Signal Strength'
[12:53:04][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[12:53:04][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[12:53:04][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[12:53:04][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[12:53:04][C][deep_sleep:049]: Setting up Deep Sleep...
```

### Additional information

[image]

fran6120, 2023-07-10:
I have the same problem in an esp32c3 in which I also use the deep_sleep function. Until now I had not opened a thread mentioning the bug, because I did not know if the problem was related to esphome, with the integration or home assistant, but since the topic has been opened here, I take this opportunity to share my experience.

Neither reloading the integration nor restarting home assistant fixes it, the only solution I have found is to re-upload the firmware to the esp32c3.

bdraco, 2023-07-14:
https://github.com/home-assistant/core/pull/96354

fran6120, 2023-07-14:
I can confirm that hass 2023.7.2 resolves this

closed 2023-11-27 by github-actions[bot]