esphome/issues #4687: deep_sleep.prevent only works while ESP is not sleeping

KaiserSoft, 2023-07-11:
### The problem

Attempting to disable deep_sleep over MQTT while the ESP is sleeping will not disable deep sleep when it wakes up. Sending the MQTT message while the ESP is running (after it wakes up) will disable deep sleep.
Tested with retrain True and False.

Also tested with exact code from https://esphome.io/components/deep_sleep.html?highlight=deep_sleep#deep-sleep-prevent-action

### Which version of ESPHome has the issue?

2023.6.5

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

NA

### What platform are you using?

ESP8266

### Board

esp12e

### Component causing the issue

deep_sleep

### Example YAML snippet

```yaml
substitutions:
  devName: "Sensor1"
  devNameLower: "sensor1"
  devId: "sensor1-e34"
  devDomain: ".intra.foo.bar"

esphome:
  name: $devNameLower
  platform: ESP8266
  board: esp12e

wifi:
  ssid: "SomeSSID"
  password: "SuperSecret42"
  domain: "$devDomain"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Recover Me"
    password: "Rot26SuperSecret42"

time:
  - platform: sntp
    id: sntp_time
    timezone: "Europe/Berlin"
    servers: "ntp.intra.foo.bar"

captive_portal:

# https://esphome.io/components/logger.html
# NONE | ERROR | WARN | INFO | DEBUG | VERBOSE | VERY_VERBOSE
logger:
  level: DEBUG

ota:
  password: "TopSecret42"

deep_sleep:
  run_duration: 2min
  sleep_duration: 1min
  id: deep_sleep_1

mqtt:
  broker: mqtt.intra.foo.bar
  on_message:
    - topic: ${devNameLower}/deepSleep
      payload: 'OFF'
      then:
        - deep_sleep.prevent: deep_sleep_1
        - logger.log: "deepSleep is OFF"
    - topic: ${devNameLower}/deepSleep
      payload: 'ON'
      then:
        - deep_sleep.enter: deep_sleep_1
        - logger.log: "deepSleep is ON"
```

### Anything in the logs that might be useful for us?

```txt
**Log When Device Starts, sending OFF to sensor1/deepSleep and I receive the log message "deepSleep is OFF"**
INFO ESPHome 2023.6.5
INFO Reading configuration /config/sensor1.yaml...
INFO Starting log output from sensor1/debug
INFO Connected to MQTT broker!
[19:33:32][C][deep_sleep:055]:   Run Duration: 120000 ms
[19:34:23][I][mqtt:286]: MQTT Connected!
[19:34:23][C][deep_sleep:037]: Setting up Deep Sleep...
[19:34:23][I][deep_sleep:042]: Scheduling Deep Sleep to start in 120000 ms
[19:34:23][I][app:062]: setup() finished successfully!
[19:34:24][I][app:102]: ESPHome version 2023.6.5 compiled on Jul 11 2023, 19:29:44
[19:34:24][C][wifi:543]: WiFi:
[19:34:24][C][wifi:379]:   Local MAC: XX:XX:XX:XX:XX:XX
[19:34:24][C][wifi:380]:   SSID: [redacted]
[19:34:24][C][wifi:381]:   IP Address: 192.168.xx.xx
[19:34:24][C][wifi:382]:   BSSID: [redacted]
[19:34:24][C][wifi:384]:   Hostname: 'sensor1'
[19:34:24][C][wifi:386]:   Signal strength: -73 dB ▂▄▆█
[19:34:24][C][wifi:390]:   Channel: 1
[19:34:24][C][wifi:391]:   Subnet: 255.255.255.XX
[19:34:24][C][wifi:392]:   Gateway: 192.168.XX.XX
[19:34:24][C][wifi:393]:   DNS1: 192.168.XX.XX
[19:34:24][C][wifi:394]:   DNS2: 0.0.0.0
[19:34:24][C][logger:301]: Logger:
[19:34:24][C][logger:302]:   Level: DEBUG
[19:34:24][C][logger:303]:   Log Baud Rate: 115200
[19:34:24][C][logger:305]:   Hardware UART: UART0
[19:34:24][C][captive_portal:088]: Captive Portal:
[19:34:24][C][sntp:053]: SNTP Time:
[19:34:24][C][sntp:054]:   Server 1: 'ntp.intra.foo.bar'
[19:34:24][C][sntp:055]:   Server 2: ''
[19:34:24][C][sntp:056]:   Server 3: ''
[19:34:24][C][sntp:057]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[19:34:24][C][mdns:112]: mDNS:
[19:34:24][C][mdns:113]:   Hostname: sensor1
[19:34:24][C][ota:093]: Over-The-Air Updates:
[19:34:24][C][ota:094]:   Address: sensor1.intra.foo.bar:8266
[19:34:24][C][ota:097]:   Using Password.
[19:34:24][C][mqtt:127]: MQTT:
[19:34:24][C][mqtt:128]:   Server Address: mqtt.intra.foo.bar:1883 (192.168.XX.XX)
[19:34:24][C][mqtt:130]:   Username: [redacted]
[19:34:24][C][mqtt:131]:   Client ID: [redacted]
[19:34:24][C][mqtt:133]:   Discovery prefix: 'homeassistant'
[19:34:24][C][mqtt:134]:   Discovery retain: YES
[19:34:24][C][mqtt:136]:   Topic Prefix: 'sensor1'
[19:34:24][C][mqtt:138]:   Log Topic: 'sensor1/debug'
[19:34:24][C][mqtt:141]:   Availability: 'sensor1/status'
[19:34:24][C][mqtt:680]: MQTT Message Trigger:
[19:34:24][C][mqtt:681]:   Topic: 'sensor1/deepSleep'
[19:34:24][C][mqtt:682]:   QoS: 0
[19:34:24][C][mqtt:680]: MQTT Message Trigger:
[19:34:24][C][mqtt:681]:   Topic: 'sensor1/deepSleep'
[19:34:24][C][mqtt:682]:   QoS: 0
[19:34:24][C][deep_sleep:049]: Setting up Deep Sleep...
[19:34:24][C][deep_sleep:052]:   Sleep Duration: 60000 ms
[19:34:24][C][deep_sleep:055]:   Run Duration: 120000 ms
[19:34:51][D][main:297]: deepSleep is OFF

**Now I remove power and plug it back in. MQTT topic is still "OFF" but deep_sleep starts after 2 minutes**
INFO ESPHome 2023.6.5
INFO Reading configuration /config/sensor1.yaml...
INFO Starting log output from sensor1/debug
INFO Connected to MQTT broker!
[19:39:10][D][main:297]: deepSleep is OFF
[19:40:19][I][mqtt:286]: MQTT Connected!
[19:40:20][C][deep_sleep:037]: Setting up Deep Sleep...
[19:40:20][I][deep_sleep:042]: Scheduling Deep Sleep to start in 120000 ms
[19:40:20][I][app:062]: setup() finished successfully!
[19:40:20][I][app:102]: ESPHome version 2023.6.5 compiled on Jul 11 2023, 19:29:44
[19:40:20][C][wifi:543]: WiFi:
[19:40:20][C][wifi:379]:   Local MAC: XX:XX:XX:XX:XX:XX
[19:40:20][C][wifi:380]:   SSID: [redacted]
[19:40:20][C][wifi:381]:   IP Address: 192.168.XX.XX
[19:40:20][C][wifi:382]:   BSSID: [redacted]
[19:40:20][C][wifi:384]:   Hostname: 'sensor1'
[19:40:20][C][wifi:386]:   Signal strength: -68 dB ▂▄▆█
[19:40:20][C][wifi:390]:   Channel: 1
[19:40:20][C][wifi:391]:   Subnet: 255.255.255.XX
[19:40:20][C][wifi:392]:   Gateway: 192.168.XX.XX
[19:40:20][C][wifi:393]:   DNS1: 192.168.XX.XX
[19:40:20][C][wifi:394]:   DNS2: 0.0.0.0
[19:40:20][C][logger:301]: Logger:
[19:40:20][C][logger:302]:   Level: DEBUG
[19:40:20][C][logger:303]:   Log Baud Rate: 115200
[19:40:20][C][logger:305]:   Hardware UART: UART0
[19:40:20][C][captive_portal:088]: Captive Portal:
[19:40:20][C][sntp:053]: SNTP Time:
[19:40:20][C][sntp:054]:   Server 1: 'ntp.intra.foo.bar'
[19:40:20][C][sntp:055]:   Server 2: ''
[19:40:20][C][sntp:056]:   Server 3: ''
[19:40:20][C][sntp:057]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[19:40:20][C][mdns:112]: mDNS:
[19:40:20][C][mdns:113]:   Hostname: sensor1
[19:40:20][C][ota:093]: Over-The-Air Updates:
[19:40:20][C][ota:094]:   Address: sensor1.intra.foo.bar:8266
[19:40:20][C][ota:097]:   Using Password.
[19:40:20][C][mqtt:127]: MQTT:
[19:40:20][C][mqtt:128]:   Server Address: mqtt.intra.foo.bar:1883 (192.168.XX.XX)
[19:40:20][C][mqtt:130]:   Username: [redacted]
[19:40:20][C][mqtt:131]:   Client ID: [redacted]
[19:40:20][C][mqtt:133]:   Discovery prefix: 'homeassistant'
[19:40:20][C][mqtt:134]:   Discovery retain: YES
[19:40:20][C][mqtt:136]:   Topic Prefix: 'sensor1'
[19:40:20][C][mqtt:138]:   Log Topic: 'sensor1/debug'
[19:40:20][C][mqtt:141]:   Availability: 'sensor1/status'
[19:40:20][C][mqtt:680]: MQTT Message Trigger:
[19:40:20][C][mqtt:681]:   Topic: 'sensor1/deepSleep'
[19:40:20][C][mqtt:682]:   QoS: 0
[19:40:20][C][mqtt:680]: MQTT Message Trigger:
[19:40:20][C][mqtt:681]:   Topic: 'sensor1/deepSleep'
[19:40:20][C][mqtt:682]:   QoS: 0
[19:40:20][C][deep_sleep:049]: Setting up Deep Sleep...
[19:40:20][C][deep_sleep:052]:   Sleep Duration: 60000 ms
[19:40:20][C][deep_sleep:055]:   Run Duration: 120000 ms
[19:42:20][I][deep_sleep:116]: Beginning Deep Sleep
[19:42:20][I][deep_sleep:118]: Sleeping for 60000000us

MQTT topic history (last changed)
[19:34:51] sensor1/deepSleep OFF
```

### Additional information

Same issue on ESPhome from 9 month ago. I upgraded to latest 2023.6.5 to see if bug is fixed. I can replicate the problem on all three devices. They are ESP8266MOD which is an ESP-12F

ssieb, 2023-07-12:
Is the message set to be retained?

photomoose, 2023-08-19:
I'm also seeing this problem. Going by the logs, it looks like the MQTT message is processed and disables the deep sleep before the deep sleep component has initialised:

```
INFO ESPHome 2023.7.1
INFO Reading configuration /config/hoseduino.yaml...
INFO Starting log output from hoseduino/debug
INFO Connected to MQTT broker!
[22:22:23][I][deep_sleep:118]: Sleeping for 7200000000us
[22:22:46][I][mqtt:286]: MQTT Connected!
[22:22:46][I][app:102]: ESPHome version 2023.7.1 compiled on Aug 19 2023, 21:54:06
...
[22:22:46][D][main:423]: Deep Sleep disabled
[22:22:46][D][main:423]: Deep Sleep disabled
...
[22:22:47][C][deep_sleep:049]: Setting up Deep Sleep...
[22:22:47][C][deep_sleep:052]:   Sleep Duration: 7200000 ms
[22:22:47][C][deep_sleep:059]:   Wakeup Pin: GPIO15
[22:22:47][C][deep_sleep:062]:   Default Wakeup Run Duration: 60000 ms
[22:22:47][C][deep_sleep:063]:   Touch Wakeup Run Duration: 60000 ms
[22:22:47][C][deep_sleep:064]:   GPIO Wakeup Run Duration: 300000 ms
...
[22:27:38][I][ota:113]: Boot seems successful, resetting boot loop counter.
[22:27:38][D][esp32.preferences:114]: Saving 1 preferences to flash...
[22:27:38][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[22:27:46][I][deep_sleep:116]: Beginning Deep Sleep
[22:27:46][I][deep_sleep:118]: Sleeping for 7200000000us
```

ssieb, 2023-08-20:
You need to watch the serial logs to determine the order and to see if the message is getting received.

EOUpSL93Y, 2024-05-16:
Same problem

ssieb, 2024-05-16:
@EOUpSL93Y and same question for you.  I suggest asking on discord.

photomoose, 2024-05-16:
I retract my previous comment, deep sleep is working fine for me. It's definitely worth viewing the logs over a wired connection, you see a lot more than what you get over wireless. For me, my problem was caused by an MQTT trigger which *enabled* deep sleep each time the ESP32 woke up. I forgot that using retained MQTT messages results in the message being received and actioned _each time the MQTT connection is established_ - and not just when the message is first published.

```
mqtt:
  broker: 10.0.0.8
  username: hoseduino
  password: !secret hoseduino_mqtt_password
  discovery: true
  on_message:
    - topic: hoseduino/deep_sleep
      payload: "DISABLED"
      then:
        - logger.log: "Deep Sleep disabled"
        - deep_sleep.prevent: deep_sleep_1
        
    # THE FOLLOWING WILL ALWAYS HAPPEN IF YOU HAVE THE RETAIN FLAG SET
    - topic: hoseduino/deep_sleep
      payload: "ENABLED"
      then:
        - logger.log: "Deep Sleep enabled"
        - deep_sleep.allow: deep_sleep_1
```

closed 2024-05-16 by ssieb