esphome/issues #4688: defining second UART might block connection to API [Errno 111] with logger enabled on one of the pins used

drahdiwaberl, 2023-07-11:
### The problem

tried to use this:

```
uart:
  - id: uart_0
    rx_pin: GPIO19
    baud_rate: 19200
    rx_buffer_size: 256
  - id: uart_1
    tx_pin: GPIO21
    rx_pin: GPIO20
    baud_rate: 9600
    rx_buffer_size: 512
```

which, after flashing, when usually the log comes up, resulted in this:
`WARNING Can't connect to ESPHome API for batt-control.local: Error connecting to ('192.168.1.196', 6053): [Errno 111]`

after a lot of debugging i happened to realize that this:

```
uart:
  - id: uart_1
    tx_pin: GPIO21
    rx_pin: GPIO20
    baud_rate: 9600
    rx_buffer_size: 512
```

brought up a warning:

`You're using the same serial port for logging and the UART component.`

.. and after setting the logger's baud_rate to zero as suggested, i'm now able to use both UARTs with the definition above just fine.

### Which version of ESPHome has the issue?

2023.6.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.7.1

### What platform are you using?

ESP32

### Board

Lilygo mini D1 plus

### Component causing the issue

UART

ssieb, 2023-07-12:
So what's the issue?

drahdiwaberl, 2023-07-12:
when having both UARTs defined, the warning doesn't come up - so it's not clear what's wrong

ssieb, 2023-07-13:
Then what is this issue about?

drahdiwaberl, 2023-07-13:
nil

closed 2023-07-13 by drahdiwaberl