esphome/issues #4689: esp 32 does not work properly

fredycz2005, 2023-07-12:
### The problem

hello,
my esp 32 does not send any data with temperature from 5AM and i have to manualy restart it every morning.
this is my config:
esphome:
  name: teplomer-bazen
  friendly_name: teplomer_bazen

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "OAm/UQxu1K81W05MTqccHVwH2Ag67P4W6CIlllWageU="

ota:
  password: "5b72e80fa210fc68e7ac509604a47b41"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  #ap:
   # ssid: "Teplome-Bazen Fallback Hotspot"
   # password: "oPpNR3V71bVp"

#captive_portal:
    
dallas:
  - pin: GPIO13
    update_interval: 10s
sensor:
  - platform: dallas
    address: 0x8d3c60e381b97728
    name: "bazen teplota"
    filters:
      - offset: -1.0

deep_sleep: 
  id: deep_sleep_1
  run_duration: 5min
  sleep_duration: 55min

### Which version of ESPHome has the issue?

2023.6.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.7.1

### What platform are you using?

ESP32

### Board

esp32

### Component causing the issue

DS18B20

ssieb, 2023-07-13:
You'll need to watch the serial logs from when it stops working.

bdraco, 2023-07-14:
https://github.com/home-assistant/core/pull/96354

closed 2023-11-27 by github-actions[bot]