esphome/issues #4690: teleinfo : bad crc on EASD02 tag

MrRady, 2023-07-12:
### The problem

I use and ESP32 with an tic2uart adapter : http://hacks.slashdirt.org/hw/tic2uart/
I have a three phase linky counter, with standard TIC mode enabled.

I have checked signals with a scope and they are clean.

So i have configured all tag to be parsed, and activated debug on UART , 
and this is always this tag appearing to be the problem **EASD02**

Here in the log : 
\nEASD02\t00134230\t%\r
correspond to : 
[09:12:27][E][teleinfo:038]: bad crc: got 37 except 62
So it's got %(37) and expect ">"(62)

If i calculate myself, i find 62 too ! The CRC sent to the counter seems to be wrong.
The electricat counter format a wrong crc for you ?

### Which version of ESPHome has the issue?

2023.2.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.07.1

### What platform are you using?

ESP32

### Board

lilygo-t-eth-poe

### Component causing the issue

teleinfo

### Example YAML snippet

```yaml
esphome:
  name: esphome-web-cb7144
  friendly_name: LINKY

esp32:
  board: esp32dev
  framework:
    type: arduino #esp-idf
external_components:
  - source: github://esphome/esphome@dev
    components: [ ethernet, teleinfo ]
# Enable logging
logger:
  baud_rate: 0   # disable logging via UART, help to avoid numerous crash with ESP_LOGD
  hardware_uart : UART0
  level: DEBUG #NONE ERROR WARN INFO DEBUG VERBOSE VERY_VERBOSE
# Enable Home Assistant API
api:
  encryption:
    key: "bTQGEVlkMMN9LswidXac521P41ex+13HrTE1X7aERXY="

ota:

ethernet:
[... 608 lines cut ...]
[09:12:27][D][sensor:127]: 'Linky Energie Soutirée D04': Sending state 0.00000 kWh with 2 decimals of accuracy
[09:12:27][D][sensor:127]: 'Linky Courant Efficace Phase 1': Sending state 3.00000 A with 0 decimals of accuracy
[09:12:27][D][sensor:127]: 'Linky Courant Efficace Phase 2': Sending state 1.00000 A with 0 decimals of accuracy
[09:12:27][D][sensor:127]: 'Linky Courant Efficace Phase 3': Sending state 0.00000 A with 0 decimals of accuracy
[09:12:27][D][sensor:127]: 'Linky Tension Efficace Phase 1': Sending state 230.00000 V with 0 decimals of accuracy
[09:12:27][D][sensor:127]: 'Linky Tension Efficace Phase 2': Sending state 238.00000 V with 0 decimals of accuracy
[09:12:27][D][sensor:127]: 'Linky Tension Efficace Phase 3': Sending state 232.00000 V with 0 decimals of accuracy
[09:12:27][D][sensor:127]: 'Linky Puissance Apparente Référence': Sending state 18000.00000 VA with 0 decimals of accuracy
[09:12:27][D][sensor:127]: 'Linky Puissance Apparente Coupure': Sending state 18000.00000 VA with 0 decimals of accuracy
[09:12:27][D][sensor:127]: 'Linky Puissance Apparente Instantanée Soutirée': Sending state 766.00000 VA with 0 decimals of accuracy
[09:12:27][D][sensor:127]: 'Linky Puissance Apparente Instantanée Soutirée Phase 1': Sending state 581.00000 VA with 0 decimals of accuracy
[09:12:27][D][sensor:127]: 'Linky Puissance Apparente Instantanée Soutirée Phase 2': Sending state 133.00000 VA with 0 decimals of accuracy
[09:12:27][D][sensor:127]: 'Linky Puissance Apparente Instantanée Soutirée Phase 3': Sending state 52.00000 VA with 0 decimals of accuracy
[09:12:27][D][sensor:127]: 'Linky Puissance Apparente Maximale Soutirée': Sending state 4536.00000 VA with 0 decimals of accuracy
[09:12:27][D][sensor:127]: 'Linky Puissance Apparente Maximale Soutirée Phase 1': Sending state 2665.00000 VA with 0 decimals of accuracy
[09:12:27][D][sensor:127]: 'Linky Puissance Apparente Maximale Soutirée Phase 2': Sending state 2063.00000 VA with 0 decimals of accuracy
[09:12:27][D][sensor:127]: 'Linky Puissance Apparente Maximale Soutirée Phase 3': Sending state 81.00000 VA with 0 decimals of accuracy
[09:12:27][D][sensor:127]: 'Linky Puissance Active Soutirée': Sending state 608.00000 W with 0 decimals of accuracy
[09:12:27][D][sensor:127]: 'Linky Puissance Active Soutirée N-1': Sending state 1316.00000 W with 0 decimals of accuracy
[09:12:27][D][sensor:127]: 'Linky Tension Moyenne Phase 1': Sending state 230.00000 V with 0 decimals of accuracy
[09:12:27][D][sensor:127]: 'Linky Tension Moyenne Phase 2': Sending state 237.00000 V with 0 decimals of accuracy
[09:12:27][D][sensor:127]: 'Linky Tension Moyenne Phase 3': Sending state 231.00000 V with 0 decimals of accuracy
[09:12:27][D][text_sensor:067]: 'Linky Registre de Statuts': Sending state '003A4001'
[09:12:27][D][uart_debug:158]: <<< "\x03"
```

Toniob, 2023-11-10:
Did you find a solution to the issue ?

alucryd, 2023-11-19:
I'm also affected by this, definitely still an issue, haven't found a solution yet.