esphome/issues #4692: Error in documentation: Automations and Template -> global variables

hiperiondev, 2023-07-12:
### The problem

in the example say:
           ESP_LOGD(TAG, "%s: %d", id(my_global_string), id(my_global_int));

but doesn't work.

must be:
        ESP_LOGD(TAG, "%s: %d", id(my_global_string).c_str(), id(my_global_int));

### Which version of ESPHome has the issue?

documentation

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

nielsnl68, 2023-07-13:
Good that you spotted this issue.
Just a suggestion, did you know you can edit the page yourself and create a PR for that change.

hiperiondev, 2023-07-14:
> Good that you spotted this issue. Just a suggestion, did you know you can edit the page yourself and create a PR for that change.

Yes. As soon as I have a little time, I will definitely do it.

KevynWithaY, 2023-07-18:
Created a pull request for this with a correction. Hopefully I did that correctly.
https://github.com/esphome/esphome-docs/pull/3076

hiperiondev, 2023-07-20:
Corrected

closed 2023-07-20 by hiperiondev