esphome/issues #4701: on the lastest ESPhome version with HA, SonOff Basic R1 lost connectivity

drumstick77, 2023-07-15:
### The problem

All sonoff basic R1 lost connectivity from the 2nd last update. flashing via USB does not resolve the issue.

### Which version of ESPHome has the issue?

2023.6.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.7.2

### What platform are you using?

ESP8266

### Board

Sonoff basic R1

### Example YAML snippet

```yaml
esphome:
  name: sonoff-basic1
  friendly_name: SonOff Basic1

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"

ota:
  password: "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Sonoff-Basic1 Fallback Hotspot"
    password: "nK3q2d6pMHAS"

captive_portal:

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode:
        input: true
        pullup: true
      inverted: true
    name: "Sonoff Basic Button"
    on_press:
      - switch.toggle: relay

switch:
  - platform: gpio
    name: "Sonoff Basic Relay"
    pin: GPIO12
    id: relay

status_led:
  pin:
    number: GPIO13
    inverted: yes    

web_server:
```

bigmoby, 2023-07-19:
The same on `Version: 2023.6.2`

drumstick77, 2023-07-19:
So, no solution then? I kinda given up and started to migrate my stuff away from ESPHOME............

drumstick77, 2023-07-19:
Sorry to digress slightly, but I got some localbytes plugs that I've converted to ESPHOME and ESPHOME is now complaining that the yml needs to be updated from restore_state to restore-mode, but the yml is from jamesswift github which means that he will need update his codes. I don't know what ESPHOME has been doing, it is so annoying that has caused this issue on my devices!

bigmoby, 2023-07-19:
> [...] the yml needs to be updated from restore_state to restore-mode, but the yml is from jamesswift github which means that he will need update his codes. [...]

Have you updated Esphome to version 2023.7?  Have a look at breaking change list.

drumstick77, 2023-07-19:
I am on 2023.7

bigmoby, 2023-07-19:
> I am on 2023.7

Ok...so have look at the pr link I posted

bkaufx, 2023-07-19:
Do you see anything in the logs when connected by USB?

bigmoby, 2023-07-20:
Sure, this reason is logged:

```
reason='probe request unsuccessful'
```