esphome/issues #4703: MQTT client subscribes to undefined topics "esphome/discover/..." and "esphome/ping/..." during setup.

sprouse, 2023-07-16:
### The problem

I believe I've found an issue with this section of code when working with the Adafruit IO mqtt service.

These topics do not always exist on the target mqtt server.

In the case of Adafruit, they ban the client for a certain amount of time if it makes too many requests for undefined topics.

Is it possible to guard these subscriptions with a compile flag so they can be controlled with a configuration paramter in the YAML file?

### Which version of ESPHome has the issue?

2023.6.5

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

mqtt

### Anything in the logs that might be useful for us?

```txt
Adafruit IO gives this error when the esphome mqtt client starts to connect:

MQTT ERROR: XXX XX  SUBSCRIBE to topic esphome/ping/<XX> rejected, not a valid resource topic
MQTT ERROR: enforcement limit reached, your account is banned for 60 seconds
```

### Additional information

I believe the issue is with these subscriptions in mqtt_client.cpp
Lines 69-76
```
this->subscribe(
      "esphome/discover", [this](const std::string &topic, const std::string &payload) { this->send_device_info_(); },
      2);

  std::string topic = "esphome/ping/";
  topic.append(App.get_name());
  this->subscribe(
      topic, [this](const std::string &topic, const std::string &payload) { this->send_device_info_(); }, 2);
```

luka6000, 2023-07-22:
confirmed, same here. Can you please make esphome respect _discovery: false_ and/or _topic_prefix: username/feeds_?

luka6000, 2023-07-22:
@sprouse I've made a fix for this but adafruit IO won't work anyway. They have very strict rules for feeds names and esphome is using a lot more complex hierarchy which doesn't fit into _username/feeds/something_

amanesamanes, 2023-08-19:
Thanks for opening the topic.
Same issue here. Esphome MQTT on a 8266 used to publish ok on Adafruit IO server till the HA/Esphome updates of last June, I had to recompile deactivating MQTT by then. Tried again it after every HA & Esphome update, no success. Last attempt today 18Aug2023 after installing HA core 2023.8.2 and Esphome 2023.8.1. No idea if HA or Esphome are to blame for the bug.
On the Adafruit side the error logger shows:
2023/08/18 07:04PM -- rejected, not a valid topic format
2023/08/18 07:04PM -- (8266 board id) 190.xx.xxx.xxx PUBLISH to topic `esphome/discover/(8266 board id)` rejected, in {username}/{type}/{key} topics for shared resources, {type} must be "feeds"_

luka6000, 2023-08-21:
> Thanks for opening the topic. Same issue here. Esphome MQTT on a 8266 used to publish ok on Adafruit IO server till the HA/Esphome updates of last June, I had to recompile deactivating MQTT by then. 

You can try my fix like this
```
external_components:
  - source: github://luka6000/esphome@devb
    components: [ mqtt ]
    refresh: 0s
mqtt:
  broker: ${io_broker}
  username: ${io_username}
  password: ${io_key}
  discovery: false
  birth_message:
  will_message:
  log_topic:
```
It basically disables discovery topics pub-sub

steven-sprouse-wdc, 2023-08-21:
@luka6000 what is the prognosis for your PR making it in to a release?

luka6000, 2023-08-21:
> @luka6000 what is the prognosis for your PR making it in to a release?

No idea. It depends on Jesse.
But you can easily just delete external_components section whenever you want. You can fork this fix and use your own repo if that sounds more secure for you.

amanesamanes, 2023-08-23:
many thanks @luka6000, in a couple of days I will try your -temporary, I hope- fix and let you know

pavelw, 2023-08-26:
for me it is kind of security issue. every time esphome device boots up it sends to broker:
(some entries edited)
esphome/discover/myDeviceName {"ip":"19.16.1.123","name":"myDeviceName ","port":6053,"version":"2023.8.1","mac":"123332123132","platform":"ESP8666","board":"myBoard","network":"wifi"}

so it sends ip, exact esphome version, mac and chip

luka6000, 2023-09-06:
Fixed in 2023.8.3

> for me it is kind of security issue. every time esphome device boots up it sends to broker: (some entries edited) esphome/discover/myDeviceName {"ip":"19.16.1.123","name":"myDeviceName ","port":6053,"version":"2023.8.1","mac":"[123332123132](tel:123332123132)","platform":"ESP8666","board":"myBoard","network":"wifi"}
> 
> so it sends ip, exact esphome version, mac and chip

That's the whole point of the discovery process, isn't it? You turn on discovery and your chips want to be discovered with all the details.

pauly7300, 2025-07-24:
This is happening to my new install of HA and an esp32 trying to connect to adafruit io.  Same error messages as above.