esphome/issues #4704: Pin Schema INPUT_PULLUP not the same as mode:/input: true/pullup: true ?

jov58, 2023-07-16:
### The problem

Using ESPHome on a Wemos S3, I was trying to define a button connected between GPIO21 and GND like this:
```
binary_sensor:

  - platform: gpio
    name: "Blue Button"
    pin:
      number: GPIO21
      inverted: true
      mode:
        input: true
        pullup: true
```
This resulted in the binary sensor going from ON to OFF several times per second, flooding the log. After trying different buttons and GPIOs, I changed the ```mode:``` definition to:
```
      mode: INPUT_PULLUP
```
According to the [documentation](https://esphome.io/guides/configuration-types.html#config-pin-schema), this should have the same effect but to my surprise the button logic now worked perfectly.

Is the documenation wrong and is there a difference between the two definitions or is this a bug in ESPHome?

### Which version of ESPHome has the issue?

2023.6.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP32

### Board

Wemos Lolin S3

### Component causing the issue

binary_sensor/gpio

### Example YAML snippet

```yaml
esphome:
  name: controlbox-1
  platformio_options: 
    board_build.flash_mode: dio
    board_build.extra_flags:
      - "-DARDUINO_USB_CDC_ON_BOOT=0"

esp32:
  board: lolin_s3
  framework:
    type: arduino
```

ssieb, 2023-07-16:
That should be exactly the same.

jov58, 2023-07-16:
OK, thx, must be something else then. I'll keep searching.

closed 2023-07-16 by jov58