esphome/issues #4706: use_address may no longer be a DNS name

klaernie, 2023-07-17:
### The problem

After esphome/esphome#5096, `use_address` is only being used, if the value is an IPv4 address. If it is not, MQTT discovery is used.

In my configs, I disable MDNS to avoid the constant chatter, and instead rely on DNS being maintained by my DHCP server.
This means that the value in `use_address` is the hostname, that can be reliably looked up in DNS. When the IP needs to change this can be done centrally and does not requires changes in the esphome configs, and similarly, slow migrations to new IPs can be done by changing the static lease in DHCP.

IMHO the best way would be, that as long as use_address has a meaningful value, it is used preferably - even if it is wrong. This way the user can fix the incorrect address in `use_address` instead of masking the problem by performing a discovery.

### Which version of ESPHome has the issue?

2023.8.0-dev20230717

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Component causing the issue

mDNS, MQTT, Ethernet, WIFI

### Example YAML snippet

```yaml
# Full yaml at https://github.com/klaernie/esphome-configs/blob/main/tuya3.yaml

wifi:
  ssid: !secret wifi-east
  password: !secret wifi-pwd

  use_address: tuya3.ak-online.be

mdns:
  disabled: true
```

### Anything in the logs that might be useful for us?

```txt
kandre@mainframe(pts/3)[main●!] ~/Projekte/ESPHome % ./esphome run tuya3.yaml
INFO ESPHome 2023.8.0-dev
INFO Reading configuration tuya3.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing tuya3 (board: esp_wroom_02; framework: arduino; platform: platformio/espressif8266@3.2.0)
----------------------------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 2MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- AsyncMqttClient-esphome @ 0.8.6
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/tuya3/src/main.cpp.o
Linking .pioenvs/tuya3/firmware.elf
RAM:   [====      ]  41.3% (used 33840 bytes from 81920 bytes)
Flash: [====      ]  42.9% (used 447865 bytes from 1044464 bytes)
Building .pioenvs/tuya3/firmware.bin
esp8266_copy_factory_bin([".pioenvs/tuya3/firmware.bin"], [".pioenvs/tuya3/firmware.elf"])
====================================================================================================
INFO Successfully compiled program.
INFO Starting looking for IP in topic esphome/discover/tuya3
INFO Connected to MQTT broker!
INFO Send discover via MQTT broker topic: esphome/ping/tuya3
ERROR Failed to find IP via MQTT
kandre@mainframe(pts/3)[main●!] ~/Projekte/ESPHome %
```

jesserockz, 2023-07-19:
The hard part of this is that `use_address` is set to a default value of `<name>.local` so it always has a "meaningful" value which is hard to know if it was manually set by the user (apart from checking if it was an ip address) without adding another flag or something somewhere. 

I will look into this though.

klaernie, 2023-07-20:
Would it be a possibility to try if the `use_address` is an IP address, and if not, if it is resolvable to an IP? If it is not an IP directly, it would be resolvable either by DNS or MDNS. And if that fails, discovery via MQTT seems like a fair attempt.

RoboMagus, 2023-10-09:
I'd propose giving `use_address` priority whenever it's present. And only use MQTT resolution when it's not there. 
This would be most in line with the situation we had before MQTT resolution was added; where `use_address` would basically be an override with top priority.

Initially I thought of falling back to MQTT resolution if `use_address` would not resolve, but then the whole thing becomes too hit-and-miss. I.e. more debug work to find out why something failed, instead of just `the address you manually specified does not respond`.

MoHf7f4, 2023-11-22:
Is there a work around for these issues with mqtt discovery?  Any documentation on what we should set in mqtt for discovery to work?
Even if I have an IP in use_address it gets stuck trying to mqtt discover.

klaernie, 2023-11-23:
@MoHf7f4  You can run `esphome compile myfile.yaml`, grab the compiled firmware.bin file, and upload it via the webserver (if your old version had it enabled). That's quite tedious, but the solution I now resorted to.

MoHf7f4, 2023-11-23:
All these mqtt errors aren't really happening on the esp right?  They are
all because how someone is trying to have esphome server try to contact the
esp correct?

Here's what I did to get my workflow sort of normalized again. Thanks to
Andre.
A firmware gets installed using esphome, esphome tries to access it via
mqtt but that fails, using the webui or api I reboot of the device, this
unsticks the stupid esphome server trying to do something nobody can
explain with mqtt ping, everything works like normal.

AJolly, 2024-11-29:
So theres no way to use a dns address instead?

Mantrid42, 2025-06-28:
I just ran across this annoyance too. I have DHCP and DNS servers that I want to use to resolve the address, but it seems that ESPHome really wants to use mDNS.
https://esphome.io/guides/faq.html#notes-on-disabling-mdns

I ran into mDNS issues when I tried to rename a device. The device thought it was one thing, and Device Builder thought it was something else. I ended up having to temporarily set manual_ip in the wifi config. It would be nice if I could just give it the proper DNS hostname to find the device.

ssieb, 2025-06-28:
@Mantrid42 you're using mqtt?

Mantrid42, 2025-06-29:
It didn't work for me. Even putting the IP address there, it kept going for the mDNS .local address, and device builder said the device with offline.