esphome/issues #4711: Unable to compile ADC for esp32-c3

fran6120, 2023-07-19:
### The problem

Since esphome version 2023.7.0 I get this error when compiling the ADC sensor.
I have tried cleaning the build files but the problem persists.

```
INFO ESPHome 2023.7.0
INFO Reading configuration /config/adc-esp32c3.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing adc-esp32c3 (board: esp32-c3-devkitm-1; framework: arduino; platform: platformio/espressif32@5.3.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/AsyncTCP-esphome @ 1.2.2
INFO Installing esphome/AsyncTCP-esphome @ 1.2.2
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@1.2.2 has been installed!
INFO AsyncTCP-esphome@1.2.2 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
Compiling .pioenvs/adc-esp32c3/src/esphome/components/adc/adc_sensor.cpp.o
Compiling .pioenvs/adc-esp32c3/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/adc-esp32c3/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/adc-esp32c3/src/esphome/components/api/api_pb2.cpp.o
src/esphome/components/adc/adc_sensor.cpp: In member function 'virtual float esphome::adc::ADCSensor::sample()':
src/esphome/components/adc/adc_sensor.cpp:164:55: error: invalid conversion from 'int32_t*' {aka 'long int*'} to 'int*' [-fpermissive]
       adc2_get_raw(channel2_, ADC_WIDTH_MAX_SOC_BITS, &raw);
                                                       ^~~~
In file included from src/esphome/components/adc/adc_sensor.h:10,
                 from src/esphome/components/adc/adc_sensor.cpp:1:
/config/.esphome/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/driver/include/driver/adc.h:381:81: note:   initializing argument 3 of 'esp_err_t adc2_get_raw(adc2_channel_t, adc_bits_width_t, int*)'
 esp_err_t adc2_get_raw(adc2_channel_t channel, adc_bits_width_t width_bit, int *raw_out);
                                                                            ~~~~~^~~~~~~
src/esphome/components/adc/adc_sensor.cpp:196:53: error: invalid conversion from 'int32_t*' {aka 'long int*'} to 'int*' [-fpermissive]
     adc2_get_raw(channel2_, ADC_WIDTH_MAX_SOC_BITS, &raw11);
                                                     ^~~~~~
In file included from src/esphome/components/adc/adc_sensor.h:10,
                 from src/esphome/components/adc/adc_sensor.cpp:1:
/config/.esphome/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/driver/include/driver/adc.h:381:81: note:   initializing argument 3 of 'esp_err_t adc2_get_raw(adc2_channel_t, adc_bits_width_t, int*)'
 esp_err_t adc2_get_raw(adc2_channel_t channel, adc_bits_width_t width_bit, int *raw_out);
                                                                            ~~~~~^~~~~~~
src/esphome/components/adc/adc_sensor.cpp:199:55: error: invalid conversion from 'int32_t*' {aka 'long int*'} to 'int*' [-fpermissive]
       adc2_get_raw(channel2_, ADC_WIDTH_MAX_SOC_BITS, &raw6);
                                                       ^~~~~
In file included from src/esphome/components/adc/adc_sensor.h:10,
                 from src/esphome/components/adc/adc_sensor.cpp:1:
/config/.esphome/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/driver/include/driver/adc.h:381:81: note:   initializing argument 3 of 'esp_err_t adc2_get_raw(adc2_channel_t, adc_bits_width_t, int*)'
 esp_err_t adc2_get_raw(adc2_channel_t channel, adc_bits_width_t width_bit, int *raw_out);
                                                                            ~~~~~^~~~~~~
src/esphome/components/adc/adc_sensor.cpp:202:57: error: invalid conversion from 'int32_t*' {aka 'long int*'} to 'int*' [-fpermissive]
         adc2_get_raw(channel2_, ADC_WIDTH_MAX_SOC_BITS, &raw2);
                                                         ^~~~~
In file included from src/esphome/components/adc/adc_sensor.h:10,
                 from src/esphome/components/adc/adc_sensor.cpp:1:
/config/.esphome/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/driver/include/driver/adc.h:381:81: note:   initializing argument 3 of 'esp_err_t adc2_get_raw(adc2_channel_t, adc_bits_width_t, int*)'
 esp_err_t adc2_get_raw(adc2_channel_t channel, adc_bits_width_t width_bit, int *raw_out);
                                                                            ~~~~~^~~~~~~
src/esphome/components/adc/adc_sensor.cpp:205:59: error: invalid conversion from 'int32_t*' {aka 'long int*'} to 'int*' [-fpermissive]
           adc2_get_raw(channel2_, ADC_WIDTH_MAX_SOC_BITS, &raw0);
                                                           ^~~~~
In file included from src/esphome/components/adc/adc_sensor.h:10,
                 from src/esphome/components/adc/adc_sensor.cpp:1:
/config/.esphome/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/driver/include/driver/adc.h:381:81: note:   initializing argument 3 of 'esp_err_t adc2_get_raw(adc2_channel_t, adc_bits_width_t, int*)'
 esp_err_t adc2_get_raw(adc2_channel_t channel, adc_bits_width_t width_bit, int *raw_out);
                                                                            ~~~~~^~~~~~~
*** [.pioenvs/adc-esp32c3/src/esphome/components/adc/adc_sensor.cpp.o] Error 1
========================== [FAILED] Took 6.42 seconds ==========================
```

### Which version of ESPHome has the issue?

2023.7.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.7.2

### What platform are you using?

ESP32

### Board

ESP32C3 (Lolin C3 Mini)

### Component causing the issue

ADC

### Example YAML snippet

```yaml
esphome:
  name: adc-esp32c3
  friendly_name: adc-esp32c3

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "***************************************"

ota:
  password: "*************************************"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Adc-Esp32C3 Fallback Hotspot"
    password: "**********"

captive_portal:

sensor:
  - platform: adc
    pin: GPIO4
    name: "Voltaje módulo Lluvia"
    id: "lluvia_analogica"
    attenuation: auto
    unit_of_measurement: "V"
    update_interval: 60s
```

nsty, 2023-07-19:
I got the exact same error message and stack trace with the adc_sensor with the same chip, same pin. Not sure if it's just this combination.

clomads, 2023-07-20:
I am also having this issue with the C3 - removing ADC config allows compile to proceed.

darekxan, 2023-07-20:
> I got the exact same error message and stack trace with the adc_sensor with the same chip, same pin. Not sure if it's just this combination.

GPIO2 doesn't work too

calle66, 2023-07-20:
beim esp32-c3-devkitm-1 besteht das gleiche Problem, diverse Pin-Deklarationen: 0, 1, 2, 3, 4, GPIO0 - GPIO4, ...
auch ich erhalte immer o.g. Fehlermeldung

Idcsl, 2023-07-22:
Exact same problem here and error message for me too. I'm using pin 4 for the ADC.

ssieb, 2023-07-22:
I can verify this.  There's something weird going on with the compiler.

kulichs, 2023-07-22:
Also same issue with pin 0 and 3 compilation

ewpa, 2023-07-22:
Possibly introduced by PR#4381, commit 3ac0165.

ssieb, 2023-07-22:
It will happen with any pin on a C3 and yes, it's caused by that commit.  Apparently we don't have a test config for a C3 with adc.

nsty, 2023-07-25:
For reference: https://github.com/esphome/esphome/pull/4381/
I guess the problem is mainly the switch from int to int32_t variable type and not the actual contribution of the PR.

Looks like a more substantial Espressif/Arduino problem when I look at numerous threads claiming the same issue.
https://github.com/espressif/esp-idf/issues/6906

A possible workaround is introduced here.
https://github.com/arendst/Tasmota/pull/11807/

kbx81, 2023-07-26:
TIL the RISC-V compiler doesn't use 32-bit `int`s. Cool.
Tested with my C3 board and https://github.com/esphome/esphome/pull/5151 corrects this issue.
Sorry for the trouble.

kbx81, 2023-07-26:
If anyone wants to test the change to confirm it's fixed for them:

```yaml
external_components:
  - source: github://pr#5151
    components: [adc]
```

louisvangeldrop, 2023-07-26:
I have tested it and it works now. Thx for the help

zwgmy, 2023-07-27:
> If anyone wants to test the change to confirm it's fixed for them:
> 
> ```yaml
> external_components:
>   - source: github://pr#5151
>     components: [adc]
> ```

Does it need to add this to  yaml for compile C3 in the future to compile?

kbx81, 2023-07-27:
> Does it need to add this to yaml for compile C3 in the future to compile?

@zwgmy we'll merge this PR real soon after which time this extra code will no longer be necessary.

closed 2023-07-30 by jesserockz