esphome/issues #4712: ili9xxx component - causing continuous log warnings

Zoogara, 2023-07-19:
### The problem

After upgrading to 2023.7.0 - device using ili9xxx component prints warning in log: 

Component ili9xxx.display took a long time for an operation (0.50 s).
Components should block for at most 20-30ms.

This starts as soon as the component is initialised and continues as long as the device is up, log example below

No apparent negative impact except of course the resources this continual logging is using.

### Which version of ESPHome has the issue?

2023.7.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

ESP32-DEVKIT-C WROVER 4MB PSRAM

### Component causing the issue

ili9xxx

### Example YAML snippet

```yaml
display:
  - platform: ili9xxx
    model: ili9488
    id: my_display
    rotation: 180
    dc_pin: GPIO02
    cs_pin: GPIO15
    reset_pin: GPIO04
    
    # Page 1 - main display.  Pages 2-5 are for anti burn-in
    pages:
      - id: page1
        lambda: |-
          // Colours - used for visiual indication of wind strength
          auto red = Color(255, 0, 0);
          auto green = Color(0, 255, 0);
          auto light_blue = Color(135, 237, 232);
          auto orange = Color(255, 170, 43);
          auto white = Color(255, 255, 255);
          auto purple = Color(97, 15, 219);
          auto grey = Color(100, 135, 135);
          // history arrays for trend display
          static float hist_wind[30];
          static int hist_dir[30];
          static int index=0;
          static int prev_index=29;
          // Convert wind direction to int, speed to float
          int dir = atoi(id(wind_dir).state.c_str());
          float speed = atof(id(wind_speed).state.c_str());
          // Calculate xy position to plot wind indicator
          int x = 120 + (90 * (cos((dir-90)*PI/180)));
          int y = 150 + (90 * (sin((dir-90)*PI/180)));
          // Store in array
          if ((hist_wind[prev_index]!=speed || hist_dir[prev_index]!=dir) && speed!=0) {
            hist_wind[index] = speed;
            hist_dir[index] = dir;
[... 429 lines cut ...]
...
...
...
[19:54:55][W][component:204]: Component ili9xxx.display took a long time for an operation (0.52 s).
[19:54:55][W][component:205]: Components should block for at most 20-30ms.
[19:54:56][W][component:204]: Component ili9xxx.display took a long time for an operation (0.52 s).
[19:54:56][W][component:205]: Components should block for at most 20-30ms.
[19:54:57][W][component:204]: Component ili9xxx.display took a long time for an operation (0.52 s).
[19:54:57][W][component:205]: Components should block for at most 20-30ms.
[19:54:58][W][component:204]: Component ili9xxx.display took a long time for an operation (0.52 s).
[19:54:58][W][component:205]: Components should block for at most 20-30ms.
[19:54:59][W][component:204]: Component ili9xxx.display took a long time for an operation (0.52 s).
[19:54:59][W][component:205]: Components should block for at most 20-30ms.
[19:55:00][W][component:204]: Component ili9xxx.display took a long time for an operation (0.52 s).
[19:55:00][W][component:205]: Components should block for at most 20-30ms.
[19:55:01][W][component:204]: Component ili9xxx.display took a long time for an operation (0.52 s).
[19:55:01][W][component:205]: Components should block for at most 20-30ms.
[19:55:02][W][component:204]: Component ili9xxx.display took a long time for an operation (0.52 s).
[19:55:02][W][component:205]: Components should block for at most 20-30ms.
[19:55:03][W][component:204]: Component ili9xxx.display took a long time for an operation (0.52 s).
[19:55:03][W][component:205]: Components should block for at most 20-30ms.
[19:55:04][W][component:204]: Component ili9xxx.display took a long time for an operation (0.52 s).
[19:55:04][W][component:205]: Components should block for at most 20-30ms.
[19:55:05][W][component:204]: Component ili9xxx.display took a long time for an operation (0.52 s).
```

nielsnl68, 2023-07-19:
Hi @Zoogara, this noting new, this message was always shown, when you used verbose logging.

Zoogara, 2023-07-19:
Hi @nielsnl68 - thanks for updating this component btw.  Note my logging level hasn't changed and is the default.  Has the default log level changed to Verbose this release?

nielsnl68, 2023-07-19:
No, Jessy changed the log level of this message from verbose to a warning.

Zoogara, 2023-07-19:
Ah - guess that's enough info if anyone finds this issue.  You may as well close it, I will turn logging down to ERROR.

nielsnl68, 2023-07-19:
On all larger displays will this be the case.

jesserockz, 2023-07-19:
https://github.com/esphome/issues/issues/4717

closed 2023-07-19 by jesserockz

billmaterial, 2023-07-20:
I have found a very simple workaround for my ssd1306 128x64 display, I simply increased the I2C frequency to 400kHz and now I do not get this warning.

Ljudet, 2023-10-02:
> I have found a very simple workaround for my ssd1306 128x64 display, I simply increased the I2C frequency to 400kHz and now I do not get this warning.

How do you change the I2C frequency? tried to google it but i cant find how you do

ssieb, 2023-10-02:
> > I have found a very simple workaround for my ssd1306 128x64 display, I simply increased the I2C frequency to 400kHz and now I do not get this warning.
> 
> How do you change the I2C frequency? tried to google it but i cant find how you do

Did you look at the i2c component docs?

Ljudet, 2023-10-02:
> > > I have found a very simple workaround for my ssd1306 128x64 display, I simply increased the I2C frequency to 400kHz and now I do not get this warning.
> > 
> > 
> > How do you change the I2C frequency? tried to google it but i cant find how you do
> 
> Did you look at the i2c component docs?

Yeah i think i found how to do it. Uploading the new code right now ;)

KuhnDaniel, 2023-10-21:
# Example configuration entry for ESP32
i2c:
  sda: 21
  scl: 22
  scan: true
  id: bus_a

https://esphome.io/components/i2c#i2c

Configuration variables:

    sda (Optional, [Pin](https://esphome.io/guides/configuration-types#config-pin)): The pin for the data line of the I²C bus. Defaults to the default of your board (usually GPIO21 for ESP32 and GPIO4 for ESP8266).

    scl (Optional, [Pin](https://esphome.io/guides/configuration-types#config-pin)): The pin for the clock line of the I²C bus. Defaults to the default of your board (usually GPIO22 for ESP32 and GPIO5 for ESP8266).

    scan (Optional, boolean): If ESPHome should do a search of the I²C address space on startup. Defaults to true.

    frequency (Optional, float): Set the frequency the I²C bus should operate on. Defaults to 50kHz. Values are 10kHz, 50kHz, 100kHz, 200kHz, … 800kHz

    id (Optional, [ID](https://esphome.io/guides/configuration-types#config-id)): Manually specify the ID for this I²C bus if you need multiple I²C buses.