esphome/issues #4713: ESPHome fails to connect to hidden SSIDs

Ralms, 2023-07-19:
### The problem

Hi there, 

I've recently acquired a couple NSPanels, to use with HomeAssistant, that need to be flashes with ESPHome. 

Everything was working fine with flashing, it would recognize the device, it would flash, give the congratulations message every time (using the Web portal to flash).

But than it wouldn't connect to the Wifi.

I've tried: 
- Using 2 different TTL to USB adapters, one of them the DSD Tech which has a genuine FTDI chip.
- Removing special characters from the Wifi Password.
- Writing the password directly on the YAML instead of using Secrets.
- Having a just letters SSID and no dashes or anything. E.g.: IOTWifi

Then, in a desperation, I decided to have it connect to my Guest Wifi and connected straight away.

My Guest Wifi is visible, however I use an smart devices dedicated Wifi called "IOTWifi", which is hidden as its only intended for smart devices. 

As such, its clear that ESPHome is not connecting or attempting to connect if the SSID is not visible.

### Which version of ESPHome has the issue?

2023.6.5

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

Sonoff NSPanel EU

### Example YAML snippet

```yaml
substitutions:

###### CHANGE ME START ######

  device_name: "YOUR NSPANEL_NAME" 
  wifi_ssid: "YOUR WIFI SSID"
  wifi_password: "YOUR WIFI PASSWORD"

  nextion_update_url: "http://HOME-ASSISTANT-IP:8123/local/nspanel_us.tft" # URL to local tft File
#  nextion_update_url: "https://raw.githubusercontent.com/Blackymas/NSPanel_HA_Blueprint/main/nspanel_us.tft" # URL to Github

##### CHANGE ME END #####

##### DO NOT CHANGE ANYTHING! #####

packages:
  ##### download esphome code from Github
  remote_package:
    url: https://github.com/Blackymas/NSPanel_HA_Blueprint
    ref: main
    files: [nspanel_esphome.yaml]
    refresh: 300s

##### DO NOT CHANGE ANYTHING! #####
```

### Additional information

Project in question where the Yaml was retrieved: 
https://github.com/Blackymas/NSPanel_HA_Blueprint/wiki/(EN)-First-Steps---Installation-and-Setup

Ralms, 2023-07-19:
Update: 
It turns out that ESPHome does support connecting to Hidden to SSIDs, but for some reason you need to enable the "fast_connect: true". 

Wouldn't it make more sense for this to be the default behavior and if it fails to connect to do the Wifi Scan?

ssieb, 2023-07-20:
The docs clearly explain that you need that option for hidden SSIDs and also the drawbacks of using it.

closed 2023-07-20 by ssieb