esphome/issues #4714: Component ssd1306_base took a long time for an operation

jov58, 2023-07-19:
### The problem

Since the update to ESPHome 2023.7 I see this log message every loop for a SH1106 128*64 display:
```
[W][component:204]: Component ssd1306_base took a long time for an operation (0.28 s).
[W][component:205]: Components should block for at most 20-30ms.
```
The first message appears rather quick, even before the i2c messages (see code snippet below. With the previous version of ESPHome I didn't see this message and the code was working fine. And still is as far as I can judge for now.

But the messages are flooding the log and I don't know how to proceed from here. Can I ignore and suppress this warning? Or should I look into the code to see if there is something specific taking a long time (nothing special though)? Or, since the code seems to work fine, is the system sending a superfluous message?

Any advice appreciated!

### Which version of ESPHome has the issue?

2023.7

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP32

### Board

Wemos Lolin S3

### Component causing the issue

ssd1306_base

### Anything in the logs that might be useful for us?

```txt
[18:18:11][I][app:102]: ESPHome version 2023.7.0 compiled on Jul 19 2023, 18:17:12
[18:18:11][C][wifi:543]: WiFi:
[18:18:11][C][wifi:394]:   <snip>
[18:18:11][C][logger:301]: Logger:
[18:18:11][C][logger:302]:   Level: DEBUG
[18:18:11][C][logger:303]:   Log Baud Rate: 115200
[18:18:11][C][logger:305]:   Hardware UART: UART0
[18:18:12][W][component:204]: Component ssd1306_base took a long time for an operation (0.28 s).
[18:18:12][W][component:205]: Components should block for at most 20-30ms.
[18:18:12][C][i2c.arduino:053]: I2C Bus:
[18:18:12][C][i2c.arduino:054]:   SDA Pin: GPIO42
[18:18:12][C][i2c.arduino:055]:   SCL Pin: GPIO41
[18:18:12][C][i2c.arduino:056]:   Frequency: 50000 Hz
[18:18:12][C][i2c.arduino:059]:   Recovery: bus successfully recovered
[18:18:12][I][i2c.arduino:069]: Results from i2c bus scan:
[18:18:12][I][i2c.arduino:075]: Found i2c device at address 0x3C
```

jesserockz, 2023-07-19:
Please comment on https://github.com/esphome/issues/issues/4717

closed 2023-07-19 by jesserockz

vasko775, 2023-07-24:
for the i2c component solved my problem.

i2c:
  sda: 4
  scl: 15
  frequency: 800kHz

Lefuneste83, 2023-08-02:
Thanks for the type this worked for me as well

DVSelectronics, 2023-11-03:
setting i2c to 800kHz worked  thanks man.

Grey-Lancaster, 2023-11-25:
+1 
setting i2c to 800kHz worked

jdituicha, 2023-12-04:
Question: It is not a very risky move to raise the frequency to those levels, the default frequency in ESPHome being 50Khz. That will demand a lot of work from the processor as a result:
- higher energy consumption
- higher working temperature
- more instability on the bus, since it is already a technology (i2c) susceptible to interference.
I have a lot of displays with this warning in the logs. But the truth is that this solution worries me even more.

apeeters, 2023-12-09:
Also does not work if other components on the i2c bus do not support the higher frequencies.

nygellyndley, 2023-12-20:
It seems this message always existed, it's just the default log level became more verbose : https://github.com/esphome/issues/issues/4717

p1ngb4ck, 2024-04-17:
pushing the frequency of the i2c bus may be a solution for some - but as soon as you have multiple i2c devices, you realize not all of them are capable to do 800kHz.
SSD1306 "fast" mode is considered 100kHZ from its datasheet, and the esphome default is 50kHz only.
A general "all around" value for i2c frequency is 100kHz.

Everybody is addressing the actual issue from the wrong end. This message should NOT be there. Its a log clutter.
If you want to achieve, what that message actually wanted to achieve the correct way, you gotta take the hard route and apply known "normal" working delays/timings for all components and then throw errors/warning only, if expected values are not met by a specific amount.
You never ever ever ever should throw warnings on components, that work as expected, it sends users down an unnecessary debugging route. Best practices rule : Never ever throw errors/warnings for wanted/expected behaviour. Throw them for unwanted/unexpected ONLY. One of those golden rules of coding, that are forgotten too often unfortunately