esphome/issues #4715: 2023.7.0 broke a ventilation component I use

spudje, 2023-07-19:
### The problem

Since updating from 2023.6.5 to 2023.7.0 a custom component I got from here https://gathering.tweakers.net/forum/list_messages/1728169/0 (and tweaked a bit) stopped working. I didn't change anything on the ventilation custom component code and verified that it is 100% caused by upgrading to 2023.7.0, since I went hence and forth between versions a couple of times and it was always only on 2023.6.5 it worked.

This uses an ESP32 combined with an nrf905 RF antenna.

Since the upgrade I only get timeouts. Even when I increase multiple timeout values in the config.

Comparing the ESPhome 2023.6.5 & 2023.7.0 code by hand I think the most probable cause is either:
- scheduler where some timeout changes have been made (**timeout=%" PRIu32 "**) in the new 2023.7.0 version
- SPI where **DATA_RATE_80MHZ** was added and maybe somehow this made the setup run at a different speed?

### Which version of ESPHome has the issue?

2023.7.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.7.2

### What platform are you using?

ESP32

### Board

esp32doit-devkit-v1

### Component causing the issue

Not sure but guessing SPI or Scheduler

### Example YAML snippet

```yaml
substitutions:
  devicename: ventilatie
  upper_devicename: Ventilatie

esphome:
  name: ventilatie
  comment: Fan controller

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: arduino

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  # password: !secret esphome_ventilatie_api_password
  encryption:
    key: !secret esphome_ventilatie_api_encryption_key

  services:
    - service: set_speed
      variables:
        run_speed: int
        run_time: int
      then:
        - lambda: |-
            zehnder_fan->setSpeed(run_speed, run_time);  
[... 733 lines cut ...]
[16:37:15][D][zehnder:666]: Receive timeout
[16:37:15][D][zehnder:670]: No data received, retry again (left: 0)
[16:37:15][V][zehnder:653]: In rfHandler --> RfStateWaitAirwayFree airway not busy
[16:37:15][D][zehnder:654]: Start TX
[16:37:15][V][nRF905:528]: Config:
  Channel 117 Band 868 MHz -> 868200000
  Rx Power normal
  Tx Retransmit Off
  Rx Address (4) 0xC11852FC
  Rx Payload width 16
  Tx Address (4)
  Tx Payload width 16
  Clk Out 500000
  XTAL Freq 16000000
  CRC On -> 16
  TX Power 10 dBm
[16:37:15][V][nRF905:227]: Write config data: 0x75 0x0E 0x44 0x10 0x10 0xFC 0x52 0x18 0xC1 0xDB
[16:37:15][V][nRF905:246]: Write config OK
[16:37:15][V][nRF905:098]: State change: 0x00 -> 0x20
[16:37:15][D][zehnder:136]: Tx Ready
[16:37:15][V][nRF905:098]: State change: 0x20 -> 0x00
[16:37:16][D][zehnder:666]: Receive timeout
[16:37:16][D][zehnder:677]: No messages received, giving up now...
[16:37:16][W][zehnder:506]: Query Timeout
```

ssieb, 2023-07-20:
External components not working is not an esphome issue unless you can reduce it to a bug in esphome code.  We can't even see what your code is.

spudje, 2023-07-20:
Nothing changed in the external component. It stopped working as soon as I upgraded to 2023.7.0.

ssieb, 2023-07-20:
It has nothing to do with those.  That message was just changed to a different logging level so that it's visible now. https://github.com/esphome/issues/issues/4717
Regardless of whether anything changed in the external component, things might have changed internally in esphome.  It's up to whoever created the component to determine if the problem is in the component or esphome.

spudje, 2023-10-26:
I'm still struggling with this (Now trying with 2023.10.3). Every ESPHome release after 2023.6 breaks this custom component. I have the idea it has something to do with performance and timeouts being too tight. Did 2023.7 introduce functionality that increases load/resources impact?

ssieb, 2023-10-26:
It is absolutely not related to that issue.
Have you talked to the person who made the component?

golles, 2024-08-30:
Is this the component you're using? https://github.com/Sanderhuisman/ESPHome-Zehnder-RF
(I've got the same problem, don't want to download from wetransfer...)