esphome/issues #4716: sps30 triggers component latency watchdog

nwf, 2023-07-19:
### The problem

I run a few `esphome` devices with `sps30` sensors attached.  As of 2023.7.0, I am told
```
[16:38:44][W][component:204]: Component sps30.sensor took a long time for an operation (0.12 s).
[16:38:44][W][component:205]: Components should block for at most 20-30ms.
```
after every set of sensor reports, such as:
```
[16:38:44][D][sensor:093]: 'Garage PM <1µm Weight concentration': Sending state 4.63284 µg/m³ with 2 decimals of accuracy
[16:38:44][D][sensor:093]: 'Garage PM <2.5µm Weight concentration': Sending state 4.78999 µg/m³ with 2 decimals of accuracy
[16:38:44][D][sensor:093]: 'Garage PM <4µm Weight concentration': Sending state 4.78999 µg/m³ with 2 decimals of accuracy
[16:38:44][D][sensor:093]: 'Garage PM <10µm Weight concentration': Sending state 4.78999 µg/m³ with 2 decimals of accuracy
[16:38:44][D][sensor:093]: 'Garage PM <0.5µm Number concentration': Sending state 32.95186 #/m³ with 2 decimals of accuracy
[16:38:44][D][sensor:093]: 'Garage PM <1µm Number concentration': Sending state 37.81256 #/m³ with 2 decimals of accuracy
[16:38:44][D][sensor:093]: 'Garage PM <2.5µm Number concentration': Sending state 37.92021 #/m³ with 2 decimals of accuracy
[16:38:44][D][sensor:093]: 'Garage PM <4µm Number concentration': Sending state 37.92162 #/m³ with 2 decimals of accuracy
[16:38:44][D][sensor:093]: 'Garage PM <10µm Number concentration': Sending state 37.92229 #/m³ with 2 decimals of accuracy
[16:38:44][D][sensor:093]: 'Garage Typical Particle size': Sending state 0.50724 µm with 0 decimals of accuracy
```

### Which version of ESPHome has the issue?

2023.7.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.7.2

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

sps30

### Example YAML snippet

```yaml
I am using the example `sps30` configuration with its many sensors, with the only changes being in names and `update_interval`:

  - platform: sps30
    pm_1_0:
      name: "Garage PM <1µm Weight concentration"
      id: "PM_1_0"
    pm_2_5:
      name: "Garage PM <2.5µm Weight concentration"
      id: "PM_2_5"
    pm_4_0:
      name: "Garage PM <4µm Weight concentration"
      id: "PM_4_0"
    pm_10_0:
      name: "Garage PM <10µm Weight concentration"
      id: "PM_10_0"
    pmc_0_5:
      name: "Garage PM <0.5µm Number concentration"
      id: "PMC_0_5"
    pmc_1_0:
      name: "Garage PM <1µm Number concentration"
      id: "PMC_1_0"
    pmc_2_5:
      name: "Garage PM <2.5µm Number concentration"
      id: "PMC_2_5"
    pmc_4_0:
      name: "Garage PM <4µm Number concentration"
      id: "PMC_4_0"
    pmc_10_0:
      name: "Garage PM <10µm Number concentration"
      id: "PMC_10_0"
    pm_size:
      name: "Garage Typical Particle size"
      id: "pm_size"
    address: 0x69
    update_interval: 60s
```
```

### Anything in the logs that might be useful for us?

```txt
As above.
```

jesserockz, 2023-07-19:
Please comment on https://github.com/esphome/issues/issues/4717

closed 2023-07-19 by jesserockz

Joshfindit, 2023-08-24:
@nwf I didn't see any other comments specifically about the SPS30 so I added my observations to that thread: https://github.com/esphome/issues/issues/4717#issuecomment-1692227072

I hope it helps you and others