esphome/issues #4717: Component xxxxxx took a long time for an operation

jesserockz, 2023-07-19:
### The problem

```
[00:00:00][W][component:204]: Component xxxxxx took a long time for an operation (x.xx s).
[00:00:00][W][component:205]: Components should block for at most 20-30ms.
```

These 2 log lines may show up in the most recent version of ESPHome due to the log level being changed from `verbose` to `warning`.
I made this change because changing the device log level to verbose just to see if these lines show up significantly slowed down
the device due to all the extra logging it had to do.

Do not report new issues related to this message, instead please comment on this issue with the logs that you see and the config for that component.

### Which version of ESPHome has the issue?

2023.7.0

### What type of installation are you using?

pip

### What platform are you using?

ESP32

nwf, 2023-07-19:
Sorry, didn't see this one before filing #4716.

Another one for the pile is `dallas`.

jesserockz, 2023-07-19:
@nwf thats alright as I posted this issue afterwards to stop future ones =)

Sebmaster, 2023-07-19:
`scd30` also causes this warning. This is the config I'm using:
```yaml
  - platform: scd30
    co2:
      name: "CO2"
      accuracy_decimals: 0
    temperature:
      name: "Temperature"
      accuracy_decimals: 1
    humidity:
      name: "Humidity"
      accuracy_decimals: 1
```

BenCos17, 2023-07-19:
Ssd1306 causes this warning also 
On mobile right now do can't paste them in like I normally would vs screenshots
[image]
[image]

BenCos17, 2023-07-19:
Ssd1306 causes it also

''display:
  - platform: ssd1306_i2c
    contrast: 0%
    model: "SH1106 128x64"
    reset_pin: D0
    address: 0x3C
    lambda: |-
      // Print "Bens' desk" in top center.
      it.printf(64, 0, id(font1), TextAlign::TOP_CENTER, "Bens' desk");

      // Print time in HH:MM format
      it.strftime(0, 60, id(font2), TextAlign::BASELINE_LEFT, "%H:%M", id(esptime).now());

      // Print inside temperature (from homeassistant sensor)
      if (id(bedroom_temp).has_state()) {
        it.printf(127, 23, id(font3), TextAlign::TOP_RIGHT , "%.1f°", id(bedroom_temp).state);
      }

      // Print bedroom co2 level (from homeassistant sensor)
      if (id(co2_bedroom).has_state()) {
        it.printf(127, 60, id(font3), TextAlign::BASELINE_RIGHT , "%.1f°", id(co2_bedroom).state);
      }
'''

ntwb, 2023-07-20:
Getting this on a BT proxy:

```log
[14:12:10][W][component:204]: Component esp32_ble_tracker took a long time for an operation (0.06 s).
[14:12:10][W][component:205]: Components should block for at most 20-30ms.
```

FYI: The config below is based on the original implementation, I've not updated it since the BT Proxy launch, so it may be out of date and I need to update it...

```yaml

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true

button:
- platform: safe_mode
  name: Safe Mode Boot
  entity_category: diagnostic

sensor:
  - platform: ble_rssi
    mac_address: xx:yy:zz:11:22:33
    name: "BT Proxy RSSI"
```

billmaterial, 2023-07-20:
A simple workaround for the ssd1306 128x64 display is to increase the i2c frequency to 400kHz. This fixed the warning for me. @BenCos17 

i2c:
  sda: GPIO21
  scl: GPIO22
  scan: True
  id: bus_a
  frequency: 400kHz

jov58, 2023-07-20:
I also got these messages:
```
[18:18:12][W][component:204]: Component ssd1306_base took a long time for an operation (0.28 s).
[18:18:12][W][component:205]: Components should block for at most 20-30ms.
```
In the default situation, a device updates every second so these messages appear every second, They clutter up the log very rapidly, making it a lot harder to follow other log messages.

I also wonder if it is useful to show these messages when the device is working properly and its functionality isn't slowed down. I guess a long-time-blocking display can be a problem in some scenarios but not in others. I don't know if it's possible with the architecture of ESPHome/ESP8266/ESP32, but wouldn't it be possible to shown these warnings only when functionality is effected, for instance when a certain heartbeat can't be maintained?

Or perhaps show the warnings once a minute?
```
[18:18:12][W][component:204]: Component ssd1306_base took a long time for an operation (0.28 s), showed up 60 times.
[18:18:12][W][component:205]: Components should block for at most 20-30ms.
```

> A simple workaround for the ssd1306 128x64 display is to increase the i2c frequency to 400kHz.

Didn't thought of that but it also worked on my SH1106 128*64, so thanks @billmaterial. At 200 kHz still warnings, at 400 kHz not anymore.

BenCos17, 2023-07-20:
> frequency: 400kHz

thanks

JanSchnacki, 2023-07-20:
I just upgraded to 2023.7.0 and recompiled my three NodeMCU stacks. After uploading I got this:

`
[15:05:53][C][htu21d:028]: HTU21D:
[15:05:53][C][htu21d:029]:   Address: 0x40
[15:05:53][C][htu21d:033]:   Update Interval: 60.0s
[15:05:53][C][htu21d:034]:   Temperature 'NodeMCUGaszaehler Temperature'
[15:05:53][C][htu21d:034]:     Device Class: 'temperature'
[15:05:53][C][htu21d:034]:     State Class: 'measurement'
[15:05:53][C][htu21d:034]:     Unit of Measurement: '°C'
[15:05:53][C][htu21d:034]:     Accuracy Decimals: 1
[15:05:53][C][htu21d:035]:   Humidity 'NodeMCUGaszaehler Humidity'
[15:05:53][C][htu21d:035]:     Device Class: 'humidity'
[15:05:53][C][htu21d:035]:     State Class: 'measurement'
[15:05:53][C][htu21d:035]:     Unit of Measurement: '%'
[15:05:53][C][htu21d:035]:     Accuracy Decimals: 1
[...]
[15:06:11][D][htu21d:065]: Got Temperature=22.2°C Humidity=67.1%
[15:06:11][W][component:204]: Component htu21d.sensor took a long time for an operation (0.11 s).
[15:06:11][W][component:205]: Components should block for at most 20-30ms.
`

The configuration is simply

`
# I2C Bus
i2c:
  sda: D2
  scl: D1
  scan: True
  id: bus_a

sensor:
  - platform: htu21d
    temperature:
      name: ${upper_devicename} Temperature
      unit_of_measurement: "°C"
      filters:
        - offset: -2.0
        - median:
    humidity:
      name: ${upper_devicename} Humidity
      unit_of_measurement: "%"
      filters:
        - median:
    update_interval: 60s
`

wtremmel, 2023-07-20:
I see this for my waveshare epaper display:

`[20:31:59][W][component:204]: Component waveshare_epaper.display took a long time for an operation (0.98 s).
`

Config:
```
display:
  - platform: waveshare_epaper
    id: epaper
    cs_pin: GPIO17
    dc_pin: GPIO16
    reset_pin: GPIO15
    busy_pin: GPIO18
    reset_duration: 2ms
    model: 7.50inv2
    update_interval: 60s
    rotation: 0
    lambda: |-

```

snechiporenko, 2023-07-20:
Look at my workaround [https://github.com/esphome/feature-requests/issues/2301](https://github.com/esphome/feature-requests/issues/2301)

tiwu, 2023-07-21:
I'm now seeing this reported for dsmr 0.8 as well:

```
[02:06:46][W][component:204]: Component dsmr took a long time for an operation (0.12 s).
[02:06:46][W][component:205]: Components should block for at most 20-30ms.
```
With the config as per https://github.com/mhendriks/esphome-p1/blob/main/p1-dongle-pro.yaml

brettmiller, 2023-07-21:
I'm seeing this:
```
[23:36:21][W][component:204]: Component esphome.coroutine took a long time for an operation (0.10 s).
[23:36:21][W][component:205]: Components should block for at most 20-30ms.
[23:36:23][W][component:204]: Component esphome.coroutine took a long time for an operation (0.10 s).
[23:36:23][W][component:205]: Components should block for at most 20-30ms.
```

When using the component:
```
external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump@develop

climate:
  - platform: mitsubishi_heatpump
    name: "${name}"
    id: ${device_id}_climate
    hardware_uart: UART1
    tx_pin: 12
    rx_pin: 13
    visual:
      temperature_step: 1.0
```

Sausages69, 2023-07-21:
> A simple workaround for the ssd1306 128x64 display is to increase the i2c frequency to 400kHz. This fixed the warning for me. @BenCos17
> 
> i2c: sda: GPIO21 scl: GPIO22 scan: True id: bus_a frequency: 400kHz

Thanks, (that below) fixed the error on the SH1106 here. Didn't fix the Dallas one though. 

i2c:   
  frequency: 400kHz

pery777, 2023-07-22:
I'm seeing this, but with pn532 as the component name:

[07:58:54][W][component:204]: Component pn532 took a long time for an operation (0.10 s).
[07:58:54][W][component:205]: Components should block for at most 20-30ms.
[07:58:55][W][component:204]: Component pn532 took a long time for an operation (0.10 s).
[07:58:55][W][component:205]: Components should block for at most 20-30ms.
[07:58:56][W][component:204]: Component pn532 took a long time for an operation (0.10 s).
[07:58:56][W][component:205]: Components should block for at most 20-30ms.

I am using the code:

https://github.com/adonno/tagreader/blob/master/tagreader.yaml

Rwin1234, 2023-07-22:
The dsmr component that reads out smart meters via a serial P1 connector shows this message every time the serial port receives data. My issue https://github.com/esphome/issues/issues/4731 was closed, but there could still be a dsmr component issue because the device frequently shows up in home assistant as available / unavailable via the Nmap Tracker integration. Taking a time-slice of 0.6 seconds is a lot and might interfere with the wifi / network stack.

INFO ESPHome 2023.7.0
INFO Reading configuration /config/esphome/p1-dongle-pro.yaml...
INFO Starting log output from p1-dongle-pro.local using esphome API
INFO Successfully connected to p1-dongle-pro.local
[00:25:44][I][app:102]: ESPHome version 2023.7.0 compiled on Jul 22 2023, 00:14:13
[00:25:44][I][app:104]: Project smartstuff.p1_dongle_pro version v23.02.1
[00:25:45][W][component:204]: Component dsmr took a long time for an operation (0.61 s).
[00:25:45][W][component:205]: Components should block for at most 20-30ms.
[00:25:55][W][component:204]: Component dsmr took a long time for an operation (0.60 s).
[00:25:55][W][component:205]: Components should block for at most 20-30ms.
[00:26:05][W][component:204]: Component dsmr took a long time for an operation (0.61 s).
[00:26:05][W][component:205]: Components should block for at most 20-30ms.

[repeats every 10 seconds]

If someone needs more info let me know.

andypnz, 2023-07-22:
With ssd1306 I got the same errors
```
[12:49:47][W][component:204]: Component interval took a long time for an operation (0.27 s).
[12:49:47][W][component:205]: Components should block for at most 20-30ms.
[12:49:47][W][component:204]: Component ssd1306_base took a long time for an operation (0.27 s).
[12:49:47][W][component:205]: Components should block for at most 20-30ms.
```
Fixed by adding
```
frequency: 400kHz
```

Sparenzi, 2023-07-22:
I have a ESP32 running an IR LED remote and a DHT11.
I get the error message only when the `climate `component is added to the code.

> [17:43:29][D][dht:048]: Got Temperature=28.3°C Humidity=54.0%
> [17:43:29][D][sensor:094]: 'Temperature': Sending state 28.30000 °C with 1 decimals of accuracy
> [17:43:29][D][climate:378]: 'AC' - Sending state:
> [17:43:29][D][climate:381]:   Mode: OFF
> [17:43:29][D][climate:386]:   Fan Mode: LOW
> [17:43:29][D][climate:398]:   Swing Mode: VERTICAL
> [17:43:29][D][climate:401]:   Current Temperature: 28.30°C
> [17:43:29][D][climate:407]:   Target Temperature: 25.00°C
> [17:43:29][D][sensor:094]: 'AC Remote Humidity': Sending state 54.00000 % with 0 decimals of accuracy
> [17:43:29][W][component:204]: Component dht.sensor took a long time for an operation (0.07 s).
> [17:43:29][W][component:205]: Components should block for at most 20-30ms.

ftw64, 2023-07-23:
I get this warning now with pzemac and modbus:

uart:
  rx_pin: GPIO4
  tx_pin: GPIO5
  baud_rate: 9600

modbus:

sensor:
  - platform: pzemac
    address: 1
    current:
      name: "${name}: Current"
    voltage:
      name: "${name}: Voltage"
    energy:
      name: "${name}: Energy"
    power:
      name: "${name}: Power"
    frequency:
      name: "${name}: Frequency"
    power_factor:
      name: "${name}: Power Factor"
    update_interval: 2s

[19:26:56][D][pzemac:049]: PZEM AC: V=230.5 V, I=0.045 A, P=5.8 W, E=10155.0 Wh, F=50.0 Hz, PF=0.56
[19:26:56][D][sensor:093]: 'PZEM-Meter-01: Voltage': Sending state 230.50000 V with 1 decimals of accuracy
[19:26:56][D][sensor:093]: 'PZEM-Meter-01: Current': Sending state 0.04500 A with 3 decimals of accuracy
[19:26:56][D][sensor:093]: 'PZEM-Meter-01: Power': Sending state 5.80000 W with 2 decimals of accuracy
[19:26:56][D][sensor:093]: 'PZEM-Meter-01: Energy': Sending state 10155.00000 Wh with 0 decimals of accuracy
[19:26:56][D][sensor:093]: 'PZEM-Meter-01: Frequency': Sending state 50.00000 Hz with 1 decimals of accuracy
[19:26:56][D][sensor:093]: 'PZEM-Meter-01: Power Factor': Sending state 0.56000  with 2 decimals of accuracy
[19:26:56][W][component:204]: Component modbus took a long time for an operation (0.07 s).
[19:26:56][W][component:205]: Components should block for at most 20-30ms.

ftw64, 2023-07-23:
Another ESP8266 reporting these warnings:

uart:
  id: uart_bus
  tx_pin: TX
  rx_pin: RX
  baud_rate: 115200
  rx_buffer_size: 1500

dsmr:
  max_telegram_length: 1500
  
text_sensor:
  - platform: dsmr
    identification:
      name: "${name}: DSMR Identification"
    p1_version:
      name: "${name}: DSMR Version"
      
sensor:
  - platform: dsmr
    energy_delivered_tariff1:
      name: "${name}: Energy Consumed Tariff 1"
    energy_delivered_tariff2:
      name: "${name}: Energy Consumed Tariff 2"
    energy_returned_tariff1:
      name: "${name}: Energy Returned Tariff 1"
    energy_returned_tariff2:
      name: "${name}: Energy Returned Tariff 2"
    power_delivered:
      name: "${name}: Power Consumed"
    power_returned:
      name: "${name}: Power Returned"
    gas_delivered:
      name: "${name}: Gas Consumed"
    electricity_failures:
      name: "${name}: Electricity Failures"
    electricity_long_failures:
      name: "${name}: Long Electricity Failures"
    voltage_l1:
      name: "${name}: Voltage Phase 1"

[19:51:25][D][sensor:093]: 'P1-DSMR-01: Uptime': Sending state 161.98500 s with 0 decimals of accuracy
[19:51:30][D][sensor:093]: 'P1-DSMR-01: Energy Consumed Tariff 1': Sending state 13168.29590 kWh with 3 decimals of accuracy
[19:51:30][D][sensor:093]: 'P1-DSMR-01: Energy Consumed Tariff 2': Sending state 12658.14258 kWh with 3 decimals of accuracy
[19:51:30][D][sensor:093]: 'P1-DSMR-01: Energy Returned Tariff 1': Sending state 5330.78613 kWh with 3 decimals of accuracy
[19:51:30][D][sensor:093]: 'P1-DSMR-01: Energy Returned Tariff 2': Sending state 12150.04785 kWh with 3 decimals of accuracy
[19:51:30][D][sensor:093]: 'P1-DSMR-01: Power Consumed': Sending state 0.00000 kW with 3 decimals of accuracy
[19:51:30][D][sensor:093]: 'P1-DSMR-01: Power Returned': Sending state 0.09200 kW with 3 decimals of accuracy
[19:51:30][D][sensor:093]: 'P1-DSMR-01: Gas Consumed': Sending state 10522.21777 m³ with 3 decimals of accuracy
[19:51:30][D][sensor:093]: 'P1-DSMR-01: Electricity Failures': Sending state 4.00000  with 0 decimals of accuracy
[19:51:30][D][sensor:093]: 'P1-DSMR-01: Long Electricity Failures': Sending state 8.00000  with 0 decimals of accuracy
[19:51:30][D][sensor:093]: 'P1-DSMR-01: Power Consumed Phase 1': Sending state 0.00000 kW with 3 decimals of accuracy
[19:51:30][D][sensor:093]: 'P1-DSMR-01: Power Produced Phase 1': Sending state 0.09300 kW with 3 decimals of accuracy
[19:51:30][D][text_sensor:064]: 'P1-DSMR-01: DSMR Identification': Sending state 'KFM5KAIFA-METER'
[19:51:30][D][text_sensor:064]: 'P1-DSMR-01: DSMR Version': Sending state '42'
[19:51:30][W][component:204]: Component dsmr took a long time for an operation (0.07 s).
[19:51:30][W][component:205]: Components should block for at most 20-30ms.

ftw64, 2023-07-23:
And another one (ESP32):

uart:
  - id: uart_2
    rx_pin: GPIO16
    tx_pin: GPIO17
    baud_rate: 9600

# A number of One-Wire sensors can be added to the ESP32.
dallas:
  - pin: GPIO27
    update_interval: 2s

output:
  - platform: ledc
    id: onboard_led
    pin:
      number: GPIO25

text_sensor:
  - platform: template
    name: "${name}: Monitor Status"
    id: monitor_status
    icon: "mdi:text-box"

  - platform: template
    # Response of "REV": ROM 1 CRC
    name: "${name}: Heater ROM 1 Checksum"
    id: rom_test_1_checksum
    icon: "mdi:text-box"

  - platform: template
    # Response of "REV": ROM 2 CRC
    name: "${name}: Heater ROM 2 Checksum"
    id: rom_test_2_checksum
    icon: "mdi:text-box"

  - platform: template
    # Response of "REV": Software release (i.e., "v2.71")
    name: "${name}: Heater Software release"
    id: software_release
    icon: "mdi:text-box"

  - platform: template
    # Response of "REV": Hardware release (i.e., "ic2kknl01*P080925*")
    name: "${name}: Heater Hardware release"
    id: hardware_release
    icon: "mdi:text-box"

# NOT SUPPORTED ON HRE 36/30 (2009, v2.71 software): "REW"
# - platform: template
#   name: "${name}: Heater DSP ROM 1 Checksum"
#   id: dsp_rom_test_1_checksum
#   icon: "mdi:text-box"
#
# - platform: template
#   name: "${name}: Heater DSP ROM 2 Checksum"
#   id: dsp_rom_test_2_checksum
#   icon: "mdi:text-box"
#
# - platform: template
#   name: "${name}: Heater DSP Hardware release"
#   id: dsp_hardware_release
#   icon: "mdi:text-box"
#
# - platform: template
#   name: "${name}: Heater DSP Software release"
#   id: dsp_software_release
#   icon: "mdi:text-box"

[... 646 lines cut ...]
[19:52:01][W][component:204]: Component custom_component took a long time for an operation (0.09 s).
[19:52:01][W][component:205]: Components should block for at most 20-30ms.

[19:53:34][D][dallas.sensor:143]: 'HRE3630: T-CV warm water temperature': Got Temperature=21.8°C
[19:53:34][D][sensor:094]: 'HRE3630: T-CV warm water temperature': Sending state 21.75000 °C with 0 decimals of accuracy
[19:53:34][D][dallas.sensor:143]: 'HRE3630: T-CV cold water temperature': Got Temperature=22.1°C
[19:53:34][D][sensor:094]: 'HRE3630: T-CV cold water temperature': Sending state 22.06250 °C with 0 decimals of accuracy
[19:53:34][D][dallas.sensor:143]: 'HRE3630: T-CV exhaust fume temperature': Got Temperature=22.1°C
[19:53:34][D][sensor:094]: 'HRE3630: T-CV exhaust fume temperature': Sending state 22.12500 °C with 0 decimals of accuracy
[19:53:36][D][dallas.sensor:143]: 'HRE3630: T-CV warm water temperature': Got Temperature=21.8°C
[19:53:36][D][sensor:094]: 'HRE3630: T-CV warm water temperature': Sending state 21.81250 °C with 0 decimals of accuracy
[19:53:36][D][dallas.sensor:143]: 'HRE3630: T-CV cold water temperature': Got Temperature=22.1°C
[19:53:36][D][sensor:094]: 'HRE3630: T-CV cold water temperature': Sending state 22.06250 °C with 0 decimals of accuracy
[19:53:36][D][dallas.sensor:143]: 'HRE3630: T-CV exhaust fume temperature': Got Temperature=22.1°C
[19:53:36][D][sensor:094]: 'HRE3630: T-CV exhaust fume temperature': Sending state 22.12500 °C with 0 decimals of accuracy
[19:53:36][D][text_sensor:064]: 'HRE3630: Current time': Sending state '2023-07-23 19:53:36 CEST'
[19:53:37][D][text_sensor:064]: 'HRE3630: Monitor Status': Sending state 'Fetching: S?\r'
[19:53:37][D][Intergas HRE:210]: Send command S?\r
[19:53:37][D][text_sensor:064]: 'HRE3630: Monitor Status': Sending state 'Processing...'
[19:53:37][D][sensor:094]: 'HRE3630: Heat Exchanger Temperature (S0)': Sending state 64.58000 °C with 0 decimals of accuracy
[19:53:37][D][sensor:094]: 'HRE3630: Delivery Temperature (S1)': Sending state 57.56000 °C with 0 decimals of accuracy
[19:53:37][D][sensor:094]: 'HRE3630: Hot water Temperature (S3)': Sending state 45.98000 °C with 0 decimals of accuracy
[19:53:37][D][sensor:094]: 'HRE3630: Heater Pressure': Sending state 1.75000 BAR with 1 decimals of accuracy
[19:53:37][W][component:204]: Component custom_component took a long time for an operation (0.09 s).
[19:53:37][W][component:205]: Components should block for at most 20-30ms.

descipher, 2023-07-24:
Tack on atm90e32

23:00:40 | [W] | [component:204] | Component atm90e32.sensor took a long time for an operation (0.05 s).
23:00:40 | [W] | [component:205] | Components should block for at most 20-30ms.

jeleniain, 2023-07-24:
Hi guys,
stumbled upon this thread trying to figure out the issue.
I am also seeing the warning following warning after updating my ESPHome to 2023.7.0 and updating my two tag readers:
```
[W][component:204]: Component pn532 took a long time for an operation (0.10 s).
[W][component:205]: Components should block for at most 20-30ms.
```
and are also using the code from: https://github.com/adonno/tagreader/blob/master/tagreader.yaml

The odd thing is that I have two of the exact same tag readers with identical code (except for a different static IP address) and only one of them is throwing the warning. I tried to re-flash the device but same behavior. Not sure what to make of it.

thatslolo, 2023-07-24:
```
[esp32.preferences:143]: Saving 7 preferences to flash: 2 cached, 5 written, 0 failed
[W][component:204]: Component preferences took a long time for an operation (0.06 s).
[component:205]: Components should block for at most 20-30ms.
```

Don't know which integration causes this. I do have multiple integration sensors and globals with restore configured, I guess these are the culprits.

```
globals:
  - id: rgb_ready
    type: bool
    restore_value: false
    initial_value: 'false'
  - id: total_energy
    type: float
    restore_value: true
    # initial_value: '5.96'
  - id: energy_current_run_
    type: float
    restore_value: true
  - id: energy_last_run_
    type: float
    restore_value: true
  - id: energy_current_month_
    type: float
    restore_value: true
  - id: start_time
    type: float
    restore_value: true
  - id: duration_last_run_
    type: float
    restore_value: true

sensor:
  - platform: total_daily_energy
    name: "${channel_1} monthly energy"
    power_id: power
    restore: true
    state_class: total_increasing
    unit_of_measurement: kWh
    filters:
      # Multiplication factor from W to kW is 0.001
      - multiply: 0.001
      - lambda: !lambda |-
          static auto last_state = x;
          if (x < last_state) { // x was reset
            id(total_energy) += last_state;
            ESP_LOGI("main", "Energy channel 1 was reset: %f", id(total_energy));
          }
          last_state = x;
          return id(total_energy) + x;

  - platform: integration
    name: "Energy Current Run"
    sensor: power
    id: energy_current_run
    time_unit: h
    unit_of_measurement: "kWh"
    restore: true
    filters:
      # Multiplication factor from W to kW is 0.001
      - multiply: 0.001
      - lambda: |-
          id(energy_current_run_) = x;
          return x;

  - platform: integration
    name: "Energy Current Month"
    sensor: power
    id: energy_current_month
    time_unit: h
    unit_of_measurement: "kWh"
    restore: true
    filters:
      # Multiplication factor from W to kW is 0.001
      - multiply: 0.001
      - lambda: |-
          id(energy_current_month_) = x;
          return x;

  - platform: template
    name: "Energy Last Run"
    icon: 'mdi:flash-outline'
    id: energy_last_run
    lambda: return id(energy_last_run_);
    accuracy_decimals: 3
    device_class: 'energy'
    state_class: 'total'
    unit_of_measurement: "kWh"
    # update_interval: 60s

  - platform: template
    name: "Duration Last Run"
    icon: 'mdi:timer'
    id: duration
    lambda: return id(duration_last_run_);
    accuracy_decimals: 0
    device_class: 'duration'
    state_class: 'total'
    unit_of_measurement: "s"
    # update_interval: 60s
```

shannonfritz, 2023-07-24:
I just saw this after updating my Emporia Vue 2 using ESPHome 2023.7.0.  This is after doing a "clean build files" before installing the updated config file.

```
Processing emporia1 (board: esp32dev; framework: espidf; platform: platformio/espressif32@5.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
```

The base of the configuration came from here https://github.com/emporia-vue-local/esphome and after installing successfully, the log starts showing this warning every ~30s

```
[17:01:14][W][component:204]: Component emporia_vue.sensor took a long time for an operation (0.06 s).
[17:01:14][W][component:205]: Components should block for at most 20-30ms.
[17:01:45][W][component:204]: Component emporia_vue.sensor took a long time for an operation (0.06 s).
[17:01:45][W][component:205]: Components should block for at most 20-30ms.
[17:02:16][W][component:204]: Component emporia_vue.sensor took a long time for an operation (0.06 s).
[17:02:16][W][component:205]: Components should block for at most 20-30ms.
```

The config is quite large, so I collapsed it in a details view below...
<details><summary>emporia.yaml</summary>

```
# Emporia Vue 2
# 2023-Jul-24
# - Changed Source for external_components (https://github.com/emporia-vue-local/esphome)
# - Added "Unaccounted Power" sensor for power not recorded on the smaller clamps
# - Added buzzer and light components
# - Added uptime sensor
# 2022-Feb-03
# - Initial build based on https://gist.github.com/flaviut/93a1212c7b165c7674693a45ad52c512

substitutions:
  devicename: "Emporia1"
  deviceid: "emporia1"

esphome:
  name: "${deviceid}"

external_components:
  - source: github://emporia-vue-local/esphome@dev
    components: [ emporia_vue ]

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: recommended

# Enable logging
logger:
  logs:
    # Don't log on each sensor reading, that's way too loud. We can see them in
    # Home Assistant anyway, or re-enable them per-device if needed.
    sensor: INFO

# Enable Home Assistant API
api:
  services:
    - service: play_rtttl
      variables:
        song_str: string
      then:
        - rtttl.play:
            rtttl: !lambda 'return song_str;'

ota:
  password: !secret ota_password

wifi:
[... 179 lines cut ...]
        id(cir03).state -
        id(cir04).state -
        id(cir05).state -
        id(cir06).state -
        id(cir07).state -
        id(cir08).state -
        id(cir09).state -
        id(cir10).state -
        id(cir11).state -
        id(cir12).state -
        id(cir13).state -
        id(cir14).state -
        id(cir15).state -
        id(cir16).state);
    update_interval: 10s
    device_class: power
    state_class: measurement
    unit_of_measurement: "W"
  # Use a Helper Sensor to record the total power readings over time (daily)
  - platform: total_daily_energy
    name: "Unaccounted Daily Energy"
    power_id: unaccounted_power
    accuracy_decimals: 0
```
</details>

Kyryer, 2023-07-24:
mh-z19 causes this warning too. 
i see trouble only with mh-z19

this is the using config:
`uart:
  rx_pin: GPIO12
  tx_pin: GPIO14
  baud_rate: 9600

dallas:
  - pin: GPIO2
    update_interval: 30s

sensor:
  - platform: mhz19
    co2:
      name: "MH-Z19 CO2 Value"
    temperature:
      name: "MH-Z19 Temperature"
    update_interval: 60s
    automatic_baseline_calibration: false
  - platform: dallas
    address: 0xdd0116003599ff28
    name: "D18B20 Temperature"
    resolution: 12`

i try update interval for mh-z19 30s and 60s - result is not different

descipher, 2023-07-25:
@jesserockz There appears to be a large number of components which do not use callbacks. Seems to be the main reason for the blocking. I did some digging with atm90e32 and its not doing anything heavy but it still runs just under 50ms to do its fundamental outputs, we could certainly lower the amount a serial logging on some of the components. But the I would assume the right path is to convert them to use callbacks where possible.

descipher, 2023-07-25:
For anyone reporting here, the blocking warning message was  previously exposed when enabling verbose logging. The blocking event message should be a warning and thus why it was changed. We should address them as time permits. Please keep adding them here.

cnovicktx, 2023-07-25:
SSD1306 using SPI suffering from the same log entry issue

The 'frequency: 400kHz' fix only applies to i2c interface and not SPI

descipher, 2023-07-25:
@jesserockz @ssieb I did a test on the atm90e32 component used on within my energy meter product PM1.

Processing power-meter-1 (board: esp12e; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash

The device is capable of much higher baud rates for logging.

The device serial logs the following data ever 15s:

[11:25:59][D][sensor:093]: 'PM1 Total Watts': Sending state 744.90210 W with 0 decimals of accuracy
[11:25:59][D][sensor:093]: 'PM1 Total kWh': Sending state 135.77835 kWh with 2 decimals of accuracy
[11:26:00][D][sensor:093]: 'PM1 Total Amps': Sending state 9.35700 A with 2 decimals of accuracy
[11:26:01][D][sensor:093]: 'PM1 Volts': Sending state 119.71000 V with 1 decimals of accuracy
[11:26:02][D][sensor:093]: 'PM1 CT1 Amps': Sending state 2.89700 A with 2 decimals of accuracy
[11:26:02][D][sensor:093]: 'PM1 CT2 Amps': Sending state 6.46600 A with 2 decimals of accuracy
[11:26:02][D][sensor:093]: 'PM1 CT1 Watts': Sending state 263.41248 W with 1 decimals of accuracy
[11:26:02][D][sensor:093]: 'PM1 CT2 Watts': Sending state 482.77182 W with 1 decimals of accuracy
[11:26:02][D][sensor:093]: 'PM1 Freq': Sending state 59.95000 Hz with 1 decimals of accuracy
[11:26:02][D][sensor:093]: 'PM1 IC Temperature': Sending state 50.00000 °C with 1 decimals of accuracy
[11:26:02][W][component:204]: Component atm90e32.sensor took a long time for an operation (0.05 s).
[11:26:02][W][component:205]: Components should block for at most 20-30ms.

This results in a 50ms blocking in the processing of publish_state calls.
The actual data retrieval from the atm90e32 is running spi at 1Mhz (was previously 250Khz .. not sure why it was set so slow) and not a significant factor in the total blocking time.

If I increase the logging baud_rate to 256000 vs the default 115200 the blocking time is reduced below the 50ms warn threshold and thus no more warning. So generally logging is a major factor of the blocking time problem.

To validate this fact I set the logging level to VERY_VERBOSE with the same configuraton and the result is 160ms of blocking time.

oivindklausen, 2023-07-25:
Also se this warning using a Custom Component for Norwegain HAN Energy Meter running UART at 2400Baud with packt length around 270Bytes. Frame sync is found and the complete packet is processed before returned. Implemented according to "Custom UART Device". Update to the Custom UART description would be appreciated to avoid possible future breaking changes.

descipher, 2023-07-25:
> Also se this warning using a Custom Component for Norwegain HAN Energy Meter running UART at 2400Baud with packt length around 270Bytes. Frame sync is found and the complete packet is processed [...]

With a HAN data format @ 2400 Baud thats 3.75ms per character * 270 = 1012.5ms if it was waiting for the sync b4 returning. What is it logging for the operation time? I doubt that it's waiting that long, something else may be happening here.

z3liff, 2023-07-25:
> I'm seeing this, but with pn532 as the component name:
>
> [07:58:54][W][component:204]: Component pn532 took a long time for an operation (0.10 s). [07:58:54][W][component:205]: Components should [...]

seeing the same thing on pn532_i2c

segator, 2023-07-25:
Same bug here after upgrading, on shelly 2.5pm for ade7953 and preferences

```
23:47:51 | [W] | [component:204] | Component ade7953.sensor took a long time for an operation (0.06 s).
23:47:51 | [W] | [component:205] | Components should block for at most 20-30ms.
23:48:40 | [W] | [component:204] | Component preferences took a long time for an operation (0.16 s).
23:48:40 | [W] | [component:205] | Components should block for at most 20-30ms.
```

patfelst, 2023-07-26:
```logs
[10:49:56][W][component:204]: Component st7789v.display took a long time for an operation (0.12 s).
[10:49:56][W][component:205]: Components should block for at most 20-30ms.
```
```yaml
display:
  - platform: st7789v
    model: TTGO_TDISPLAY_135X240
    backlight_pin: GPIO4
    cs_pin: GPIO5
    dc_pin: GPIO16
    reset_pin: GPIO23
    update_interval: 1s
    rotation: 90
```

ESPSabre, 2023-07-26:
bme680_bsec 
and
scd4x.sensor

alexruffell, 2023-07-26:
ssd1306.display
frequency: 400kHz - fixed it

sht3xd.sensor
frequency: 200kHz - fixed it on multiple different configs, but not all needed it
Note: I am assuming sht4xd is a close relative... it did not have any warnings at the default 50kHz

htu21d.sensor
frequency: 100 to 800 kHz - did not fix on multiple systems w/ different configs
Default implementation
```
[12:56:16][W][component:204]: Component htu21d.sensor took a long time for an operation (0.12 s).
[12:56:16][W][component:205]: Components should block for at most 20-30ms.
```

mics_4514.sensor
frequency: 100 to 800 kHz - did not fix
Default implementation
```
[13:18:05][W][component:204]: Component mics_4514.sensor took a long time for an operation (0.06 s).
[13:18:05][W][component:205]: Components should block for at most 20-30ms.
```

All systems running 2023.7

pepe59, 2023-07-26:
```
spi:
 clk_pin: 18
 mosi_pin: 23
 miso_pin: 19
 
display:
  - platform: ili9xxx
    model: ILI9341
    cs_pin: 14
    dc_pin: 27
    reset_pin: 33
    rotation: 90
```
```
[23:59:42][W][component:204]: Component ili9xxx.display took a long time for an operation (0.15 s).
[23:59:42][W][component:205]: Components should block for at most 20-30ms.
[23:59:43][W][component:204]: Component ili9xxx.display took a long time for an operation (0.15 s).
[23:59:43][W][component:205]: Components should block for at most 20-30ms.
```

pepe59, 2023-07-26:
```
i2c:
  sda: D1
  scl: D2
  
display:
  - platform: lcd_pcf8574
    dimensions: 16x2
    address: 0x3F
```
```
[00:15:57][W][component:204]: Component lcd_base took a long time for an operation (0.12 s).
[00:15:57][W][component:205]: Components should block for at most 20-30ms.
```

pepe59, 2023-07-26:
```
[00:32:52][W][component:204]: Component htu21d.sensor took a long time for an operation (0.12 s).
[00:32:52][W][component:205]: Components should block for at most 20-30ms.
```

TheWanderingTurkey, 2023-07-26:
`[18:12:41][W][component:204]: Component api took a long time for an operation (0.28 s).
[18:12:41][W][component:205]: Components should block for at most 20-30ms.`

This only shows up once at the initial API connection and does not repeat.

steelsparky, 2023-07-27:
Just updated and seeing on circuitsetup energy-meter. 

`<html>
<body>

20:13:11 | [W] | [component:204] | Component atm90e32.sensor took a long time for an operation (0.06 s).
-- | -- | -- | --
20:13:11 | [W] | [component:204] | Component atm90e32.sensor took a long time for an operation (0.06 s).
20:13:11 | [W] | [component:205] | Components should block for at most 20-30ms.

</body>
</html>`

Cougar, 2023-07-27:
```
[17:23:53][W][component:204]: Component st7789v.display took a long time for an operation (1.37 s).
```
I guess that this is a way too much :D I'm using this config as an example https://community.home-assistant.io/t/need-help-with-ttgo-t-display/448695/3

m0jek, 2023-07-27:
I just saw this "thread" when wondering why I was see the below in the espHome logs from **modbus_controller**. 
I am using with Epever Tracer8415AN MPPT controllers. Only noticed this after the recent update update ....
Here is a snippet of my logs.

```
17:00:53 | [W] | [component:204] | Component modbus_controller took a long time for an operation (0.05 s).
17:00:53 | [W] | [component:205] | Components should block for at most 20-30ms.
17:00:54 | [W] | [component:204] | Component modbus_controller took a long time for an operation (0.13 s).
17:00:54 | [W] | [component:205] | Components should block for at most 20-30ms.
```

espHome is running on a ESP32 PICO

The config I am using is based on this one, but  edited for my setup, but generally the same.

https://devices.esphome.io/devices/epever_mptt_tracer_an

PaFcio, 2023-07-27:
Using really rudimentary config for PN532:
...
esphome:
  name: nfc
  friendly_name: nfc

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "<redacted>"

ota:
  password: !secret ota_pass

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "NFC"
    password: !secret ap_pass

captive_portal:

i2c:
  sda: 4
  scl: 5
  scan: true
  frequency: 200kHz
  

pn532_i2c:
  address: 0x24
  update_interval: 1s

  on_tag:
    then:
      - homeassistant.tag_scanned: !lambda 'return x;'
...

Receive below as with others that posted before:
```
[20:23:54][W][component:204]: Component pn532 took a long time for an operation (0.10 s).
[20:23:54][W][component:205]: Components should block for at most 20-30ms.
```

Tried 400kHz to same effect as based on following quote from datasheet:
_The PN532 is configured with I2C address 0x48 and is able to support a clock frequency
up to 400 kHz.
The data order used is MSB first._

tungmeister, 2023-07-28:
seeing this constantly for esp32_touch:
```
[07:37:01][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:01][W][component:205]: Components should block for at most 20-30ms.
[07:37:01][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:01][W][component:205]: Components should block for at most 20-30ms.
[07:37:01][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:01][W][component:205]: Components should block for at most 20-30ms.
[07:37:01][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:01][W][component:205]: Components should block for at most 20-30ms.
[07:37:01][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:01][W][component:205]: Components should block for at most 20-30ms.
[07:37:01][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:01][W][component:205]: Components should block for at most 20-30ms.
[07:37:02][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:02][W][component:205]: Components should block for at most 20-30ms.
[07:37:02][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:02][W][component:205]: Components should block for at most 20-30ms.
[07:37:02][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:02][W][component:205]: Components should block for at most 20-30ms.
[07:37:02][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:02][W][component:205]: Components should block for at most 20-30ms.
[07:37:02][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:02][W][component:205]: Components should block for at most 20-30ms.
[07:37:02][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:02][W][component:205]: Components should block for at most 20-30ms.
[07:37:02][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:02][W][component:205]: Components should block for at most 20-30ms.
[07:37:02][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:02][W][component:205]: Components should block for at most 20-30ms.
[07:37:02][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:02][W][component:205]: Components should block for at most 20-30ms.
[07:37:02][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:02][W][component:205]: Components should block for at most 20-30ms.
[07:37:02][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:02][W][component:205]: Components should block for at most 20-30ms.
[07:37:03][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:03][W][component:205]: Components should block for at most 20-30ms.
[07:37:03][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:03][W][component:205]: Components should block for at most 20-30ms.
[07:37:03][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:03][W][component:205]: Components should block for at most 20-30ms.
[07:37:03][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:03][W][component:205]: Components should block for at most 20-30ms.
[07:37:03][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:03][W][component:205]: Components should block for at most 20-30ms.
[07:37:03][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:03][W][component:205]: Components should block for at most 20-30ms.
[07:37:03][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:03][W][component:205]: Components should block for at most 20-30ms.
[07:37:03][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:03][W][component:205]: Components should block for at most 20-30ms.
[07:37:03][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:03][W][component:205]: Components should block for at most 20-30ms.
[07:37:03][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:03][W][component:205]: Components should block for at most 20-30ms.
[07:37:03][W][component:204]: Component esp32_touch took a long time for an operation (0.06 s).
[07:37:03][W][component:205]: Components should block for at most 20-30ms.
```
I have 8 touch sensors configured, all controlling and LED behind each "button" depending on hass states and input with delays etc, this being one of the most complex ones:

```
  - platform: esp32_touch
    name: "Dan Blanket Touch"
    pin: GPIO12
    threshold: 100
    on_multi_click:
    - timing:
        - ON for at most 400ms
      then:
[... 51 lines cut ...]
                  id: dblanketled
                  brightness: 100%
              - delay: 0.1s
              - light.turn_on: 
                  id: dblanketled
                  brightness: 30%
              - delay: 0.1s
            else:
              - delay: 0.1s
              - light.turn_on: 
                  id: dblanketled
                  brightness: 30%
              - delay: 0.1s
        - if:
            condition:
              - binary_sensor.is_on: blanket
            then:
              - light.turn_on:
                  id: dblanketled
                  brightness: 80%
              - light.turn_on:
                  id: lblanketled
                  brightness: 80%
```
would this be the reason for the long times rather than an actual issue?

Littletiger, 2023-07-28:
[11:48:14][W][component:204]: Component sgp4x.sensor took a long time for an operation (0.06 s).
[11:48:14][W][component:205]: Components should block for at most 20-30ms.

  - platform: sgp4x
    nox:
      id: nox
      name: ${upper_devicename} NOx
    voc:
      id: voc
      name: ${upper_devicename} VOC
    store_baseline: yes
    update_interval: 60s
    compensation:
      temperature_source: temp
      humidity_source: humidity

I also had the message for the ssd1306 and this has been fixed by setting the i2c frequency to 400kHz

pgolawsk, 2023-07-28:
Although i2c frequency set to 400kHz I'm getting those for 2 sensors:
`

2023-07-28 21:53:46.338 | [W][component:204]: Component tcs34725.sensor took a long time for an operation (0.08 s). |  
-- | -- | --
  |   | 2023-07-28 21:53:16.340 | [W][component:205]: Components should block for at most 20-30ms.

`
`

2023-07-28 22:04:27.954 | [W][component:204]: Component bme680.sensor took a long time for an operation (0.05 s). |  
-- | -- | --
  |   | 2023-07-28 22:03:57.961 | [W][component:205]: Components should block for at most 20-30ms.

`

PhilippeLRQ, 2023-07-29:
Hello, thanks to all developers to work so hard on this marvelous project.

I have the same issue but with a brand new esp32 connected to SM300D2 Air Quality Sensor.

"
[18:20:46][D][sm300d2:045]: Successfully read SM300D2 data
[18:20:46][D][sm300d2:058]: Received CO₂: 403 ppm
[18:20:46][D][sensor:094]: 'SM300D2 CO2': Sending state 403.00000 ppm with 0 decimals of accuracy
[18:20:46][D][sm300d2:062]: Received Formaldehyde: 0 µg/m³
[18:20:47][D][sensor:094]: 'SM300D2 Formaldehyde': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[18:20:47][D][sm300d2:066]: Received TVOC: 2 µg/m³
[18:20:47][D][sensor:094]: 'SM300D2 TVOC': Sending state 2.00000 µg/m³ with 0 decimals of accuracy
[18:20:47][D][sm300d2:070]: Received PM2.5: 33 µg/m³
[18:20:47][D][sensor:094]: 'SM300D2 PM2.5': Sending state 33.00000 µg/m³ with 0 decimals of accuracy
[18:20:47][D][sm300d2:074]: Received PM10: 59 µg/m³
[18:20:47][D][sensor:094]: 'SM300D2 PM10': Sending state 59.00000 µg/m³ with 0 decimals of accuracy
[18:20:47][D][sm300d2:078]: Received Temperature: 27.80 °C
[18:20:47][D][sensor:094]: 'SM300D2 Temperature': Sending state 27.80000 °C with 1 decimals of accuracy
[18:20:47][D][sm300d2:082]: Received Humidity: 52.30 percent
[18:20:47][D][sensor:094]: 'SM300D2 Humidity': Sending state 52.30000 % with 1 decimals of accuracy
[18:20:47][W][component:204]: Component sm300d2.sensor took a long time for an operation (0.08 s).
[18:20:47][W][component:205]: Components should block for at most 20-30ms.
"

My configuration is the one included in the dedicated page for this sensor, and it looks to work fine.

"
uart:
  rx_pin:
    number: 3
    mode: INPUT_PULLDOWN
  tx_pin: 1
  baud_rate: 9600

Ssensor: 
   - platform: sm300d2
    co2:
      name: "SM300D2 CO2"
    formaldehyde:
      name: "SM300D2 Formaldehyde"
    tvoc:
      name: "SM300D2 TVOC"
    pm_2_5:
      name: "SM300D2 PM2.5"
    pm_10_0:
      name: "SM300D2 PM10"
    temperature:
      name: "SM300D2 Temperature"
    humidity:
      name: "SM300D2 Humidity"
    update_interval: 60s
"

Thanks again.

Philippe

Protoncek, 2023-07-30:
Hi guys!
As all of you, i also have same problem with PN532, running on esp32. But that's not why i'm writing - it's because these messages noticeably slow down PN532. When i move card to a reader it reads it and it should respond with 3 beeps. Sometimes these beeps are "funny" - either one is lost, or second or third is played after small delay, or one of them is longer than it should be... when looking log output it's clear that this delay or corruption is caused by above message. Hiding these messages in logger (with component: error) doesn't help. 
So, pn532 is unusable at this moment. Changing bus speed doesn't help - i tried from 10kHz to 800kHz - message always appears. 
Are there any solutions other than throwing pn532 to the drawer and waiting for better times?

RC522 works flawlessly, though.

nielsnl68, 2023-07-30:
I don't belief this is mentioned before so i will add this as well:

```
[VV][api.service:324]: send_media_player_state_response: MediaPlayerStateResponse {
  key: 1433258598
  state: MEDIA_PLAYER_STATE_PLAYING
  volume: 1
  muted: NO
}
[W][component:204]: Component i2s_audio.media_player took a long time for an operation (0.57 s).
[W][component:205]: Components should block for at most 20-30ms.
[W][component:204]: Component i2s_audio.media_player took a long time for an operation (0.34 s).
[W][component:205]: Components should block for at most 20-30ms.

[VV][api.service:324]: send_media_player_state_response: MediaPlayerStateResponse {
  key: 1433258598
  state: MEDIA_PLAYER_STATE_IDLE
  volume: 1
  muted: NO
}
[W][component:204]: Component i2s_audio.media_player took a long time for an operation (0.35 s).
[W][component:205]: Components should block for at most 20-30ms.
```

xavitron-1, 2023-07-30:
My two esp32 with displays are reporting this after the update:

- Waveshare E-Paper  2.13in (TTGO) 122px x 250px
- TTGO T-Display 135x240 (ST7789V )

descipher, 2023-07-30:
> Hi guys! As all of you, i also have same problem with PN532, running on esp32. But that's not why i'm writing - it's because these messages noticeably slow down PN532. When i move card to a reader [...]

With an Esp32 try setting the logger to

baud_rate: 921600

Protoncek, 2023-07-30:
Sadly it's no difference. I tried 921600, i tried zero, i tried log level "error", but lag is still present. my beeper code is this:

```
output: 
  - platform: template
    id: tag_yes
    type: binary
    write_action:
      - switch.turn_on: buzzer
      - delay: 50ms
      - switch.turn_off: buzzer
      - delay: 50ms
      - switch.turn_on: buzzer
      - delay: 50ms
      - switch.turn_off: buzzer
      - delay: 50ms
      - switch.turn_on: buzzer
      - delay: 50ms
      - switch.turn_off: buzzer
```
but in real these 3 short beeps are mostly like "bep beeeeep   bep",  or "bep (1s pause) bep bep" etc. So something is seriously causing lagging. 

BTW - this is in arduino environment. I tried with esp-idf, but on 400kHz reader is not even found, at 100kHz is found but it's not working - nothing happens when tag should be read, like it's not there.

descipher, 2023-07-30:
Yikes, clearly the 50ms delay is not compatible with the blocking check routine. Bring it down to 40ms and see what happens.

Protoncek, 2023-07-30:
Tried, unsuccesfull. I uploaded video on my youtube channel how thing works. It can happen that it's ok a few times, but not constantly:
_https://youtube.com/shorts/4KYrt_ZMZDo?feature=share_

PaFcio, 2023-07-30:
> Sadly it's no difference. I tried 921600, i tried zero, i tried log level "error", i even tried with logger disabled, but lag is still present. my beeper code is this:
> 
> ```
> output: 
>   - platform: template
>     id: tag_yes
>     type: binary
>     write_action:
>       - switch.turn_on: buzzer
>       - delay: 50ms
>       - switch.turn_off: buzzer
>       - delay: 50ms
>       - switch.turn_on: buzzer
>       - delay: 50ms
>       - switch.turn_off: buzzer
>       - delay: 50ms
>       - switch.turn_on: buzzer
>       - delay: 50ms
>       - switch.turn_off: buzzer
> ```
> but in real these 3 short beeps are mostly like "bep beeeeep   bep",  or "bep (1s pause) bep bep" etc. So something is seriously causing lagging. 
> 
> BTW - this is in arduino environment. I tried with esp-idf, but on 400kHz reader is not even found, at 100kHz is found but it's not working - nothing happens when tag should be read, like it's not there.

Not sure that it would fix the issue but you might try to run the buzzer with RTTTL Buzzer component.

Batzi1967, 2023-07-30:
the same with the media_player:
```
[20:57:52][D][media_player:059]: 'ESP32 Assistant03Media Player' - Setting
[20:57:52][D][media_player:066]:   Media URL: http://192.168.178.101:8123/media/local/03%20AliBabo.mp3?authSig=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiI4MWFiZGQ2NWFlYmE0MGMyOTI3NmFjOWMyMWM1YTM4YyIsInBhdGgiOiIvbWVkaWEvbG9jYWwvMDMgQWxpQmFiby5tcDMiLCJwYXJhbXMiOltdLCJpYXQiOjE2OTA3NDM0NzIsImV4cCI6MTY5MDgyOTg3Mn0.eQ4SQq0OSmakp084fzlBXFz_VknqM_gVs0Hbn4octck
[20:57:52][W][component:204]: Component i2s_audio.media_player took a long time for an operation (0.54 s).
[20:57:52][W][component:205]: Components should block for at most 20-30ms.
```

mrc1600, 2023-07-30:
> I'm seeing this:
>
> ```
> [23:36:21][W][component:204]: Component esphome.coroutine took a long time for an operation (0.10 s).
> [23:36:21][W][component:205]: Components should block for at most 20-30ms. [...]

Also using Geoff Davis' Mitsubishi heat pump custom component, and seeing the same issue since the update.

[283 more replies]