esphome/issues #4718: Inkplate 6 partial update doesn't work since 2023.7.0

mariusbaumann, 2023-07-19:
### The problem

After installing the update, the display (inkplate 6) began to fully refresh every 5 seconds. It was configured like this:
- platform: inkplate6
   id: inkplate_display
   greyscale: false
   partial_updating: true
   full_update_every: 100

Then I've changed it to this:

- platform: inkplate6
  id: inkplate_display
  greyscale: false
  partial_updating: true
  update_interval: 60s
  full_update_every: 100

Now the display refreshes every 60s but doesn't partially update anymore. 

### Which version of ESPHome has the issue?

2023.7.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.7.2

### What platform are you using?

ESP32

### Board

inkplate 6

### Example YAML snippet

```yaml
display:
- platform: inkplate6
  id: inkplate_display
  greyscale: false
  partial_updating: true
  full_update_every: 100
```

alrouen, 2023-07-31:
Same issue

xaxix, 2023-08-14:
In my Inkplate 6 Setup (Home Assistant 2023.8.2, ESPHome 2023.7.1) partial Update works fine.

```
- platform: inkplate6
  id: inkplate_display
  greyscale: false
  partial_updating: true
  update_interval: 60s
  full_update_every: 30
```
But my inkplate 10 refuses the partial update.

solarkennedy, 2024-10-23:
Probably the same as https://github.com/esphome/issues/issues/4730