esphome/issues #4719: Animation does not work without image component

TillFleisch, 2023-07-19:
### The problem

The animation component won't work without the image component.

When adding an animation like this:
```
animation:
  - file: "nyan_64x32.gif"
    id: nyan_cat
    type: RGB565
    resize: 64x32
```

ESPHome will throw the following error:
```
Failed config

image: None
  
  'file' is a required option for [0].
  []
required key not provided
```

A workaround to this issue is adding a image component to the configuration:
```
image:
  - file: "pxArt.png"
    id: pxArt
```

The issue was not present prior to the 2023.7.0 update.

### Which version of ESPHome has the issue?

2023.7.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

animation

guillempages, 2023-07-20:
Not a solution, but I did see this and created a PR to add a test, so that when this is fixed, it does not happen again: https://github.com/esphome/esphome/pull/5057

guillempages, 2023-11-22:
This issue is still happening; not stale unfortunately

scarolan, 2024-03-02:
Same problem here.

guillempages, 2024-03-02:
@scarolan which ESPhome version are you using? This was fixed with 2023.12.0; there are now even tests to check that "animation" works without an explicit "image"

randybb, 2024-03-02:
@guillempages he has a different "problem". The usual "undefined reference to..." that needs cleaning of build files.
https://github.com/esphome/issues/issues/5560

guillempages, 2024-03-02:
Which "undefined reference to..." do you mean? On the original description of the issue, it was about the refactoring done to display/image/..., where image and animation were moved to their own components (separate form display), and thus the image code was not auto-included anymore for animations. 
At least that's what I understood when I commented the "issue is not stale"; at that point it had not been fixed yet ;-)

randybb, 2024-03-02:
@guillempages My comment was related to your last comment for scarolan + I added his issue that is totally unrelated to this one.

guillempages, 2024-03-02:
So this issue *could* be closed, right? ;-)

randybb, 2024-03-02:
This one probably yes as I tried for the very first time an animation and it works

scarolan, 2024-03-02:
@guillempages i tested it, is working now without the image.  Bravo!

closed 2024-03-03 by jesserockz