esphome/issues #4720: 2023.7.0 broke my YAML.....

andypnz, 2023-07-19:
### The problem

YAML broken by 2023.7.0 - was working with 2023.6.x

### Which version of ESPHome has the issue?

2023.7.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.7.2

### What platform are you using?

ESP32

### Board

esp-wrover-kit

### Example YAML snippet

```yaml
esphome:
  name: esp32-solar-mon

esp32:
  board: esp-wrover-kit
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "------------------"

ota:
  password: "--------------------------------------"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: -------------
    # Set this to the IP address of the router. Often ends with .1
    gateway: ----------------
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-Solar-Mon Fallback Hotspot"
    password: "--------------"

captive_portal:

substitutions:

# Weather entity from home assistant
  weather_entity: weather.home
# Weather now icon size
  icon_xy: '20x20'
[... 703 lines cut ...]
                          ^~~~~~~
/config/esphome/esp32-solar-mon.yaml:481:26: error: assignment of read-only location '"windy"[imgDict]'
          imgDict["windy"] = id(wpng_13);
                          ^~~~~~~
/config/esphome/esp32-solar-mon.yaml:482:34: error: assignment of read-only location '"windy-variant"[imgDict]'
          imgDict["windy-variant"] = id(wpng_14);
                                  ^~~~~~~
/config/esphome/esp32-solar-mon.yaml:483:32: error: assignment of read-only location '"exceptional"[imgDict]'
          imgDict["exceptional"] = id(wpng_15);
                                ^~~~~~~
/config/esphome/esp32-solar-mon.yaml:484:21: error: assignment of read-only location '""[imgDict]'
          imgDict[""] = id(wpng_0);
                     ^~~~~~
/config/esphome/esp32-solar-mon.yaml:597:32: error: no match for 'operator[]' (operand types are 'int' and 'std::__cxx11::string' {aka 'std::__cxx11::basic_string<char>'})
            it.image(0, h2, imgDict[id(weather_condition).state]);
                                ^
Compiling /data/esp32-solar-mon/.pioenvs/esp32-solar-mon/lib280/ESPAsyncWebServer-esphome/WebResponses.cpp.o
Compiling /data/esp32-solar-mon/.pioenvs/esp32-solar-mon/lib280/ESPAsyncWebServer-esphome/WebServer.cpp.o
*** [/data/esp32-solar-mon/.pioenvs/esp32-solar-mon/src/main.cpp.o] Error 1
========================= [FAILED] Took 31.27 seconds =========================
```

### Additional information

Was working with 2023.6.x

nielsnl68, 2023-07-20:
We are aware of this issue. From my knowledge it should have been fixed in a PR, but somehow it did not got merged yet.

For now could you add a small `image` this will add the image class and your YAML script should work again.

andypnz, 2023-07-20:
Thanks...

Changing this Image to image (I presume that is what you are referring to?)
```
//Map icons//
         std::map<std::string, image *> imgDict;
```
Just results in a slightly changed error list:
```
Compiling /data/esp32-solar-mon/.pioenvs/esp32-solar-mon/libbd6/libsodium/crypto_aead/chacha20poly1305/sodium/aead_chacha20poly1305.c.o
/config/esphome/esp32-solar-mon.yaml: In lambda function:
/config/esphome/esp32-solar-mon.yaml:468:36: error: template argument 2 is invalid
          std::map<std::string, image *> imgDict;
                                    ^
/config/esphome/esp32-solar-mon.yaml:468:36: error: template argument 4 is invalid
/config/esphome/esp32-solar-mon.yaml:469:32: error: assignment of read-only location '"clear-night"[imgDict]'
          imgDict["clear-night"] = id(wpng_1);
                                ^~~~~~
/config/esphome/esp32-solar-mon.yaml:470:27: error: assignment of read-only location '"cloudy"[imgDict]'
          imgDict["cloudy"] = id(wpng_2);
                           ^~~~~~
/config/esphome/esp32-solar-mon.yaml:471:24: error: assignment of read-only location '"fog"[imgDict]'
          imgDict["fog"] = id(wpng_3);
                        ^~~~~~
/config/esphome/esp32-solar-mon.yaml:472:25: error: assignment of read-only location '"hail"[imgDict]'
          imgDict["hail"] = id(wpng_4);
                         ^~~~~~
/config/esphome/esp32-solar-mon.yaml:473:30: error: assignment of read-only location '"lightning"[imgDict]'
          imgDict["lightning"] = id(wpng_5);
                              ^~~~~~
/config/esphome/esp32-solar-mon.yaml:474:36: error: assignment of read-only location '"lightning-rainy"[imgDict]'
          imgDict["lightning-rainy"] = id(wpng_6);
                                    ^~~~~~
/config/esphome/esp32-solar-mon.yaml:475:33: error: assignment of read-only location '"partlycloudy"[imgDict]'
          imgDict["partlycloudy"] = id(wpng_7);
                                 ^~~~~~
/config/esphome/esp32-solar-mon.yaml:476:28: error: assignment of read-only location '"pouring"[imgDict]'
          imgDict["pouring"] = id(wpng_8);
                            ^~~~~~
/config/esphome/esp32-solar-mon.yaml:477:26: error: assignment of read-only location '"rainy"[imgDict]'
          imgDict["rainy"] = id(wpng_9);
                          ^~~~~~
/config/esphome/esp32-solar-mon.yaml:478:26: error: assignment of read-only location '"snowy"[imgDict]'
          imgDict["snowy"] = id(wpng_10);
                          ^~~~~~~
/config/esphome/esp32-solar-mon.yaml:479:32: error: assignment of read-only location '"snowy-rainy"[imgDict]'
          imgDict["snowy-rainy"] = id(wpng_11);
                                ^~~~~~~
/config/esphome/esp32-solar-mon.yaml:480:26: error: assignment of read-only location '"sunny"[imgDict]'
          imgDict["sunny"] = id(wpng_12);
                          ^~~~~~~
/config/esphome/esp32-solar-mon.yaml:481:26: error: assignment of read-only location '"windy"[imgDict]'
          imgDict["windy"] = id(wpng_13);
                          ^~~~~~~
/config/esphome/esp32-solar-mon.yaml:482:34: error: assignment of read-only location '"windy-variant"[imgDict]'
          imgDict["windy-variant"] = id(wpng_14);
                                  ^~~~~~~
/config/esphome/esp32-solar-mon.yaml:483:32: error: assignment of read-only location '"exceptional"[imgDict]'
          imgDict["exceptional"] = id(wpng_15);
                                ^~~~~~~
/config/esphome/esp32-solar-mon.yaml:484:21: error: assignment of read-only location '""[imgDict]'
          imgDict[""] = id(wpng_0);
                     ^~~~~~
/config/esphome/esp32-solar-mon.yaml:597:32: error: no match for 'operator[]' (operand types are 'int' and 'std::__cxx11::string' {aka 'std::__cxx11::basic_string<char>'})
            it.image(0, h2, imgDict[id(weather_condition).state]);
```

jesserockz, 2023-07-20:
The issue here is that the `Image` class was moved from the display namespace (which lambdas have direct access to) to the `image` namespace.

The fix is to simply change to `std::map<std::string, image::Image *> imgDict;`

andypnz, 2023-07-20:
Thanks very much! Solved with that change :-)

closed 2023-07-20 by andypnz