esphome/issues #4721: 2023.7.0 WARNING Can't connect to ESPHome API [Errno 111]

PipeDeveloper, 2023-07-20:
### The problem

ESP32 connects to wifi, but not to HA. All entities are down, however the device is connected to wifi and respond to ESPhome

### Which version of ESPHome has the issue?

2023.7.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

The latets at the date

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

i2c, but this is not the issue

### Example YAML snippet

```yaml
#for documentation see project at https://github.com/Dilbert66/esphome-dsckeybus

substitutions:
  # change the name to suit your needs. This is what esphome will use as the friendly name for your component.
  # also used as a prefix for all fields
  systemName: "alarma" 
  
  #Only comes into effect if a password prompt occurs when arming eg. night mode
  accessCode: "0823"
  
  #used to select the default partition associated with the alarm panel messages
  defaultPartition: "1" 

  #zone expander addresses:
  # 9  - zones 9-16
  # 10 - zones 17-24
  # 11 - zones 25-32
  # 12 - zones 33-40 (for systems with 64 zone support)
  # 13 - zones 41-48 (for systems with 64 zone support)
  # 14 - zones 49-56 (for systems with 64 zone support)
  # 16 - zones 57-64 (for systems with 64 zone support)  
  expanderAddr1: "0" # 1st zone expander emulator address to use . Set to 0 to disable. 
  expanderAddr2: "0" # 2nd expander emulator address to use . Set to 0 to disable. 
  
  #ESP32 Pins
  dscClockPin: "22"
  dscReadPin: "21"
  dscWritePin: "18"
  
  maxZones: "32" # maximum amount of zones your system supports
  
esphome:
  name: $systemName
  platform: ESP32
  board: nodemcu-32s

  includes:
    # subdirectory path where custom component *.h and *.cpp files are located
    - dscKeybusInterface/
[... 485 lines cut ...]
  - platform: template
    name: "$systemName Connection"
    id: connection_status_switch
    lambda: |-
      return dsc.keybusConnected;
    icon: "mdi:shield-link-variant"
    turn_on_action:
      - switch.toggle: restart_switch
    turn_off_action:
      - lambda: |-
          disconnectKeybus();
  - platform: restart
    id: restart_switch
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.7.0
INFO Reading configuration /config/esphome/alarma.yaml...
INFO Detected timezone 'America/Santiago'
INFO Starting log output from alarma.local using esphome API
WARNING Can't connect to ESPHome API for alarma.local: Error connecting to ('192.168.1.30', 6053): [Errno 111] Connect call failed ('192.168.1.30', 6053) (SocketAPIError)
INFO Trying to connect to alarma.local in the background
```

PipeDeveloper, 2023-07-20:
I pluged one of my devices on USB, here is the log
[logs (1).txt](https://github.com/esphome/issues/files/12101511/logs.1.txt)

dscao, 2023-07-20:
I have the same issues , one esp8266 in esphome 2023.7.0.
No problem in esphome 2023.6.4

ijusss, 2023-07-20:
same

[image]

coolstuff99, 2023-07-21:
> same
> [image]
> 
> EDIT - resolved Resolved by properly adding a quots to mqtt login and password.
> 
> ```
> mqtt:
>   broker: 192.168.0.123
>   username: "login"
>   password: "pass"
> ```

May or may not help to others since in my case is still failing (as described by @PipeDeveloper) and I don't have mqtt enabled for this specific esp32.
My workaround has been to switch to latest 2023.6.5 version for now :(

PipeDeveloper, 2023-07-23:
Thank you @JohnDepon got it online again

closed 2023-07-23 by PipeDeveloper

v12345vtm, 2023-07-30:
same issue here  : but i dont use dsckey.....

my yaml file  : ` 
esphome:
  name: arduinotest
  friendly_name: Arduinotest
  #includes:
  #- customvith.h

web_server:
  port : 80

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "J2+3oqEO6eDX0EleMvkLcTooGO0x2MaNFk+C/nFkRbU="

ota:
  password: "db1f0ee5a4d2cc1040c0958489be95e2"

wifi:
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_password
  - ssid: !secret wifi_ssid2
    password: !secret wifi_password 

      # Optional manual IP
  manual_ip:
    static_ip: 192.168.1.147
    gateway: 192.168.1.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Arduinotest Fallback Hotspot"
    password: "zr6NLwN4Bbjm"

captive_portal:
    

 `