esphome/issues #4724: Neopixelbus send my 8266 into a bootloop

DerEmder, 2023-07-20:
### The problem

If I use the neopixelbus lib to address my two RGB LEDs, then my Wemos D1 mini is sent to a boot loop instantly. This happened after updating to the latest ESPHome version. Before the update all went fine. The 8266 is still connected to my WiFi and I can perform OTA updates but the API is disconnected (of course). Logs, yaml and info below.

### Which version of ESPHome has the issue?

2023.7.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2023.7.2 Supervisor 2023.07.1 Operating System 10.3 Frontend 20230705.1 - latest

### What platform are you using?

ESP8266

RIGeek, 2023-07-22:
Same issue here and all of my devices that use neopixelbus.

ssieb, 2023-07-22:
I don't know about the others posters, but the problem with the original posted config is that the switch is initializing and trying to control the light before the light is initialized.  Try using `restore_mode: DISABLED` on the switch and let the `on_boot:` (that you already have) handle setting the state.

DerEmder, 2023-07-22:
Hi ssieb,
I will try that but then _restore_mode_ functionality is completely unusable. Anyway, I will report back later.

RIGeek, 2023-07-22:
I set my restore to disabled but I'm still stuck in a boot loop. Here is my code:

`substitutions:
  device_name: office-desk-light
  device_description: Office Desk Lighting
  friendly_name: Office Desk Light
  restore_state: DISABLED
  led_count: "82"
  led_pin: GPIO3
  button_pin: D2
  encoder_pin1: D3
  encoder_pin2: D4
#Try to restore light state after reboot/power-loss event.
#RESTORE_DEFAULT_OFF (Default) - Attempt to restore state and default to OFF if not possible to restore. Uses flash write cycles.
#RESTORE_DEFAULT_ON - Attempt to restore state and default to ON. Uses flash write cycles.
#ALWAYS_OFF - Always initialize the pin as OFF on bootup. Does not use flash write cycles.
#ALWAYS_ON - Always initialize the pin as ON on bootup. Does not use flash write cycles.

esphome:
  name: ${device_name}
  comment: ${device_description}

esp8266:
  board: d1_mini
  restore_from_flash: true

preferences:
  flash_write_interval: 15min

logger:

api:

ota:
  password: "1234"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: ${friendly_name}
    password: "1234"

captive_portal:

switch:
  - platform: template
    name: ${friendly_name} Half
    id: half_brightness_switch
    lambda: |-
      if (id(desk_strip).current_values.is_on()) {
        return true;
      } else {
        return false;
      }
    turn_on_action:
      - light.turn_on:
          id: "desk_strip"
          brightness: 60% 
          red: 0%
          green: 0%
          blue: 0%
          white: 100%
      - light.turn_on:
          id: "desk_strip"
          effect: "Wipe In"
      - delay: 2s
      - light.turn_on:
          id: "desk_strip"
[... 341 lines cut ...]
3fffff20:  feefeffe feefeffe feefeffe feefeffe  
3fffff30:  feefeffe feefeffe feefeffe feefeffe  
3fffff40:  feefeffe feefeffe 3ffffdf0 00000004  
3fffff50:  3fff3f04 3fff3f24 feefeffe feefeffe  
3fffff60:  3ffffef0 3ffef964 3fff17bc 3fff3d24  
3fffff70:  3ffe934a 3fff3ebc 00000005 3fff3424  
3fffff80:  3fffff50 feefeffe feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3ffefd6c  
3fffffa0:  3fffdad0 00000000 3ffefd58 40229924  
3fffffb0:  feefeffe feefeffe 3ffe85d8 40100669  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00073cf0
~ld `

RIGeek, 2023-07-22:
For my error. It seems I can keep in my restore mode but one of my effects seems to cause the crash. 

      - addressable_lambda:
          name: "Wipe Out"
          update_interval: 12ms
          lambda: |-
            static int x = 0;
            if (initial_run) {
              x = it.size();
            }
            if (x > 0) {
              x -= 1; 
              it[x] = ESPColor::BLACK;
            }

DerEmder, 2023-07-22:
> `substitutions: device_name: office-desk-light device_description: Office Desk Lighting friendly_name: Office Desk Light restore_state: DISABLED led_count: "82" led_pin: GPIO3 button_pin: D2 [...]

Wasn't it _restore_mode_ and not _restore_state_ ?

DerEmder, 2023-07-22:
> I don't know about the other posters, but the problem with the original posted config is that the switch is initializing and trying to control the light before the light is initialized. Try using `restore_mode: DISABLED` on the switch and let the `on_boot:` (that you already have) handle setting the state.

I used to program the ESP32 and ESP8266 nativeley back in the day and normally **all** hardware is initialized in the _setup_ part **before** any hardware access is done (and possible). It would be very weird if the devs of ESPHome did not follow this procedure. In other words: the bootcode which is either a _value restore_ or a _on_boot_ call must have the light function initialized before the call - there is no way you can make a cold call without the library initialized. AFAIK.I don't know if this is any different in ESPHome code.

RIGeek, 2023-07-22:
It was restore_state that I had set to DISABLED after reading this thread but that still resulted in a boot loop. My code does not have any on_boot function so that did not apply to me. I found that I have an effect that uses an addressable_lambda function that was resulting in the bootloop. I've not been able to figure out why but the section of code that I had to comment out (for now) to get the devices to boot was the following:

` - addressable_lambda:
      name: "Wipe Out"
      update_interval: 12ms
      lambda: |-
        static int x = 0;
        if (initial_run) {
          x = it.size();
        }
        if (x > 0) {
          x -= 1; 
          it[x] = ESPColor::BLACK;
        } `

If you see anything within that function wrong or that could be better, I'm open to ideas. I believe I copy+pasted that from some other code I found in another project. I cannot make much sense of it.

DerEmder, 2023-07-22:
>   static int x = 0;
>   if (initial_run) {

What is _initial_run_ ?

>   x = it.size();

x is now the number of LEDs you want to address

>   if (x > 0) {
>   x -= 1;

It counts back by one the number of LEDs until it is 0

>   it[x] = ESPColor::BLACK;

The LED address with index x is then set to black i.e. switched off. If you comment out the  _it[x] = ESPColor::BLACK;_ line it should at least not crashing because thet's the only place where something is actually writen to the LEDs. 

Maybe you could try and if it's the case then the problem is inside the neopixel library (or how parameters are written). In my case the boot loop is an exception 29 and something is wrongfully written to excvaddr=0x00000090 which is, in return, resetting the controller.

I guess only the devs know what is going on here.

RIGeek, 2023-07-22:
I have modified it to the following for testing but still a boot loop when loading this function:

      - addressable_lambda:
          name: "Wipe Out"
          update_interval: 12ms
          lambda: |-
            static int x = 0;
            if (initial_run) {
              x = it.size();
            }
            if (x > 0) {
              x -= 1; 
            }

RIGeek, 2023-07-22:
I have essentially the opposite effect to this that gives no issue. That code, for reference is:

          lambda: |-
            static int x = 0;
            if (initial_run) {
              x = 0;
              it.all() = Color::BLACK;
            }
            if (x < it.size()) {
              it[x] = current_color;
              x += 1;
            }

This has no issue but the "Wipe Out" causes a bootloop.

DerEmder, 2023-07-22:
Hmm, that's confusing. I'm afraid we have to wait for the devs to provide an update since the problem seems to be inside the library itself.

DerEmder, 2023-07-25:
Hmm, it's funny - if you need an update urgently then there is none but if you don't want or need any you get 'em twice a day.

DerEmder, 2023-08-01:
I can confirm: no more boot loop anymore with latest update ...7.1!

Sofwtare is working again. Thanks devs :)