esphome/issues #4727: ESPhome 2023.7.0 no working with SPI Display 1306

CjonesLAB, 2023-07-20:
### The problem

Update to 2023.7.0 and the NodeMCU blinking only Blue and not working.
With basis code without SPI Display in the Code, working the ESP Chip

### Which version of ESPHome has the issue?

2023.7.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.7.2

### What platform are you using?

ESP8266

### Board

NodeMCU 8266

### Component causing the issue

SPI

### Example YAML snippet

```yaml
esphome:
  name: wetterstation-schlafzimmer

esp8266:
  board: nodemcuv2
logger:
api:
ota:
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: "Wetterstation-Schlafzimmer"
    password: "E7TE90zuyCxr"

captive_portal:

web_server:
  port: 80
  

    
sensor:
  - platform: dht
    pin: D2
    temperature:
      name: "Schlafzimmer Temperatur"
      id: "temp_sz"
      filters:
        - calibrate_linear:
            - 0.0 -> 0.0
            - 24.4 -> 21.7
    humidity:
      name: "Schlafzimmer Luftfeuchtigkeit"
      id: "feuchte_sz"
    update_interval: 10s

  - platform: homeassistant
    id: temp_aussen
    entity_id: sensor.temp_aussen

time:
  - platform: homeassistant
    id: homeassistant_time
    
sun:
  latitude: 52.3923°
  longitude: 13.5170°
  on_sunrise:
    - then:
        - light.turn_on:
            id: oled_light
  on_sunset:
    - then:
        - light.turn_off:
            id: oled_light

font:
  - file: 'arial.ttf'
    id: font1
    size: 20
    glyphs:
          ['&', '@', '!', ',', '.', '?', '"', '%', '(', ')', '+', '-', '_', ':', '°', '0',
           '1', '2', '3', '4', '5', '6', '7', '8', '9', 'A', 'B', 'C', 'D', 'E',
          'F', 'G', 'H', 'I', 'J', 'K', 'L', 'M', 'N', 'O', 'P', 'Q', 'R', 'S',
          'T', 'U', 'V', 'W', 'X', 'Y', 'Z', ' ', 'a', 'b', 'c', 'd', 'e', 'f',
           'g', 'h', 'i', 'j', 'k', 'l', 'm', 'n', 'o', 'p', 'q', 'r', 's', 't',
           'u', 'v', 'w', 'x', 'y', 'z','å', 'Ä', 'ä', 'Ö', 'ö', 'Ü', 'ü', 'ß', '/']
  - file: 'arial.ttf'
    id: font2
    size: 12
    glyphs:
          ['&', '@', '!', ',', '.', '?', '"', '%', '(', ')', '+', '-', '_', ':', '°', '0',
           '1', '2', '3', '4', '5', '6', '7', '8', '9', 'A', 'B', 'C', 'D', 'E',
           'F', 'G', 'H', 'I', 'J', 'K', 'L', 'M', 'N', 'O', 'P', 'Q', 'R', 'S',
           'T', 'U', 'V', 'W', 'X', 'Y', 'Z', ' ', 'a', 'b', 'c', 'd', 'e', 'f',
           'g', 'h', 'i', 'j', 'k', 'l', 'm', 'n', 'o', 'p', 'q', 'r', 's', 't',
           'u', 'v', 'w', 'x', 'y', 'z','å', 'Ä', 'ä', 'Ö', 'ö', 'Ü', 'ü', 'ß', '/']
  - file: 'arial.ttf'
    id: font3
    size: 16
    glyphs:
          ['&', '@', '!', ',', '.', '?', '"', '%', '(', ')', '+', '-', '_', ':', '°', '0',
           '1', '2', '3', '4', '5', '6', '7', '8', '9', 'A', 'B', 'C', 'D', 'E',
           'F', 'G', 'H', 'I', 'J', 'K', 'L', 'M', 'N', 'O', 'P', 'Q', 'R', 'S',
           'T', 'U', 'V', 'W', 'X', 'Y', 'Z', ' ', 'a', 'b', 'c', 'd', 'e', 'f',
           'g', 'h', 'i', 'j', 'k', 'l', 'm', 'n', 'o', 'p', 'q', 'r', 's', 't',
           'u', 'v', 'w', 'x', 'y', 'z','å', 'Ä', 'ä', 'Ö', 'ö', 'Ü', 'ü', 'ß', '/']
          
  - file: 'arial.ttf'
    id: font4
    size: 18
    glyphs:
          ['&', '@', '!', ',', '.', '?', '"', '%', '(', ')', '+', '-', '_', ':', '°', '0',
           '1', '2', '3', '4', '5', '6', '7', '8', '9', 'A', 'B', 'C', 'D', 'E',
           'F', 'G', 'H', 'I', 'J', 'K', 'L', 'M', 'N', 'O', 'P', 'Q', 'R', 'S',
           'T', 'U', 'V', 'W', 'X', 'Y', 'Z', ' ', 'a', 'b', 'c', 'd', 'e', 'f',
           'g', 'h', 'i', 'j', 'k', 'l', 'm', 'n', 'o', 'p', 'q', 'r', 's', 't',
           'u', 'v', 'w', 'x', 'y', 'z','å', 'Ä', 'ä', 'Ö', 'ö', 'Ü', 'ü', 'ß', '/']
image:
  - file: "temp.gif"
    id: thermometer
  - file: "trash.gif"
    id: trash
    
spi:
  clk_pin: D5 
  mosi_pin: D7 
display:
  - platform: ssd1306_spi
    model: "SSD1306 128x64"
    id: my_display
    cs_pin: D8 
    dc_pin: D6 
    reset_pin: D1 
    pages:
      - id: page1
        lambda: |-
          it.printf(1, 0, id(font1), "%.1f°C", id(temp_sz).state);
          it.printf(70, 0, id(font1), "%.1f%%", id(feuchte_sz).state);
          it.printf(4, 22, id(font2), "Außentemperatur");
          it.printf(1, 33, id(font3), "%.1f°C", id(temp_aussen).state);
          it.strftime(56, 45, id(font4), "%X", id(homeassistant_time).now());
switch:
  - platform: template
    name: "Oled Backlights"
    id: oled_backlight
    assumed_state: true
    optimistic: true
    turn_on_action:
      then:
        - lambda: |-
            id(my_display).turn_on();
            id(my_display).fill(COLOR_ON);
    turn_off_action:
      then:
        - lambda: |-
            id(my_display).turn_off();
            id(my_display).fill(COLOR_OFF);
            
light:
  - platform: monochromatic
    name: "OLED Contrast"
    output: oled_bc
    id: oled_light
   
output:
  - platform: template
    id: oled_bc
    type: float
    write_action:
      then:
        - lambda: id(my_display).set_contrast(state);
```

ssieb, 2023-07-21:
Can you get serial logs?

CjonesLAB, 2023-07-21:
> 
Flashed the esp chip again but this time with DEV version 8.0.
The problem remains the same.
But for finding the problem I thought it would be important.
Unfortunately, there isn't much you can do with this log.

INFO ESPHome 2023.8.0-dev
INFO Reading configuration /config/esphome/wetterstation-schlafzimmer.yaml...
INFO Detected timezone 'Europe/Berlin'
INFO Starting log output from wetterstation-schlafzimmer.local using esphome API
WARNING Can't connect to ESPHome API for wetterstation-schlafzimmer.local: Error connecting to ('192.168.1.118', 6053): [Errno 111] Connect call failed ('192.168.1.118', 6053) (SocketAPIError)
INFO Trying to connect to wetterstation-schlafzimmer.local in the background

Log from ESPHome flasher:
Using 'COM5' as serial port.
Unexpected error: Cannot configure port, something went wrong. Original message: PermissionError(13, 'Ein an das System angeschlossenes Gerät funktioniert nicht.', None, 31)

I hope that, despite the little information, it narrows down the problem

ssieb, 2023-07-21:
You need to get the serial logs when it's not working.  The working logs won't help.  Try using the web installer to get the logs instead of esphome-flasher.

z3liff, 2023-08-05:
FYI, I found with the recent updates (2023.7.+), any code that calls the update() method in a lambda causes the device to go into a bootloop / panic. verified with my own device -- applies to I2C variants as well.  Same code worked perfectly before 2023.7.x