esphome/issues #4728: ESP 2023.7.0 failed config

grssll, 2023-07-21:
### The problem

After updating to 7.0 having following issue: can not update, download logs, download yaml.
INFO ESPHome 2023.7.0
INFO Reading configuration /config/esphome/esphome-web-34f45c.yaml...

Failed config

image: None
  
  'file' is a required option for [0].
  []
required key not provided

### Which version of ESPHome has the issue?

2023.7.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

esp32dev

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.7.0
INFO Reading configuration /config/esphome/esphome-web-34f45c.yaml...

Failed config

image: None
  
  'file' is a required option for [0].
  []
required key not provided
```

ssieb, 2023-07-21:
You didn't provide the config.

chanchal1987, 2023-07-31:
I also faced the same issue with animation. I needed to add one dummy image to compile my old yaml.

grssll, 2023-08-01:
Same on 2023.7.0.1 failed to install.
Failed config
image: None
  'id' is a required option for [0].
  []
required key not provided
My cofing file below:
```
substitutions:
  name: "Redia Centre" # Location and Entity Names (Capitals and Spaces Allowed)
  max_temp: "65" # Threshold for MAX Fan Speed and Alarm

esphome:
  name: "fan-redia-centre"
esp32:
  board: esp32dev
  framework:
    type: arduino

preferences:
  flash_write_interval: 10000s

# Enable logging
logger:
  logs:
    dallas.sensor: none
    number: none
    sensor: none
    light: none
    fan: none
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    

ota:
 

wifi:
  ssid: 
  password: 
  # Optional manual IP
  

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "
    password: "
    
captive_portal:

spi:
  mosi_pin: GPIO23
  clk_pin: GPIO18
  miso_pin: GPIO19

display:
  - platform: ili9xxx
    model: TFT 2.4
    cs_pin: GPIO5
    dc_pin: GPIO32
    reset_pin: GPIO33
    rotation: 270
    lambda: |-
      it.printf(0, 20, id(bold), Color(255, 0, 0), " %.1f", id(temp_sensor_1).state);
      it.print(250, 20, id(bold), Color(255, 255, 255), "/");
      it.printf(300, 20, id(bold), Color(0, 0, 255), " %.1f", id(rpm_1).state);
      auto red = Color(255, 0, 0);
[... 208 lines cut ...]
    output: gp8403_1
    gamma_correct: 1.2
  

output:
  - platform: ledc
    id: fan_pwm
    pin:
      number: GPIO4
    min_power: 0.05 #Slowest Speed for Noctua Fans is 5% - Set to 0 for 'Other' Fans
    frequency: 22.5kHz #Pulse the fan fast to prevent noise
    
  - platform: gp8403
    channel: 0
    id: gp8403_0
  - platform: gp8403
    channel: 1
    id: gp8403_1
  
  - platform: ledc
    id: d4_light_pwm
    pin:
      number: GPIO16
      inverted: true
```

ssieb, 2023-08-01:
For now, add a small image to your config to fix it.

closed 2023-08-01 by ssieb