esphome/issues #4729: ESPhome 2023.7.0 compile with api/services failed

WestYao666, 2023-07-21:
### The problem

I met a failure while compiling with api/services component on HA add-on. I thought the add-on was going wrong, I used pip to install a another ESPHome to compile it, but got a same result. If I remove the api/services component , it compiles successfully.

### Which version of ESPHome has the issue?

ESPhome 2023.7.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

esp32-c3-devkitm-1

### Component causing the issue

api/services

### Example YAML snippet

```yaml
esphome:
  name: esp32-environment-monitor
  friendly_name: esp32-Environment-monitor
  platformio_options:
    board_build.flash_mode: dio

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:

remote_transmitter:
  pin: GPIO8
  carrier_duty_percent: 50%

# Enable Home Assistant API
api:
  encryption:
    key: ""

  services:
    - service: ir_send_raw_command
      variables:
        command: int[]
      then:
        - remote_transmitter.transmit_raw:
            carrier_frequency: 38kHz
            code: !lambda 'return command;'

ota:
  password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-Environment-Monitor"
    password: !secret AP_password

captive_portal:

i2c:
  sda: GPIO04
  scl: GPIO05

sensor:
  - platform: hdc1080
    temperature:
      name: "Living Room Temperature"
    humidity:
      name: "Living Room Humidity"
    update_interval: 5s

  - platform: ccs811
    eco2:
      name: "Living Room eCO2 Value"
    tvoc:
      name: "Living Room Total Volatile Organic Compound"
    address: 0x5A
    baseline: 0x66AD
    update_interval: 0.5s

  - platform: bme280
    temperature:
      name: "BME280 Temperature"
      id: bme280_temperature
    pressure:
      name: "BME280 Pressure"
      id: bme280_pressure
    humidity:
      name: "BME280 Relative Humidity"
      id: bme280_humidity
    address: 0x76
    update_interval: 5s
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.7.0
INFO Reading configuration /config/esphome/esp32-environment-monitor.yaml...
WARNING GPIO8 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing esp32-environment-monitor (board: esp32-c3-devkitm-1; framework: arduino; platform: platformio/espressif32@5.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Wire @ 2.0.0
|-- noise-c @ 0.1.4
Compiling /data/esp32-environment-monitor/.pioenvs/esp32-environment-monitor/src/main.cpp.o
Linking /data/esp32-environment-monitor/.pioenvs/esp32-environment-monitor/firmware.elf
/data/cache/platformio/packages/toolchain-riscv32-esp@8.4.0+2021r2-patch5/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /data/esp32-environment-monitor/.pioenvs/esp32-environment-monitor/src/main.cpp.o: in function `esphome::api::ListEntitiesServicesArgument::ListEntitiesServicesArgument()':
/config/esphome/.esphome/build/esp32-environment-monitor/src/esphome/components/api/api_pb2.h:836: undefined reference to `esphome::api::enums::ServiceArgType esphome::api::to_service_arg_type<std::vector<long, std::allocator<long> > >()'
collect2: error: ld returned 1 exit status
*** [/data/esp32-environment-monitor/.pioenvs/esp32-environment-monitor/firmware.elf] Error 1
========================== [FAILED] Took 8.42 seconds ==========================
```

evilone68, 2023-07-24:
I have the same sort of issue but had worked to update via the notification page, not the esphome page but now nothing works at all, I now get " *** [/data/ld2410/.pioenvs/ld2410/src/esphome/components/api/api_pb2.cpp.o] Error 1" . on all of my boards but the reason could change so some its API, web_server, src, captive_portal, but all are Error1.

ssieb, 2023-07-24:
@evilone68 what is the reason for the error?

WestYao666, 2023-07-25:
> 

I think the same as the issue.

CarlosGS, 2023-11-21:
There is a temporary workaround in this thread https://github.com/esphome/issues/issues/3564#issuecomment-1821603239