esphome/issues #4730: display full_update_every not working after uograde from 2023.6.3 to 2023.7.0

Vulle66, 2023-07-21:
### The problem

I have 3 InkPlate6 displays running, after upgrading to ESPHome version 2023.7.0 the full_update_every is ignored. Before updating i was running with partial update every 10s and full_update_every 600 giving full update every 100 minutes. On version the full update is every 10s.

### Which version of ESPHome has the issue?

2023.7.o

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.7.2

### What platform are you using?

ESP32

### Board

Esp-wrover-kit

### Component causing the issue

Display 

### Example YAML snippet

```yaml
display:
- platform: inkplate6
  id: inkplate_display
  greyscale: false
  partial_updating: true
  update_interval: 10s
  full_update_every: 600
```

ghettodev, 2023-07-23:
Can confirm. Seeing the same issues.

alrouen, 2023-07-31:
Same issue

Vulle66, 2023-09-01:
I solved my issue by using the esphome/esphome docker image version 2023.06.05, to build the code for my inkplate modules. I just started the docket image which setup a web ui, form here you can setup esphome configurations. I just copied the config from my home assistant to the docker web UI. This made it possible to recompile the config using ESPHome 2023.6.3, and do wireless update of the modules. I did not change or remove the config from HA. So new i run ESP Home 2023.6.5 on the inkplate 6, and the latest version on other modules. The inkplate6 is now running with partial update, and only full update as configured. I know this will keep my inkplate on version ESP Home 2023.6.3, and all other ESP Home on newer versions, but this is better than not having partial update. I hope this will keep running until the issue is fixed in the latest ESP Home

alrouen, 2024-08-04:
Well, the issue is still present (just tried with esphome-2024.7.3), so I will roll back my device again to 2023.6.5

solarkennedy, 2024-10-23:
I'm unable to reproduce, adding in some logging I think the actual mechanism for full_update_every is working for me on latest:
```
[13:55:58][V][inkplate:149]: Setting block_partial_ to true, num partial updates: 7 and full_update_every_ is 7
[13:56:02][V][inkplate:319]: Display finished (full) (4180ms)
[13:56:03][V][inkplate:314]: Display finished (partial) (668ms)
[13:56:12][V][inkplate:314]: Display finished (partial) (669ms)
[13:56:22][V][inkplate:314]: Display finished (partial) (663ms)
[13:56:32][V][inkplate:314]: Display finished (partial) (662ms)
[13:56:42][V][inkplate:314]: Display finished (partial) (670ms)
[13:56:52][V][inkplate:314]: Display finished (partial) (664ms)
[13:57:02][V][inkplate:314]: Display finished (partial) (670ms)
[13:57:11][V][inkplate:149]: Setting block_partial_ to true, num partial updates: 7 and full_update_every_ is 7
[13:57:16][V][inkplate:319]: Display finished (full) (4189ms)
[13:57:22][V][inkplate:314]: Display finished (partial) (667ms)
[13:57:32][V][inkplate:314]: Display finished (partial) (666ms)
[13:57:42][V][inkplate:314]: Display finished (partial) (660ms)
[13:57:52][V][inkplate:314]: Display finished (partial) (663ms)
[13:58:02][V][inkplate:314]: Display finished (partial) (668ms)
[13:58:12][V][inkplate:314]: Display finished (partial) (662ms)
[13:58:22][V][inkplate:314]: Display finished (partial) (666ms)
[13:58:31][V][inkplate:149]: Setting block_partial_ to true, num partial updates: 7 and full_update_every_ is 7
```

is it possible something is _calling_ `update()` much more often?
I'm not using `update_interval: 10s`, and I actually can't find display or inkplate code that uses that setting?
My updates are being triggered by the text and other sensors that are on there.

The big change related to displays that occurred for 2023.7.0 is the Display/DisplayRef refactor:
https://github.com/esphome/esphome/pull/5002/files#diff-a53f36e1c599e069b5ab469dd8c6b6c588627182284922ff811d0e7fac77e6f7
(big changelog here: https://github.com/esphome/esphome/pull/5122)