esphome/issues #4731: ESP 2023.7.0 Component dsmr took a long time for an operation (0.60 s)

Rwin1234, 2023-07-21:
### The problem

I observe a new warning in my log for my P1 dsmr reader. It occurs every 10 seconds, exactly the frequency of getting serial data from the P1 port. See below.

Note dsmr was updated in this release.

### Which version of ESPHome has the issue?

ESPHome 2023.7.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.7.3

### What platform are you using?

ESP32

### Board

ESP32-C3

### Component causing the issue

dsmr

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.7.0
INFO Reading configuration /config/esphome/p1-dongle-pro.yaml...
INFO Starting log output from p1-dongle-pro.local using esphome API
INFO Successfully connected to p1-dongle-pro.local
[00:25:44][I][app:102]: ESPHome version 2023.7.0 compiled on Jul 22 2023, 00:14:13
[00:25:44][I][app:104]: Project smartstuff.p1_dongle_pro version v23.02.1
[00:25:45][W][component:204]: Component dsmr took a long time for an operation (0.61 s).
[00:25:45][W][component:205]: Components should block for at most 20-30ms.
[00:25:55][W][component:204]: Component dsmr took a long time for an operation (0.60 s).
[00:25:55][W][component:205]: Components should block for at most 20-30ms.
[00:26:05][W][component:204]: Component dsmr took a long time for an operation (0.61 s).
[00:26:05][W][component:205]: Components should block for at most 20-30ms.

[repeats every 10 seconds]
```

ssieb, 2023-07-21:
Did you not see the pinned issue at the top of the issues page?
Duplicate of https://github.com/esphome/issues/issues/4717

closed 2023-07-21 by ssieb

Rwin1234, 2023-07-22:
Sorry, I did not see that issue, google did not find it yet.

Eric64124, 2023-09-06:
Same here on a DSMR sensor, the device has been working properly until end of june at least.
Then after an update (don't recall to which version) I started to get this error

**LOG (some text, where indicated, removed for privacy reasons):**

[22:46:26][I][app:102]: ESPHome version 2023.8.2 compiled on Aug 29 2023, 22:57:58
[22:46:26][C][wifi:543]: WiFi:
[22:46:26][C][wifi:379]:   Local MAC: 0C:B8:15:F6:4E:28
[22:46:26][C][wifi:380]:   SSID: '..removed..'[redacted]
[22:46:26][C][wifi:381]:   IP Address: 192.168.0.51
[22:46:26][C][wifi:383]:   BSSID: 3C:84:6A:..partly removed... [redacted]
[22:46:26][C][wifi:384]:   Hostname: 'slimmemeter'
[22:46:26][C][wifi:386]:   Signal strength: -77 dB ..here a nice graph that cannot be pasted was present..
[22:46:26][C][wifi:390]:   Channel: 6
[22:46:26][C][wifi:391]:   Subnet: 255.255.255.0
[22:46:26][C][wifi:392]:   Gateway: 192.168.0.1
[22:46:26][C][wifi:393]:   DNS1: ..removed...21
[22:46:26][C][wifi:394]:   DNS2: ..removed...20
[22:46:26][C][logger:301]: Logger:
[22:46:26][C][logger:302]:   Level: DEBUG
[22:46:26][C][logger:303]:   Log Baud Rate: 115200
[22:46:26][C][logger:305]:   Hardware UART: UART0
[22:46:26][C][uart.arduino_esp32:124]: UART Bus 1:
[22:46:26][C][uart.arduino_esp32:126]:   RX Pin: GPIO13
[22:46:26][C][uart.arduino_esp32:128]:   RX Buffer Size: 1800
[22:46:26][C][uart.arduino_esp32:130]:   Baud Rate: 115200 baud
[22:46:26][C][uart.arduino_esp32:131]:   Data Bits: 8
[22:46:26][C][uart.arduino_esp32:132]:   Parity: NONE
[22:46:26][C][uptime.sensor:031]:   State Class: 'total_increasing'
[22:46:26][C][uptime.sensor:031]:   Unit of Measurement: 's'
[22:46:26][C][uptime.sensor:031]:   Accuracy Decimals: 0
[22:46:26][C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
[22:46:26][D][sensor:094]: 'Energy Consumed Tariff 1': Sending state 1681.50403 kWh with 3 decimals of accuracy
[22:46:26][D][sensor:094]: 'Energy Consumed Tariff 2': Sending state 1798.13696 kWh with 3 decimals of accuracy
[22:46:26][D][sensor:094]: 'Energy Produced Tariff 1': Sending state 0.00000 kWh with 3 decimals of accuracy
[22:46:26][D][sensor:094]: 'Energy Produced Tariff 2': Sending state 0.00000 kWh with 3 decimals of accuracy
[22:46:26][D][sensor:094]: 'Power Consumed': Sending state 2032.00012 W with 0 decimals of accuracy
[22:46:26][D][sensor:094]: 'Power Produced': Sending state 0.00000 W with 0 decimals of accuracy
[22:46:26][D][sensor:094]: 'Electricity Failures': Sending state 7.00000  with 0 decimals of accuracy
[22:46:26][D][sensor:094]: 'Long Electricity Failures': Sending state 2.00000  with 0 decimals of accuracy
[22:46:27][D][sensor:094]: 'Voltage Phase 1': Sending state 233.39999 V with 1 decimals of accuracy
[22:46:27][D][sensor:094]: 'Voltage Phase 2': Sending state 233.80000 V with 1 decimals of accuracy
[22:46:27][D][sensor:094]: 'Voltage Phase 3': Sending state 231.70000 V with 1 decimals of accuracy
[22:46:27][D][sensor:094]: 'Current Phase 1': Sending state 7.00000 A with 1 decimals of accuracy
[22:46:27][D][sensor:094]: 'Current Phase 2': Sending state 0.00000 A with 1 decimals of accuracy
[22:46:27][D][sensor:094]: 'Current Phase 3': Sending state 0.00000 A with 1 decimals of accuracy
[22:46:27][D][sensor:094]: 'Power Consumed Phase 1': Sending state 1857.00000 W with 0 decimals of accuracy
[22:46:27][D][sensor:094]: 'Power Consumed Phase 2': Sending state 110.00000 W with 0 decimals of accuracy
[22:46:27][D][sensor:094]: 'Power Consumed Phase 3': Sending state 62.00000 W with 0 decimals of accuracy
[22:46:27][D][sensor:094]: 'Power Produced Phase 1': Sending state 0.00000 W with 0 decimals of accuracy
[22:46:27][D][sensor:094]: 'Power Produced Phase 2': Sending state 0.00000 W with 0 decimals of accuracy
[22:46:27][D][sensor:094]: 'Power Produced Phase 3': Sending state 0.00000 W with 0 decimals of accuracy
[22:46:27][D][sensor:094]: 'Gas Consumed': Sending state 1515.78296 mÂ³ with 3 decimals of accuracy
[22:46:27][D][text_sensor:064]: 'DSMR Identification': Sending state 'ISK5\2M550T-1013'
[22:46:27][D][text_sensor:064]: 'DSMR Version': Sending state '50'
[22:46:27][W][component:204]: Component dsmr took a long time for an operation (0.22 s).
[22:46:27][C][dsmr:286]:     Accuracy Decimals: 3
[22:46:27][C][dsmr:286]:   energy_delivered_tariff2 'Energy Consumed Tariff 2'
[22:46:27][C][dsmr:286]:     Device Class: 'energy'
[22:46:27][C][dsmr:286]:     State Class: 'total_increasing'
[22:46:27][C][dsmr:286]:     Unit of Measurement: 'kWh'
[22:46:27][C][dsmr:286]:     Accuracy Decimals: 3
[22:46:27][C][dsmr:286]:     State Class: 'total_increasing'
[22:46:27][C][dsmr:286]:     Unit of Measurement: 'kWh'
[22:46:27][C][dsmr:286]:     Accuracy Decimals: 3
[22:46:27][C][dsmr:286]:     State Class: 'total_increasing'
[22:46:27][C][dsmr:286]:   power_returned 'Power Produced'
[22:46:27][C][dsmr:286]:     Device Class: 'power'
[22:46:27][C][dsmr:286]:     State Class: 'measurement'
[22:46:27][C][dsmr:286]:     Unit of Measurement: 'W'
[... 218 lines cut ...]
#    gas_delivered_be:
#      name: "Gas Consumed Belgium"
#      state_class: total_increasing
  - platform: uptime
    name: "Uptime"
  - platform: wifi_signal
    name: "Wi-Fi Signal"
    update_interval: 60s
      
text_sensor:
  - platform: dsmr
    identification:
      name: "DSMR Identification"
    p1_version:
      name: "DSMR Version"
  - platform: wifi_info
    ip_address:
      name: "IP Address"
    ssid:
      name: "Wi-Fi SSID"
    bssid:
      name: "Wi-Fi BSSID"
  - platform: version
    name: "ESPHome Version"
    hide_timestamp: true