esphome/issues #4736: alpha3 component sensors not working

jan-hofmeier, 2023-07-22:
### The problem

When using the alpha3 component, no values are reported by the sensors, despite the pump being connected and no errors shown in the log.

### Which version of ESPHome has the issue?

2023.7.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

alpha3

### Example YAML snippet

```yaml
ble_client:
  - mac_address: "04:ee:03:bc:d8:70"
    id: radiator_pump

sensor:
  - platform: alpha3
    ble_client_id: radiator_pump
    flow:
      name: "Radiator FLOW"
```

### Additional information

It seems on some pumps the requests and responses look different than on the pumps then I have to test. So I need more information, so I can implement a generic solution, that works for every pump.

If you run into this problem, please comment with the following information
- From which country is your pump
- What is written on the type label of the pump
- does it work with the attached [alpha3_denmark.zip](https://github.com/esphome/issues/files/12137363/alpha3_denmark.zip) external comonent
- If it doesn't work with the attached external comonent a bluetooth log if you can generate one

If neither the version upstream in esphome nor the modified external component works, I need a bluetooth log of the communication of the GO Remote app with the pump, to see how the requests and responses look like in your case. On Android this is called 'HCI Snoop log' and on iOS 'bluetooth debug'.

jan-hofmeier, 2023-07-22:
@TorbenHansen @uwec I think you have the same problem. Could you try the external component and see if that helps?

TorbenHansen, 2023-07-23:
> @TorbenHansen @uwec I think you have the same problem. Could you try the external component and see if that helps?

I just tried the "alpha3_denmark" component.. And it didn't work.

But then I tried to connect the pump to the GO-app and I reset the pump to Factory-settings. Because earlier I had named the pump a different name. 

After I reset the pump to Factory settings. I get the readings fine. with the "alpha3_denmark" component

uwec, 2023-07-23:
> @TorbenHansen @uwec I think you have the same problem. Could you try the external component and see if that helps?

Very good hint. Can confirm, i had as well chnanged the name of the pump in earlier times and had no success with the "alpha3_denmark" as external component.

Did a reset of my pump the factory defaults and it works now!!!

After I configured the pump name back to my individual name and it still works?!

jan-hofmeier, 2023-07-23:
This is very interesting, thank you for the feedback. 
@TorbenHansen does your pump now also work with the upstream version of the component?

TorbenHansen, 2023-07-23:
> This is very interesting, thank you for the feedback. @TorbenHansen does your pump now also work with the upstream version of the component?

@jan-hofmeier 
Can confirm. It also works with the upstream version of the component.

MiRoCZ, 2023-11-03:
Hello guys, would you be so kind and describe step by step how to install and configure ESP for HA and Grundfos Alpha 3. I obtained Alpha MAC address. Then added code from https://esphome.io/components/sensor/alpha3.html via HA ESPHome / Edit, but getting error "Platform not found: 'sensor.alpha3"

jan-hofmeier, 2023-11-03:
Did you make sure your esphome is up to date?
Try doing a `pip install --upgrade esphome`

MiRoCZ, 2023-11-03:
> Did you make sure your esphome is up to date? Try doing a `pip install --upgrade esphome`

THX. 
[image]