esphome/issues #4742: Not able to change interrupt pin for lilygo_t5_47 touchscreen

special-k-uk, 2023-07-24:
### The problem

The lilygo T5 4.7 Touchscreen component enforces GPIO13 as the interrupt pin, as specified in the docs.
> interrupt_pin (Optional, [Pin Schema](https://esphome.io/guides/configuration-types.html#config-pin-schema)): The touch detection pin. Must be GPIO13.

This works fine for the [Lilygo T5 4.7 ESP32 v3](https://www.lilygo.cc/products/t5-4-7-inch-e-paper) boards where the interrupt is GPIO13, but not for the newer [ESP32-S3](https://www.lilygo.cc/products/t5-4-7-inch-e-paper-v2-3) board version where the interrupt is GPIO09.

I have 6 of the newer ESP32-S3 boards that the seller shipped instead of the purchased older v3 boards :-(

The Touchscreens docs specify the interrupt_pin is optional, yet if it is omitted from the yaml ESPHome warns that it IS a required option.

Setting the interrupt pin to GPIO09 in the yaml has no affect, and touches are not registered.

### Which version of ESPHome has the issue?

2023.7.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.7.3

### What platform are you using?

ESP32

### Board

Lilygo T5 4.7 ESP32-S3 v2.3

### Component causing the issue

Touchscreen

### Example YAML snippet

```yaml
touchscreen:
  - platform: lilygo_t5_47
    id: lilygo_touchscreen
    interrupt_pin: GPIO09
    address: 0x5A
    setup_priority: -100
    on_touch:
      then:
        - display.page.show_next: 't5_display'
        - component.update: 't5_display'
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.7.0
INFO Reading configuration /config/esphome/test.yaml...
INFO Detected timezone 'Europe/London'
INFO Detected timezone 'Europe/London'
INFO Starting log output from 192.168.1.51 using esphome API
INFO Successfully connected to 192.168.1.51
[20:51:14][I][app:102]: ESPHome version 2023.7.0 compiled on Jul 24 2023, 19:32:54
[20:51:14][C][wifi:543]: WiFi:
[20:51:14][C][wifi:379]:   Local MAC: 7C:DF:A1:FF:D2:2b
[20:51:14][C][wifi:380]:   SSID: [redacted]
[20:51:14][C][wifi:381]:   IP Address: 192.168.1.51
[20:51:14][C][wifi:383]:   BSSID: [redacted]
[20:51:14][C][wifi:384]:   Hostname: 'test'
[20:51:14][C][wifi:386]:   Signal strength: -69 dB ▂▄▆█
[20:51:14][V][wifi:388]:   Priority: 0.0
[20:51:14][C][wifi:390]:   Channel: 1
[20:51:14][C][wifi:391]:   Subnet: 255.255.255.0
[20:51:14][C][wifi:392]:   Gateway: 192.168.1.254
[20:51:14][C][wifi:393]:   DNS1: 192.168.1.253
[20:51:14][C][wifi:394]:   DNS2: 8.8.8.8
[20:51:14][C][logger:301]: Logger:
[20:51:14][C][logger:302]:   Level: VERBOSE
[20:51:14][C][logger:303]:   Log Baud Rate: 115200
[20:51:14][C][logger:305]:   Hardware UART: UART0
[20:51:14][C][i2c.arduino:053]: I2C Bus:
[20:51:14][C][i2c.arduino:054]:   SDA Pin: GPIO18
[20:51:14][C][i2c.arduino:055]:   SCL Pin: GPIO17
[20:51:14][C][i2c.arduino:056]:   Frequency: 50000 Hz
[20:51:14][C][i2c.arduino:059]:   Recovery: bus successfully recovered
[20:51:14][I][i2c.arduino:069]: Results from i2c bus scan:
[20:51:14][I][i2c.arduino:075]: Found i2c device at address 0x51
[20:51:14][I][i2c.arduino:075]: Found i2c device at address 0x5A
[20:51:14][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'button21'
[20:51:14][C][gpio.binary_sensor:016]:   Pin: GPIO21
[20:51:14][C][PCF8563:022]: PCF8563:
[20:51:14][C][PCF8563:023]:   Address: 0x51
[20:51:14][C][PCF8563:027]:   Timezone: 'GMT0BST,M3.5.0/1,M10.5.0'
[20:51:14][C][homeassistant.time:010]: Home Assistant Time:
[20:51:14][C][homeassistant.time:011]:   Timezone: 'GMT0BST,M3.5.0/1,M10.5.0'
[20:51:14][C][t574:053]: T547
[20:51:14][C][t574:053]:   Rotations: 180 °
[20:51:14][C][t574:053]:   Dimensions: 960px x 540px
[20:51:14][C][t574:054]:   Update Interval: 7200.0s
[20:51:14][C][captive_portal:088]: Captive Portal:
[20:51:14][C][mdns:112]: mDNS:
[20:51:14][C][mdns:113]:   Hostname: test
[20:51:14][V][mdns:114]:   Services:
[20:51:14][V][mdns:116]:   - _esphomelib, _tcp, 6053
[20:51:14][V][mdns:118]:     TXT: friendly_name = test
[20:51:14][V][mdns:118]:     TXT: version = 2023.7.0
[20:51:14][V][mdns:118]:     TXT: mac = 7cdfa1ffd228
[20:51:14][V][mdns:118]:     TXT: platform = ESP32
[20:51:14][V][mdns:118]:     TXT: board = lilygo-t5-47-plus
[20:51:14][V][mdns:118]:     TXT: network = wifi
[20:51:14][V][mdns:118]:     TXT: api_encryption = Noise_NNpsk0_25519_ChaChaPoly_SHA256
[20:51:14][C][ota:093]: Over-The-Air Updates:
[20:51:14][C][ota:094]:   Address: 192.168.1.51:3232
[20:51:14][C][ota:097]:   Using Password.
[20:51:14][C][api:138]: API Server:
[20:51:14][C][api:139]:   Address: 192.168.1.51:6053
[20:51:14][C][api:141]:   Using noise encryption: YES
[20:51:14][C][lilygo_t5_47.touchscreen:174]: Lilygo T5 47 Touchscreen:
[20:51:14][C][lilygo_t5_47.touchscreen:175]:   Address: 0x5A
[20:51:14][C][lilygo_t5_47.touchscreen:176]:   Interrupt Pin: GPIO9
```

nodacat, 2023-09-07:
Same issue here! any progress on your end?

nodacat, 2023-09-08:
FYI I have it working. Try using GPIO47 instead of GPIO09 - I'm using esphome's component.

special-k-uk, 2023-09-11:
DOH! 

Well spotted @nodacat . I don't know why I didn't check the schematics, I've worked with a few different Lilygo screens and I ALWAYS need to check them as their printed docs tend to be very wrong. 

:scratches_head: Why didn't I check them this time? Don't I feel stupid for not checking :-)

Yep, all working with GPIO47

nodacat, 2023-09-11:
Hahah oh you weren’t the only one! I was so locked in on that pinmap graphic from lilygo, many others mistook the RTC pins for the touchscreen, which lilygo left off this time oddly. Glad I could help!

special-k-uk, 2023-09-11:
Yeah, sometimes you just go so far down the rabbit hole, that the obvious just isn't obvious anymore! 

I had the same issue with the TTGO T5s 2.7". Cracking little board (used them for air quality sensors), but the silk screening and docs are a mess. I'd swear the schematics were for a different board compared to the AliExpress listing!

Anyways, the WAF has significantly increased now these 6 screens are touchable again!

special-k-uk, 2023-09-11:
@nodacat . Probably a dumb question, as it's been a really long day, I'm looking at the T5-4.7-plus.pdf from [here ](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/tree/master/schematic). Where abouts did you pick up on GPIO47 for the interrupt?

nodacat, 2023-09-11:
@special-k-uk  check out section B4 (using the row/column headers around the pdf). There's a 6 pin connector (only 6 pin connector on the physical board) and to the left a pinout of INT/SDA/SCL.  Match that pinout to the circuit diagram and you'll see INT=>47.   You'll find a similar 6 pin connector on the other older [T5-4.7.pdf ](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/master/schematic/T5-4.7.pdf).  Comparing the two, you'll see the change from GPIO13 to GPIO47 (as well as the SDA/SCL changes).

special-k-uk, 2023-09-11:
Ah, got it. It's the big red letters I didnt see. Thanks for your help with this one, really appreciated

closed 2023-09-11 by special-k-uk

nodacat, 2023-09-11:
Haha no problem, it wasn’t super obvious that that was related to the touchscreen connector at first, I only figured that out by process of elimination with the other header/connectors when inspecting the board. But anyway, cheers man! Glad I could save someone else from further headache… now on to the rest of the yaml lol