esphome/issues #4743: 'ESPTime' is not a member of 'esphome::time' error in external component BM8563

gsgxnet, 2023-07-24:
### The problem

The external component sensor BM8563 from the [jesserockz m5paper](https://github.com/jesserockz/m5paper-esphome) repo can not be used any more. Compilation fails, see log below. Reason same as in 
[esphome issue 4617](https://github.com/esphome/issues/issues/4617)

I am reporting this issue here, as there is no issue reporting option at the external component repository

I do not have enough c++ and esphome knowledge to fork and fix myself

### Which version of ESPHome has the issue?

2023.7.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.7.3

### What platform are you using?

ESP32

### Board

M5stack core2

### Component causing the issue

BM8563

### Example YAML snippet

```yaml
external_components:
  - source: github://gpambrozio/esphome@FT6336U-touch
    components: [ ft63x6 ]
  - source: github://martydingo/esphome-axp192  # https://github.com/martydingo/esphome-axp192
    components: [axp192]
  - source:
      type: git
      url: https://github.com/jesserockz/m5paper-esphome
      ref: main
    components: [ bm8563 ]  

...

time:
  - platform: homeassistant
    id: esptime
    on_time_sync: 
      - bm8563.write_time
  - platform: bm8563
    i2c_id: bus_a
    address: 0x51
    id: rtc_time
    sleep_duration: 3600000ms
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.7.0
INFO Reading configuration /config/esphome/m5t3d1.yaml...
INFO Cloning https://github.com/jesserockz/m5paper-esphome@main
INFO Detected timezone 'Europe/Dublin'
INFO Detected timezone 'Europe/Dublin'
INFO Detected timezone 'Europe/Dublin'
WARNING GPIO15 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO0 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO12 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO2 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING Pin 10 (9-10) might already be used by the flash interface in QUAD IO flash mode.
WARNING GPIO2 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing m5t3d1 (board: m5stack-core2; framework: arduino; platform: platformio/espressif32@5.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 4.31MB RAM, 16MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|-- SPI @ 2.0.0
|-- Wire @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
|-- HTTPClient @ 2.0.0
|-- ESP32-audioI2S @ 2.0.7
Compiling /data/m5t3d1/.pioenvs/m5t3d1/src/esphome/components/bm8563/bm8563.cpp.o
src/esphome/components/bm8563/bm8563.cpp: In member function 'void esphome::bm8563::BM8563::read_time()':
src/esphome/components/bm8563/bm8563.cpp:84:9: error: 'ESPTime' is not a member of 'esphome::time'
   time::ESPTime rtc_time{.second = uint8_t(BM8563_TimeStruct.seconds),
         ^~~~~~~
src/esphome/components/bm8563/bm8563.cpp:84:9: note: suggested alternative:
In file included from src/esphome/components/time/real_time_clock.h:8,
                 from src/esphome/components/bm8563/bm8563.h:5,
                 from src/esphome/components/bm8563/bm8563.cpp:3:
src/esphome/core/time.h:12:8: note:   'esphome::ESPTime'
 struct ESPTime {
        ^~~~~~~
src/esphome/components/bm8563/bm8563.cpp:93:3: error: 'rtc_time' was not declared in this scope
   rtc_time.recalc_timestamp_utc(false);
   ^~~~~~~~
src/esphome/components/bm8563/bm8563.cpp:93:3: note: suggested alternative: 'write_time'
   rtc_time.recalc_timestamp_utc(false);
   ^~~~~~~~
   write_time
Compiling /data/m5t3d1/.pioenvs/m5t3d1/src/main.cpp.o
*** [/data/m5t3d1/.pioenvs/m5t3d1/src/esphome/components/bm8563/bm8563.cpp.o] Error 1
```

ssieb, 2023-07-24:
Use this repo instead: https://github.com/paveldn/m5paper_esphome

closed 2023-07-24 by ssieb

gsgxnet, 2023-07-25:
Sorry but that repo is no useful as it is.  
As that repo is not set up to be used within a yaml definition as it does not contain a `components` or an `esphome` folder. So including it, I get an error message from the linter:
```
Could not find components folder for source. Please check the source contains a 'components' or 'esphome/components' folder.
```

gsgxnet, 2023-07-25:
To really close this issue I have forked the recommend repo and modified it.
Created a new branch named `cc2c`.
In this branch there is a folder named `components`, which is a plain copy of the `custom_components` folder.

Now the external components can be used as follows:

```
external_components:
  - source:
      type: git
      #url: https://github.com/jesserockz/m5paper-esphome  
      #url: https://github.com/paveldn/m5paper_esphome
      url: https://github.com/gsgxnet/m5paper_esphome
      ref: cc2c
    components: [ bm8563 ] 
```

It is important to use the `ref:` declaration here, as the main branch is unmodified and still does not contain a `components` folder.  

The `bm8563` time device for my M5TOUGH is defined as:

```
time:
  - platform: bm8563
    i2c_id: bus_a
    address: 0x51
    id: rtc_time
    sleep_duration: 3600000ms
```