esphome/issues #4744: class esphome::wifi::WiFiComponent' has no member named 'get_dns_address'; did you mean 'get_ip_address'?

zibous, 2023-07-25:
### The problem

Compile Error:
```
compiling .pioenvs/devkit4-test/src/esphome/core/component_iterator.cpp.o
In file included from src/esphome/components/wifi_info/wifi_info_text_sensor.cpp:1:
src/esphome/components/wifi_info/wifi_info_text_sensor.h: In member function 'virtual void esphome::wifi_info::DNSAddressWifiInfo::update()':
src/esphome/components/wifi_info/wifi_info_text_sensor.h:32:49: error: 'class esphome::wifi::WiFiComponent' has no member named 'get_dns_address'; did you mean 'get_ip_address'?
     auto dns_one = wifi::global_wifi_component->get_dns_address(0);
                                                 ^~~~~~~~~~~~~~~
                                                 get_ip_address
src/esphome/components/wifi_info/wifi_info_text_sensor.h:33:49: error: 'class esphome::wifi::WiFiComponent' has no member named 'get_dns_address'; did you mean 'get_ip_address'?
     auto dns_two = wifi::global_wifi_component->get_dns_address(1);
                                                 ^~~~~~~~~~~~~~~
                                                 get_ip_address
*** [.pioenvs/devkit4-test/src/esphome/components/wifi_info/wifi_info_text_sensor.cpp.o] Error 1
======================================================================= [FAILED] Took 12.27 seconds
```

### Which version of ESPHome has the issue?

ESPHome 2023.7.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

az-delivery-devkit-v4

### Component causing the issue

platform: wifi_info

### Example YAML snippet

```yaml
## ------------------------------------------------------------------
## WIFI TESTCASE
## ------------------------------------------------------------------
substitutions:
  ## device settings
  device_name_short: "devkit4-test"
  device_description: "WIFI Testcase"
  projectname: "devkit4.testcase"
  friendly_name: "DEVKIT4 TESTCASE"
  appversion: "1.0.0"
  domain: !secret domain

## -----------------------------------------------------------------
## HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
## -----------------------------------------------------------------
esp32:
  board: az-delivery-devkit-v4

## ----------------------------------------------------------------
## APPLICATION ESPHOME
## ----------------------------------------------------------------
esphome:
  name: "${device_name_short}"
  comment: ${device_description}
  build_path: ./build/${device_name_short}
  project:
    name: ${projectname}
    version: ${appversion}

## ---------------------------------------------------
## LOGGER COMPONENT
## ---------------------------------------------------
logger:
  
## ---------------------------------------------------
## WIFI Settings
## ---------------------------------------------------
wifi:
  ssid: !secret ssid4_name
  password: !secret ssid4_pswd
  domain: !secret domain
  
## ---------------------------------------------------
## SENSORS
## ---------------------------------------------------
sensor:

  # Wifi quality RSSI (%)
  - platform: wifi_signal
    name: "${friendly_name}WLAN Signal"
    id: wifi_signal_db
    update_interval: 60s
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    entity_category: "diagnostic"
    unit_of_measurement: "%"

  # Device uptime in hours
  - platform: uptime
    name: "Device Uptime"
    filters:
      - lambda: return x / 3600;
    unit_of_measurement: "h"
    entity_category: "diagnostic"
    state_class: "measurement"
    accuracy_decimals: 2
    icon: mdi:clock-start

## ---------------------------------------------------
## TEXT SENSOR
## ---------------------------------------------------
text_sensor:

  # connected to ssid
  - platform: wifi_info
    ssid:
      name: ${friendly_name} WLAN SSID
      id: wlan_ssid
      entity_category: "diagnostic"
      disabled_by_default: true
      icon: mdi:wifi-settings

  # Device application version
  - platform: version
    name: ${friendly_name} Version
    hide_timestamp: true
    entity_category: "diagnostic"
    id: appver
    icon: mdi:application-cog-outline
    disabled_by_default: true
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.7.0
INFO Reading configuration ble_monitor_test.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Processing devkit4-test (board: az-delivery-devkit-v4; framework: arduino; platform: platformio/espressif32@5.3.0)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 520KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
Compiling .pioenvs/devkit4-test/src/esphome/components/esp32/core.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/components/esp32/gpio.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/components/esp32/preferences.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/components/logger/logger.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/components/mdns/mdns_component.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/components/mdns/mdns_esp32.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/components/mdns/mdns_host.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/components/mdns/mdns_rp2040.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/components/network/util.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/components/sensor/automation.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/components/sensor/filter.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/components/sensor/sensor.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/components/text_sensor/filter.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/components/text_sensor/text_sensor.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/components/uptime/uptime_sensor.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/components/version/version_text_sensor.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/components/wifi/wifi_component.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/components/wifi/wifi_component_esp32_arduino.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/components/wifi/wifi_component_esp8266.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/components/wifi/wifi_component_esp_idf.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/components/wifi/wifi_component_pico_w.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/components/wifi_info/wifi_info_text_sensor.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/components/wifi_signal/wifi_signal_sensor.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/core/application.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/core/color.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/core/component.cpp.o
Compiling .pioenvs/devkit4-test/src/esphome/core/component_iterator.cpp.o
In file included from src/esphome/components/wifi_info/wifi_info_text_sensor.cpp:1:
src/esphome/components/wifi_info/wifi_info_text_sensor.h: In member function 'virtual void esphome::wifi_info::DNSAddressWifiInfo::update()':
src/esphome/components/wifi_info/wifi_info_text_sensor.h:32:49: error: 'class esphome::wifi::WiFiComponent' has no member named 'get_dns_address'; did you mean 'get_ip_address'?
     auto dns_one = wifi::global_wifi_component->get_dns_address(0);
                                                 ^~~~~~~~~~~~~~~
                                                 get_ip_address
src/esphome/components/wifi_info/wifi_info_text_sensor.h:33:49: error: 'class esphome::wifi::WiFiComponent' has no member named 'get_dns_address'; did you mean 'get_ip_address'?
     auto dns_two = wifi::global_wifi_component->get_dns_address(1);
                                                 ^~~~~~~~~~~~~~~
                                                 get_ip_address
*** [.pioenvs/devkit4-test/src/esphome/components/wifi_info/wifi_info_text_sensor.cpp.o] Error 1
======================================================================= [FAILED] Took 12.27 seconds =======================================================================
```

zibous, 2023-09-15:
Error allready exits with ESPHome 2023.8.3... 

```
src/esphome/components/wifi_info/wifi_info_text_sensor.h:32:49: error: 'class esphome::wifi::WiFiComponent' has no member named 'get_dns_address'; did you mean 'get_ip_address'?
```

ssieb, 2023-09-16:
I can't reproduce this and the CI test cases pass as well.  Are you sure you don't have a `custom_components` directory or something?

zibous, 2023-09-16:
@ssieb 
Negativ, simple testcase:

```
# ---------------------------------------------------
# ESPHome by Nabu Casa | Fund development | 2023.8.3 
# ---------------------------------------------------
esphome:
  name: esp-testdevice
  friendly_name: esp-testdevice

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "QqRGDlbZxhvCgY1Ht9CQhy/T8xkv5hPg67N0Iq0qEiY="

ota:
  password: "31ea82aa3bf50a66dcffe0d1af5ff971"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp-Testdevice Fallback Hotspot"
    password: "LfHA8EzmvGkp"

captive_portal:

text_sensor:

  # connected to ssid

  # In file included from src/esphome/components/wifi_info/wifi_info_text_sensor.cpp:1:
  # src/esphome/components/wifi_info/wifi_info_text_sensor.h: In member function 'virtual void esphome::wifi_info::DNSAddressWifiInfo::update()': 
  # src/esphome/components/wifi_info/wifi_info_text_sensor.h:32:49: error: 'class esphome::wifi::WiFiComponent' has no member named 'get_dns_address'; did you mean 'get_ip_address'?
  #   auto dns_one = wifi::global_wifi_component->get_dns_address(0);
  #                                               ^~~~~~~~~~~~~~~
  #                                               get_ip_address
  # src/esphome/components/wifi_info/wifi_info_text_sensor.h:33:49: error: 'class esphome::wifi::WiFiComponent' has no member named 'get_dns_address'; did you mean 'get_ip_address'?
  #   auto dns_two = wifi::global_wifi_component->get_dns_address(1);
  #                                               ^~~~~~~~~~~~~~~
  #                                               get_ip_address
  # *** [.pioenvs/esp-testdevice/src/esphome/components/wifi_info/wifi_info_text_sensor.cpp.o] Error 1

  - platform: wifi_info
    ssid:
      name: WLAN SSID
      id: wlan_ssid
      entity_category: "diagnostic"
      disabled_by_default: true
      icon: mdi:wifi-settings

  # Device application version
  - platform: version
    name: Version
    hide_timestamp: true
    entity_category: "diagnostic"
    id: appver
    icon: mdi:application-cog-outline
    disabled_by_default: true    
```

Using Docker Version: esphome/esphome:latest

ssieb, 2023-09-16:
There must be something wrong with your system.  I copied that and compiled it with no errors.  And that method is definitely there.  What does "esphome version" show in the container?

zibous, 2023-09-16:
@ssieb 

Thank you for taking care of my problem.

> What does "esphome version" show in the container?

Version: 2023.8.3

Yes, there must be something wrong with the Docker installation, because when I try with the esphome cli version, it works. 
```
INFO ESPHome 2023.9.0b1
INFO Reading configuration esp-testdevice.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Please wait while upgrading PlatformIO...
PlatformIO has been successfully upgraded to 6.1.11!
```

Compile on docker console:
```
root@nuc2023-wlan:/config# esphome compile esp-testdevice.yaml
INFO ESPHome 2023.8.3
INFO Reading configuration esp-testdevice.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Deleting .esphome/build/esp-testdevice/.pioenvs
INFO Deleting .esphome/build/esp-testdevice/.piolibdeps
INFO Compiling app...
Processing esp-testdevice (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
---------------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing platformio/espressif32 @ 5.4.0
INFO Installing platformio/espressif32 @ 5.4.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif32@5.4.0 has been installed!
INFO espressif32@5.4.0 has been installed!
Tool Manager: Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
INFO Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-xtensa-esp32@8.4.0+2021r2-patch5 has been installed!
INFO toolchain-xtensa-esp32@8.4.0+2021r2-patch5 has been installed!
Tool Manager: Installing platformio/framework-arduinoespressif32 @ ~3.20005.0
INFO Installing platformio/framework-arduinoespressif32 @ ~3.20005.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: framework-arduinoespressif32@3.20005.220925 has been installed!
INFO framework-arduinoespressif32@3.20005.220925 has been installed!
Tool Manager: Installing platformio/tool-esptoolpy @ ~1.40400.0
INFO Installing platformio/tool-esptoolpy @ ~1.40400.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-esptoolpy@1.40400.0 has been installed!
INFO tool-esptoolpy@1.40400.0 has been installed!
Library Manager: Installing esphome/AsyncTCP-esphome @ 1.2.2
INFO Installing esphome/AsyncTCP-esphome @ 1.2.2
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@1.2.2 has been installed!
INFO AsyncTCP-esphome@1.2.2 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
[... 58 lines cut ...]
Compiling .pioenvs/esp-testdevice/src/esphome/components/web_server_base/web_server_base.cpp.o
Compiling .pioenvs/esp-testdevice/src/esphome/components/wifi/wifi_component.cpp.o
Compiling .pioenvs/esp-testdevice/src/esphome/components/wifi/wifi_component_esp32_arduino.cpp.o
Compiling .pioenvs/esp-testdevice/src/esphome/components/wifi/wifi_component_esp8266.cpp.o
Compiling .pioenvs/esp-testdevice/src/esphome/components/wifi/wifi_component_esp_idf.cpp.o
Compiling .pioenvs/esp-testdevice/src/esphome/components/wifi/wifi_component_pico_w.cpp.o
Compiling .pioenvs/esp-testdevice/src/esphome/components/wifi_info/wifi_info_text_sensor.cpp.o
Compiling .pioenvs/esp-testdevice/src/esphome/core/application.cpp.o
Compiling .pioenvs/esp-testdevice/src/esphome/core/color.cpp.o
Compiling .pioenvs/esp-testdevice/src/esphome/core/component.cpp.o
In file included from src/esphome/components/wifi_info/wifi_info_text_sensor.cpp:1:
src/esphome/components/wifi_info/wifi_info_text_sensor.h: In member function 'virtual void esphome::wifi_info::DNSAddressWifiInfo::update()':
src/esphome/components/wifi_info/wifi_info_text_sensor.h:32:49: error: 'class esphome::wifi::WiFiComponent' has no member named 'get_dns_address'; did you mean 'get_ip_address'?
     auto dns_one = wifi::global_wifi_component->get_dns_address(0);
                                                 ^~~~~~~~~~~~~~~
                                                 get_ip_address
src/esphome/components/wifi_info/wifi_info_text_sensor.h:33:49: error: 'class esphome::wifi::WiFiComponent' has no member named 'get_dns_address'; did you mean 'get_ip_address'?
     auto dns_two = wifi::global_wifi_component->get_dns_address(1);
                                                 ^~~~~~~~~~~~~~~
                                                 get_ip_address
*** [.pioenvs/esp-testdevice/src/esphome/components/wifi_info/wifi_info_text_sensor.cpp.o] Error 1
============================================================= [FAILED] Took 211.64 seconds =============================================================
```

But I can't figure out what it could be

zibous, 2023-09-16:
@ssieb 

> Are you sure you don't have a `custom_components` directory or something? 💯 

Thank you for this hint. The problem was that I had installed esphome-components in the zdzichu6969 component wifi. 
> see: https://github.com/zdzichu6969/esphome-components/tree/main/components/wifi

Didn't realize these were always loaded.
After I deleted it, everything worked again.

closed 2023-09-16 by zibous

ssieb, 2023-09-16:
Use `external_components:` instead to avoid this.