esphome/issues #4745: pn532_i2c excessive slowness

z3liff, 2023-07-25:
### The problem

pn532_i2c appears to be operating slowly.  Problem started after updating (went from ESPHome 2023.4.4 (which worked), to 2023.07.0 (which gives constant warnings).

Not sure what underlying change has caused the issue, but the device logs are constantly flooding with the warning.  More problematic is the fact the the .11s delay appears to be blocking while waiting to read from the device.  This seems to be causing the host device and peripherals to be running slowly.  For example, RTTTL playback is skewed and neopixel effects are jittery.

### Which version of ESPHome has the issue?

2023.07.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.07.3

### What platform are you using?

ESP32-IDF

### Board

ESP32-POE

### Component causing the issue

pn532_i2c

### Example YAML snippet

```yaml
pn532_i2c:
  update_interval: 1.0s
  id: "nfc_rfid_unit"
  on_tag:
    then:
      - text_sensor.template.publish:
          id: rfid_tag
          state: !lambda "return x;"
```

### Anything in the logs that might be useful for us?

```txt
[W][component:204]: Component pn532 took a long time for an operation (0.11 s).
```

z3liff, 2023-07-25:
My apologies. just saw [the notice](https://github.com/esphome/issues/issues/4717)

closed 2023-07-25 by z3liff

sweetpants, 2023-09-28:
I'm experiencing the same issue. Any news on it?