esphome/issues #4749: Speed Fan component - issue with speed_count

ESP4Ever, 2023-07-27:
### The problem

It looks like the speed_count setting is counted as a max_speed: when it is set to 10 the max speed is 10 at 100%

### Which version of ESPHome has the issue?

2023.6.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

latest

### What platform are you using?

ESP8266

### Component causing the issue

Speed Fan

### Example YAML snippet

```yaml
fan:
  - platform: speed
    id: fan1
    output: dimmer1
    name: "Ventilation Fan"
    speed_count: 10
    #restore_mode: ${fan_restore_mode}
    on_speed_set:
      lambda: |-
        if (id( fan1).state) {
          float temp = id( fan1).speed;
          ESP_LOGD( "TEST: temp", "speed: %f", temp);
        }
```

elik745i, 2023-07-27:
Try this temperature controlled fan speed:

`
output:
  - platform: ledc
    pin: GPIO33
    id: pwm_output
    frequency: 1000Hz

fan:
  - platform: speed
    output: pwm_output
    name: "Colling Fan"
    id: "the_fan"

# Sets the speed of the fan based on a linear calculation
# between the high and low temperature thresholds and
# the minimum specified fan speed
script:
  - id: set_fan_state
    then:
      - if:
          condition:
            lambda: |-
              return id(temperature_sensor).state < id(${temperature_threshold_low});
          then:
            - fan.turn_off: the_fan
          else:
            - fan.turn_on:
                id: the_fan
                speed: !lambda |-
                  if (id(temperature_sensor).state >= id(${temperature_threshold_high})) {
                    // Over upper threshold, fan speed at maximum
                    ESP_LOGD("Fan speed calc", "Temperature is above or equal to upper threshold so setting to max");
                    return 100;
                  }
                  else {
                    float calc_speed = ((100-id(${minimum_fan_speed})) / (id(${temperature_threshold_high})-id(${temperature_threshold_low})))*(id(temperature_sensor).state-id(${temperature_threshold_low}))+id(${minimum_fan_speed});
                    ESP_LOGD("Fan speed calc", "calculated speed = %f", calc_speed);
                    return calc_speed;
                  }
`

ssieb, 2023-07-28:
@ESP4Ever what are you expecting?

ESP4Ever, 2023-07-28:
> @ESP4Ever what are you expecting?

Hi! I expected that it works according to its description:

speed_count (Optional, int): Set the number of supported discrete speed levels. The value is used to calculate the percentages for each speed. E.g. 2 means that you have 50% and 100% while 100 will allow 1% increments in the output.

ssieb, 2023-07-28:
And that's how it works.  You need to explain more about what you think is going wrong.

ESP4Ever, 2023-07-28:
Sorry, I expected that this setting will give me a step granularity, f.e. if set to 10 the output will be 0, 10 , 20...100% (temp =0.00, 0.1, 0.2..1.0) but it always returns 10.0 instead and the MCU is loosing connection and reboots randomly

ssieb, 2023-07-28:
You're not reading the output, you're reading the speed that you set.  github issues aren't for support requests.  If you want help with whatever you're trying to do, come ask on the esphome discord server or one of the forums.

closed 2023-07-28 by ssieb

SenseiGinko, 2023-08-30:
@ESP4Ever email me at jparadnikas gmail com i have task for you . paid one