esphome/issues #4750: DSMR module (dev) not showing correct values for 13 month average peek

VerstrooidProfke, 2023-07-28:
### The problem

While merging the (dev) version of DSMR; I see a small bug in the calculation of "13_month_average_peak"
It does not show the 13_months_average peak, but te month peak of the month before...

I checked this via validation on "my fluvius",  in May, the number "13-month-average-peak" showed 

I'm not sure if the code is wrong, or the smartmeter delivers wrong numbers, but the number that is parsed, currently is "last monthts peak, and not "last years average peak".

Some screenshots to show

[image]

"13_month_average" shows 4,627 in june, while it should show 4,1314

[image]

"13_month_average" shows 4,043 in july, while it should show 4,11667

[image]

### Which version of ESPHome has the issue?

2023.7.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

All versions, i suppose. running 2023,7,2 right now

### What platform are you using?

ESP8266

### Example YAML snippet

```yaml
ESPHOME:

#https://www.esphome.io/components/external_components.html?highlight=external
external_components:
   - source:
      type: git
      url: https://github.com/bwynants/esphome
      ref: capacity_test
    components: [ dsmr ]

sensor:
  - platform: dsmr
    energy_delivered_tariff1:
      name: "Electriciteit DAG"
    energy_delivered_tariff2:
      name: "Electriciteit NACHT"
    #energy_returned_tariff1:
    #  name: "Teruggeleverde energie Tarief 1"
    #energy_returned_tariff2:
    #  name: "Teruggeleverde energie Tarief 2"
    power_delivered:
      name: "Verbruikt Vermogen"
    #power_returned:
    #  name: "Geleverd Vermogen"
    voltage_l1:
      name: "Spanning Fase 1"
    voltage_l2:
      name: "Spanning Fase 2"
    voltage_l3:
      name: "Spanning Fase 3"
    current_l1:
      name: "Stroom Fase 1"
    current_l2:
      name: "Stroom Fase 2"
    current_l3:
      name: "Stroom Fase 3"
    gas_delivered_be:
      name: "Verbruik Gas"
    active_energy_import_maximum_demand_running_month:
      name: "piekgebruik huidige maand"
    active_energy_import_current_average_demand:
      name: "Huidig kwartierverbruik"
    active_energy_import_maximum_demand_last_13_months:
      name: "13 maand piek kwartierverbruik"

HA dashboard config:
type: vertical-stack
cards:
  - show_name: true
    show_icon: false
    show_state: true
    type: glance
    entities:
      - entity: sensor.huidig_kwartierverbruik
        name: Huidig Kwartier
      - entity: sensor.piekgebruik_huidige_maand
        name: Maand
      - entity: sensor.13_maand_piek_kwartierverbruik
        name: Jaar
    title: Piekgebruik
  - type: history-graph
    hours_to_show: 12
    entities:
      - entity: sensor.huidig_kwartierverbruik
        name: Huidig Kwartier
      - entity: sensor.piekgebruik_huidige_maand
        name: Huidige maandpiek
      - entity: sensor.13_maand_piek_kwartierverbruik
        name: Huidige jaarpiek
    show_names: false
```

mhaket, 2023-11-15:
Example raw telegram from a BE meter:

```
/FLU5\253769484_A

0-0:96.1.4(50217)
0-0:96.1.1(3153414733313030303633383033)
0-0:1.0.0(231115094532W)
1-0:1.8.1(006780.240*kWh)
1-0:1.8.2(011645.231*kWh)
1-0:2.8.1(023545.888*kWh)
1-0:2.8.2(009317.351*kWh)
0-0:96.14.0(0001)
1-0:1.4.0(00.002*kW)
1-0:1.6.0(231105170000W)(02.391*kW)
0-0:98.1.0(11)(1-0:1.6.0)(1-0:1.6.0)(230101000000W)(221206183000W)(06.134*kW)(230201000000W)(230127174500W)(05.644*kW)(230301000000W)(230226063000W)(04.895*kW)(230401000000S)(230305181500W)(04.879*kW)(230501000000S)(230416094500S)(04.395*kW)(230601000000S)(230522084500S)(03.242*kW)(230701000000S)(230623053000S)(01.475*kW)(230801000000S)(230724060000S)(02.525*kW)(230901000000S)(230819174500S)(02.491*kW)(231001000000S)(230911063000S)(02.342*kW)(231101000000W)(231031234500W)(02.048*kW)
1-0:1.7.0(00.133*kW)
1-0:2.7.0(00.000*kW)
1-0:21.7.0(00.657*kW)
1-0:41.7.0(00.000*kW)
1-0:61.7.0(00.000*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.293*kW)
1-0:62.7.0(00.230*kW)
1-0:32.7.0(234.9*V)
1-0:52.7.0(239.2*V)
1-0:72.7.0(240.0*V)
1-0:31.7.0(003.37*A)
1-0:51.7.0(001.65*A)
1-0:71.7.0(001.26*A)
0-0:96.3.10(1)
0-0:17.0.0(999.9*kW)
1-0:31.4.0(999*A)
0-0:96.13.0()
0-1:24.1.0(003)
0-1:96.1.1(37464C4F32313139313233303733)
0-1:24.4.0(1)
0-1:24.2.3(231115094007W)(11722.671*m3)
!ab1b
```

This is the value for 'active_energy_import_maximum_demand_last_13_months':

` 0-0:98.1.0(11)(1-0:1.6.0)(1-0:1.6.0)(230101000000W)(221206183000W)(06.134*kW)(230201000000W)(230127174500W)(05.644*kW)(230301000000W)(230226063000W)(04.895*kW)(230401000000S)(230305181500W)(04.879*kW)(230501000000S)(230416094500S)(04.395*kW)(230601000000S)(230522084500S)(03.242*kW)(230701000000S)(230623053000S)(01.475*kW)(230801000000S)(230724060000S)(02.525*kW)(230901000000S)(230819174500S)(02.491*kW)(231001000000S)(230911063000S)(02.342*kW)(231101000000W)(231031234500W)(02.048*kW)`

With the current implementation of DSRM the value 2.048 kW is returned in stead of the complete array of values.

bwynants, 2024-01-06:
https://github.com/glmnet/arduino-dsmr/pull/18

bwynants, 2024-01-23:
add 
```
external_components:
  - source:
      type: git
      url: https://github.com/bwynants/esphome
      ref: dsmr
    components: [dsmr]
    refresh: 0s

```
to your yaml  for this fix until esphome is updated

mhaket, 2024-01-23:
While compiling it for a  esp32-c3-devkitm-1 board, I get the following error:

```
INFO ESPHome 2023.12.9
INFO Reading configuration /config/esphome/device.yaml...
INFO Updating https://github.com/bwynants/esphome@dsmr
INFO Generating C++ source...
INFO Compiling app...
Processing device (board: esp32-c3-devkitm-1; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.0.1
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.3
|-- Dsmr @ 0.9.0+sha.48b1dad
|-- Crypto @ 0.4.0
|-- noise-c @ 0.1.4
Compiling .pioenvs/device/src/esphome/components/dsmr/dsmr.cpp.o
Compiling .pioenvs/device/src/esphome/components/esp32/gpio.cpp.o
Compiling .pioenvsdevice/src/esphome/components/esp32/preferences.cpp.o
In file included from src/esphome/components/dsmr/dsmr.h:14,
                 from src/esphome/components/dsmr/dsmr.cpp:3:
.piolibdeps/device/arduino-dsmr/src/dsmr/fields.h: In member function 'dsmr::ParseResult<void> dsmr::AveragedFixedField<T, _unit, _int_unit>::parse(const char*, const char*)':
.piolibdeps/device/arduino-dsmr/src/dsmr/fields.h:188:16: error: 'uint32' was not declared in this scope
           for (uint32 i = 0; i < numberOfValues.result; i++)
                ^~~~~~
.piolibdeps/device/arduino-dsmr/src/dsmr/fields.h:188:16: note: suggested alternative: 'uint'
           for (uint32 i = 0; i < numberOfValues.result; i++)
                ^~~~~~
                uint
.piolibdeps/device/arduino-dsmr/src/dsmr/fields.h:188:30: error: 'i' was not declared in this scope
           for (uint32 i = 0; i < numberOfValues.result; i++)
                              ^
*** [.pioenvs/device/src/esphome/components/dsmr/dsmr.cpp.o] Error 1
========================== [FAILED] Took 2.78 seconds ==========================
```

Full YAML file:

```
substitutions:
  device_name: <hidden>

esphome:
  name: ${device_name}
  name_add_mac_suffix: false
  platformio_options:
    upload_speed: 1500000

esp32:
  board: esp32-c3-devkitm-1

improv_serial:
  next_url: http://{{ip_address}}

status_led:
  pin:
    number: 7
    inverted: true
    id: led

# temporary fix https://github.com/esphome/issues/issues/4750#issuecomment-1906850672
[... 185 lines cut ...]
  - platform: wifi_signal
    name: "Wi-Fi Signal"
    update_interval: 60s
      
text_sensor:
  - platform: dsmr
    identification:
      name: "DSMR Identification"
    p1_version:
      name: "DSMR Version"
    p1_version_be:
      name: "DSMR Version Belgium"
  - platform: wifi_info
    ip_address:
      name: "IP Address"
    ssid:
      name: "Wi-Fi SSID"
    bssid:
      name: "Wi-Fi BSSID"
  - platform: version
    name: "ESPHome Version"
    hide_timestamp: true
```

I cleared the cache and retried but same result.

bwynants, 2024-01-24:
> While compiling it for a esp32-c3-devkitm-1 board, I get the following error:
>
> ```
> INFO ESPHome 2023.12.9
> INFO Reading configuration /config/esphome/device.yaml...
> INFO Updating [...]

> In file included from src/esphome/components/dsmr/dsmr.h:14,
>                  from src/esphome/components/dsmr/dsmr.cpp:3:
> .piolibdeps/device/arduino-dsmr/src/dsmr/fields.h: In member function [...]

fixed, try again

mhaket, 2024-01-24:
Compiled and works.

Question, what value is now returned? I expected an array of 13 values but I receive only one.

bwynants, 2024-01-24:
> Compiled and works.
> 
> Question, what value is now returned? I expected an array of 13 values but I receive only one.

the single value as reported under https://mijn.fluvius.be/netkosten/ 
this is the average value (gemiddelde) of those 13  (or possibly less) values

mhaket, 2024-01-24:
Clear. Thanks for the library update.

closed 2026-01-06 by bdraco