esphome/issues #4751: 'esp_log_printf_' was not declared in this scope

pjanning, 2023-07-28:
### The problem

While trying to build with an external library i always get the error, that a logging function is not declared in scope.

### Which version of ESPHome has the issue?

 2023.7.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.7

### What platform are you using?

ESP32

### Board

wemos_d1_mini32

### Example YAML snippet

```yaml
esphome:
  name: display-esp32
  libraries:
    - arduino-libraries/Wire
    - SPI
    - Adafruit BusIO
    #- ESP-IDF Logging Library
  platformio_options:
    lib_ldf_mode: "chain+"
    build_type: debug

esp32:
  board: wemos_d1_mini32
  framework:
    type: arduino

# Enable Logger
logger:
  level: INFO

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

#captive_portal:

font:
  - id: digits
    # 3x5 pixel font: https://www.dafont.com/de/pixelzim3x5.font
    file: "fonts/pzim3x5.ttf"
    size: 32

# Enable Home Assistant API
api:
  services:
  #- service: lukas
  #  then:
  #  - display.page.show: page1
  #- service: show_rakete
  #  then:
  #  - display.page.show: page_rakete
  - service: show_time
    then:
    - display.page.show: page_time
  - service: show_black
    then:
    - display.page.show: page_black
  - service: show_pvgraph
    then:
    - display.page.show: page_pvgraph

external_components:
  - source:
      url: https://github.com/pjanning/esphome-pxmatrix-component.git
      type: git
    components: [ dmamatrix ]
#  - source: 'github://pr#3255'
#    components:
#      - display
#      - online_image
time:
 - platform: homeassistant
   id: ha_time

sensor:
  - platform: homeassistant
    id: solarpower
    entity_id: sensor.total_dc_power
  - platform: homeassistant
    id: load
    entity_id: sensor.load_power
  - platform: homeassistant
    id: openelements
    entity_id: sensor.offene_elemente

graph:
  # Show multi-trace graph
  - id: powergraph
    duration: 3h
    width: 128
    height: 64
    traces:
      - sensor: load
        line_type: SOLID
        line_thickness: 1
        color: my_red
      - sensor: solarpower
        line_type: DASHED
        line_thickness: 1
        color: my_green

color:
  - id: my_red
    red: 0%
    green: 0%
    blue: 100%
  - id: my_green
    red: 0%
    green: 100%
    blue: 0%

text_sensor:
  - platform: homeassistant
    id: weather
    name: "Weather Forecast From Home Assistant"
    entity_id: weather.pj_home

display:
  platform: dmamatrix
  id: my_display
  brightness: 100
  pages:
  - id: page_time
    lambda: |-
       it.strftime(10, 10, id(digits), "%H:%M", id(ha_time).now());
       it.strftime(10, 40, id(digits), "%H:%M", id(ha_time).now());
       ESP_LOGI("custom", "%H:%M", id(ha_time).now());
  - id: page_black
    lambda: |-
       it.fill(COLOR_OFF);
  - id: page_pvgraph
    lambda: |-
      it.graph(0, 0, id(powergraph));
```

### Anything in the logs that might be useful for us?

```txt
/data/display-esp32/.piolibdeps/display-esp32/ESP32 HUB75 LED MATRIX PANEL DMA Display/src/ESP32-HUB75-MatrixPanel-I2S-DMA.h: In member function 'void MatrixPanel_I2S_DMA::setBrightness(uint8_t)':
src/esphome/core/log.h:105:3: error: 'esp_log_printf_' was not declared in this scope
   esp_log_printf_(ESPHOME_LOG_LEVEL_INFO, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
```

ssieb, 2023-07-28:
github issues are not for support for external components unless you can show that it's a bug in esphome.  However, logging is working for everyone else's components.  If you need help, then come ask on the esphome discord server or one of the forums.

closed 2023-07-28 by ssieb

pjanning, 2023-07-28:
The component is using a standard logging function... The error is in ESPHome and not in someone elses lib...

`src/esphome/core/log.h:105:3: error: 'esp_log_printf_' was not declared in this scope
   esp_log_printf_(ESPHOME_LOG_LEVEL_INFO, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)`

amcfague, 2023-10-25:
This issue is the same as #3196; when the `ESP_LOGD` macro is expanded by the compiler, it looks for `esp_log_printf_` which is _not_ in the global namespace--it's in `esphome::esp_log_printf_`. The real "fix" would be to move it into the global namespace, assuming there are no conflicts.

For the short term, you can do something like the following:

* `using namespace esphome;`
* Wrap your code with `namespace esphome { ...  }`

If you only have a header and no implementation file, keep in mind this WILL pollute the namespace--although it's not likely a huge issue.