esphome/issues #4752: BME680 with BSEC: Unexpected humidity data with temperature offset

xaviergr, 2023-07-29:
### The problem

I have a suspicion that my humidity readings with the bme680_bsec component are wrong so I did the following test:
Connect two bme680 sensors (side by side, and away from the board) on an esp32 board and use the bm680_bsec component to get temperature and humidity data in lp mode (sample every 3 seconds).

As expected the two sensors report slightly higher temperature than the actual temperature and both require different offsets to report the same temperature (-0.5 and -0.3). After adjusting the offset both sensors show the exact same temperature but the humidity between them is off by 2 points (68% vs 70% at 25C). This humidity mismatch will become even bigger if the temperature offset has a bigger difference between the two (e.g. if one of the sensors is inside an enclosure).

Is this unexpected behavior or should I assume that also the humidity sensor needs to be calibrated with a point of reference after the temperature compensation?

Another thing that bugs me is that if I don't use the bsec library (just the normal bme680 component) the humidity readings are completely different, even when trying to apply temperature compensation. Without bsec the only option is to use the generic sensor filter to apply a temperature offset, but this offset is only applied to temperature and humidity is not affected so the readings are way off.

Going further and to try to measure humidity adjusted by the temperature offset I found the following formula from Texas Instrument. But even with this, humidity readings can vary by up to 10% compared to the bsec library (given the same temperature offsets).
[image]

My next step is to use bme680_bsec, bme680 and bme680+temperature compensation with a salt chamber to see who is closer to the expected 75% humidity. Until then if someone can give any additional hints it would be very helpful.

### Which version of ESPHome has the issue?

2023.7.0-dev

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

bme680

### Example YAML snippet

```yaml
bme680_bsec:
  - id: bsec0
    address: 0x77
    sample_rate: lp
    temperature_offset: 0.3
  - id: bsec1
    address: 0x76
    sample_rate: lp
    temperature_offset: 0.5

sensor:
  - platform: bme680_bsec
    bme680_bsec_id: bsec0
    temperature:
      name: "BSEC0 Temperature"
    humidity:
      name: "BSEC0 Humidity"

  - platform: bme680_bsec
    bme680_bsec_id: bsec1
    temperature:
      name: "BSEC1 Temperature"
    humidity:
      name: "BSEC1 Humidity"
```

DAVe3283, 2023-07-30:
Assuming your sensors are genuine Bosch, the relative humidity (RH) accuracy is specified as [±3%](https://www.bosch-sensortec.com/products/environmental-sensors/gas-sensors/bme680/#technical). So you could have 2 samples that are as far as 6% apart while both being fully in spec.

The reason you see the RH value change as you adjust the offset is because the bsec library is attempting to calculate what the relative humidity should be at the corrected temperature, so any sensor error will manifest as non-linearity in the "corrected" RH.

The [Absolute Humidity component docs](https://esphome.io/components/sensor/absolute_humidity.html) have more details on absolute humidity and how temperature affects relative humidity, with links to good technical information.

I don't think you are seeing a bug in the ESPHome code or even in the bsec library (which is proprietary from Bosch). The BME680 is a good sensor, but IMO you will never get lab-grade results from it, regardless of individual calibration. And a single-point calibration like the bsec library has will only really get you close at that one point.

xaviergr, 2023-07-31:
Hello Dave, thanks for you answer. I think this whole calibration topic and normal vs bsec library differences might be more than I can chew on but still worth exploring.

> Assuming your sensors are genuine Bosch, the relative humidity (RH) accuracy is specified as [±3%](https://www.bosch-sensortec.com/products/environmental-sensors/gas-sensors/bme680/#technical). So you could have 2 samples that are as far as 6% apart while both being fully in spec.

The sensors are from AliExpress so there is a chance that they are not genuine. Having said that I had some rough temperature calibration attempts, which left me quite impressed with these sensors, genuine or not. Basically the sensors are 0.2/0.3 degrees away from 0°C when placed in an ice/water bath and the same treatment with boiling water resulted in a temperature of 99.6°C (the other sensor shorted in the process but is still operational after drying).

Humidity wise I am still running the salt chamber calibration process, which so far, gives quite positive results too, showing that both sensors are within spec (although I am still monitoring to see if the humidity changes further). After more than 24 hours (at 25°C) one sensor reads 73.8% while the other is at 75.3%, which is amazing considering the numbers given from [this source](https://www.engineeringtoolbox.com/salt-humidity-d_1887.html). 

> The reason you see the RH value change as you adjust the offset is because the bsec library is attempting to calculate what the relative humidity should be at the corrected temperature, so any sensor error will manifest as non-linearity in the "corrected" RH.

My problem is that even the slightest temperature offset with bsec overshoots the expected results. For example the 2nd sensor that is bang on the expected 75.3% humidity is overshooting the temperature by 0.2°C, applying this correction with bsec gives a reading of 76.2% which seems wrong. On the other hand manually compensating for this 0.2 temperature difference also results in an erroneous 75.6% but closer to a real value than with bsec. In a real scenario the temperature difference can be at least 2-3°C and in that case the bsec readings seem completely useless.
 
> The [Absolute Humidity component docs](https://esphome.io/components/sensor/absolute_humidity.html) have more details on absolute humidity and how temperature affects relative humidity, with links to good technical information.

Thanks, I wasn't aware of this, it will help cross checking calculations and get more points of reference.

> I don't think you are seeing a bug in the ESPHome code or even in the bsec library (which is proprietary from Bosch). The BME680 is a good sensor, but IMO you will never get lab-grade results from [...]

I guess that's the key point  here, even if somehow the bsec temperature offset is erroneous this has nothing to do with esphome. The other good thing that I discovered by accident is that a user can use one sensor both with or without the bsec library simultaneously and get samples at a greater granularity than the lp/ulp modes of bsec provides.

Moving on, I think I will use these 2 sensors as a reference and calibrate manually all the others that are inside enclosures and susceptible to a noticeable temperature offset. But so far it looks to me that bsec is over-compensating way too much to be reliable for this scenario.

xaviergr, 2023-08-07:
After a lot of experimenting the best result for me was the following:
 - First calibrate a sensor with the salt chamber method and use it as a reference for humidity.
 - For temperature you can carefully use an ice/boiling water bath and/or another reference (a laser thermometer was close enough for me).
 - Place the calibrated sensor away from any heat source but near the desired measuring spot to compare with another sensor that is in an enclosure and subject to a temperature humidity offset.
 - Adjust the BSEC library temperature offset of the enclosed sensor to match the temperature readings of the reference sensor.
 - Hopefully the BSEC humidity readings should be close to the reference, if not you can adjust with a fixed sensor filter offset.
 - Alternatively, manually apply the August-Roche-Magnus formula as a lambda filter, but in my case the BSEC compensation was actually closer to the calibrated sensor.

Ideally, you would do all of the above at different humidity/temperature levels so that you can get a linear/polynomial relationship that can be applied as an additional filter, but that is too much for me.

In the end there is no bug or unexpected behavior besides my sensors not being calibrated, so I will close this issue.

closed 2023-08-07 by xaviergr