esphome/issues #4753: INA226 not reporting "power" and "bus voltage"

hirsts, 2023-07-30:
### The problem

I have connected and INA226 to an ESP32 and have missing values and incorrect values. I am using a default INA226 with an R100 resistor. 

### Which version of ESPHome has the issue?

2023.7.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP32

### Board

Quinled-eps32

### Component causing the issue

ina226

### Example YAML snippet

```yaml
i2c:
  - id: bus_a
    frequency: 400kHz
    sda: 17
    scl: 22
    scan: true

sensor:        
  - platform: total_daily_energy
    name: "PebblePower1 CH1  Daily Energy"
    power_id: power
    id: my_power

  - platform: ina226
    i2c_id: bus_a
    address: 0x40
    shunt_resistance: 0.1 ohm
    current:
      name: "PebblePower1 CH1 Current"
      id: current
    power:
      name: "PebblePower1 CH1 Power"
      id: power
    bus_voltage:
      name: "PebblePower1 CH1 Bus Voltage"
      id: voltage
    shunt_voltage:
      name: "PebblePebble1 CH1 Shunt Voltage"
      id: shunt_voltage
      unit_of_measurement: "mV"
    max_current: 3.2A
    update_interval: 1s
```

### Anything in the logs that might be useful for us?

```txt
[21:58:54][C][i2c.arduino:053]: I2C Bus:
[21:58:54][C][i2c.arduino:054]:   SDA Pin: GPIO17
[21:58:54][C][i2c.arduino:055]:   SCL Pin: GPIO22
[21:58:54][C][i2c.arduino:056]:   Frequency: 400000 Hz
[21:58:54][C][i2c.arduino:059]:   Recovery: bus successfully recovered
[21:58:54][I][i2c.arduino:069]: Results from i2c bus scan:
[21:58:54][I][i2c.arduino:075]: Found i2c device at address 0x40
[21:58:54][C][switch.gpio:068]: GPIO Switch 'Relay'
[21:58:54][C][switch.gpio:091]:   Restore Mode: always OFF
[21:58:54][C][switch.gpio:031]:   Pin: GPIO23
WARNING pebble1-quinled-esp32.local: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for pebble1-quinled-esp32.local
WARNING Disconnected from API
INFO Successfully connected to pebble1-quinled-esp32.local
[21:59:14][D][sensor:094]: 'PebblePower1 CH1 Bus Voltage': Sending state 0.00000 V with 2 decimals of accuracy
[21:59:14][D][sensor:094]: 'PebblePebble1 CH1 Shunt Voltage': Sending state -0.07837 mV with 2 decimals of accuracy
[21:59:14][D][sensor:094]: 'PebblePower1 CH1 Current': Sending state -0.77998 A with 3 decimals of accuracy
[21:59:14][D][sensor:094]: 'PebblePower1 CH1 Power': Sending state 0.00000 W with 2 decimals of accuracy
[21:59:14][D][sensor:094]: 'PebblePower1 CH1  Daily Energy': Sending state 0.00002 Wh with 4 decimals of accuracy
[21:59:14][W][component:204]: Component ina226.sensor took a long time for an operation (0.05 s).
[21:59:14][W][component:205]: Components should block for at most 20-30ms.
[21:59:15][D][sensor:094]: 'PebblePower1 CH1 Bus Voltage': Sending state 0.00000 V with 2 decimals of accuracy
[21:59:15][D][sensor:094]: 'PebblePebble1 CH1 Shunt Voltage': Sending state -0.07824 mV with 2 decimals of accuracy
[21:59:15][D][sensor:094]: 'PebblePower1 CH1 Current': Sending state -0.78576 A with 3 decimals of accuracy
[21:59:15][D][sensor:094]: 'PebblePower1 CH1 Power': Sending state 0.00000 W with 2 decimals of accuracy
[21:59:15][D][sensor:094]: 'PebblePower1 CH1  Daily Energy': Sending state 0.00002 Wh with 4 decimals of accuracy
[21:59:15][W][component:204]: Component ina226.sensor took a long time for an operation (0.05 s).
[21:59:15][W][component:205]: Components should block for at most 20-30ms.
[21:59:16][D][sensor:094]: 'PebblePower1 CH1 Bus Voltage': Sending state 0.00000 V with 2 decimals of accuracy
[21:59:16][D][sensor:094]: 'PebblePebble1 CH1 Shunt Voltage': Sending state -0.07871 mV with 2 decimals of accuracy
[21:59:16][D][sensor:094]: 'PebblePower1 CH1 Current': Sending state -0.78537 A with 3 decimals of accuracy
[21:59:16][D][sensor:094]: 'PebblePower1 CH1 Power': Sending state 0.00000 W with 2 decimals of accuracy
[21:59:16][D][sensor:094]: 'PebblePower1 CH1  Daily Energy': Sending state 0.00002 Wh with 4 decimals of accuracy
[21:59:16][W][component:204]: Component ina226.sensor took a long time for an operation (0.05 s).
[21:59:16][W][component:205]: Components should block for at most 20-30ms.
[21:59:16][D][switch:016]: 'Relay' Turning OFF.
[21:59:17][D][sensor:094]: 'PebblePower1 CH1 Bus Voltage': Sending state 0.00000 V with 2 decimals of accuracy
[21:59:17][D][sensor:094]: 'PebblePebble1 CH1 Shunt Voltage': Sending state -0.07823 mV with 2 decimals of accuracy
[21:59:17][D][sensor:094]: 'PebblePower1 CH1 Current': Sending state -0.78077 A with 3 decimals of accuracy
[21:59:17][D][sensor:094]: 'PebblePower1 CH1 Power': Sending state 0.00000 W with 2 decimals of accuracy
[21:59:17][D][sensor:094]: 'PebblePower1 CH1  Daily Energy': Sending state 0.00002 Wh with 4 decimals of accuracy
[21:59:17][W][component:204]: Component ina226.sensor took a long time for an operation (0.05 s).
[21:59:17][W][component:205]: Components should block for at most 20-30ms.
```

### Additional information

Values for "current" seem correct, all be it provided as a negative number.

"power" and "bus voltage" are always zero.

[image]

DAVe3283, 2023-07-30:
From your diagram, I believe you have the IN- and IN+ wiring backwards. That would explain the negative current value and the 0 voltage reading.

hirsts, 2023-07-30:
When I swap IN- and IN+ I get Amps as a positive current value but still zero bus voltage and zero power. Basically the same issues but with a positive reading for current.

Thanks for the feedback.

DAVe3283, 2023-07-30:
You also need to have the +12v rail hooked to the INA226 `VBUS` input. You don't show that on your schematic. I don't have an INA226 to help further if it doesn't show voltage after hooking up that wire.

hirsts, 2023-07-31:
Hi @DAVe3283, thanks for your help. I have now connected the INA226 VBUS input to IN+ and its now working correctly. I assumed that this was connected via the INA226 module. Thanks again.

hirsts, 2023-07-31:
Resolved with help from @DAVe3283.

Additional info that might help others: 
[https://how2electronics.com/how-to-use-ina226-dc-current-sensor-with-arduino/ ](https://how2electronics.com/how-to-use-ina226-dc-current-sensor-with-arduino/ )

closed 2023-07-31 by hirsts