esphome/issues #4762: Compilation fails with error "no matching function for call" - Alarm Control Panel

blackscreener, 2023-08-03:
### The problem

Can't compile arm and disarm with function of Template Alarm Control Panel within lambdas with pin code. Without code it works.

id(acp1).arm_away(); - works
id(acp1).arm_home(); - works
id(acp1).disarm("1234"); - doesn't work

### Which version of ESPHome has the issue?

2023.7.0, 2023.7.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.7.4, 2023.8.0

### What platform are you using?

ESP32

### Component causing the issue

alarm_control_panel alarm_control_panel/template

### Example YAML snippet

```yaml
sim800l:
  on_sms_received:
    - lambda: |-
        id(sms_sender).publish_state(sender);
        id(sms_message).publish_state(message);
    - lambda: |-
        if  (id(sms_sender).state == "xxx" and  id(sms_message).state == "disarm" or id(sms_message).state == "Disarm" ) {
          int code = 1234;
          id(acp1).disarm("1234");
        }
    - lambda: |-
        if ( id(sms_sender).state == "xxx" and  id(sms_message).state == "arm_away" or id(sms_message).state == "Arm_away") {
          id(acp1).arm_away();
        }
    - lambda: |-
        if ( id(sms_sender).state == "xxx" and id(sms_message).state == "arm_home" or id(sms_message).state == "Arm_home") {
          id(acp1).arm_home();
        }
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.7.1
INFO Reading configuration /config/esphome/alarm.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing alarm (board: nodemcu-32s; framework: arduino; platform: platformio/espressif32@5.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Wire @ 2.0.0
|-- noise-c @ 0.1.4
Compiling /data/alarm/.pioenvs/alarm/src/main.cpp.o
/config/esphome/alarm.yaml: In lambda function:
/config/esphome/alarm.yaml:219:28: error: no matching function for call to 'esphome::template_::TemplateAlarmControlPanel::disarm(const char [5])'
           id(acp1).disarm("1234");
                            ^
In file included from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/components/alarm_control_panel/alarm_control_panel.h:105:8: note: candidate: 'void esphome::alarm_control_panel::AlarmControlPanel::disarm(esphome::optional<std::__cxx11::basic_string<char> >)'
   void disarm(optional<std::string> code = nullopt);
        ^~~~~~
src/esphome/components/alarm_control_panel/alarm_control_panel.h:105:8: note:   no known conversion for argument 1 from 'const char [5]' to 'esphome::optional<std::__cxx11::basic_string<char> >'
*** [/data/alarm/.pioenvs/alarm/src/main.cpp.o] Error 1
========================= [FAILED] Took 196.43 seconds =========================
```

ssieb, 2023-08-03:
The parameter is an `optional`.  You could try using `std::string("1234")`, but you might need to do `optional(std::string("1234"))`.

blackscreener, 2023-08-03:
Thanks. The first method works.

closed 2023-08-04 by blackscreener