esphome/issues #4763: WiFi Signal Sensor - unit_of_measurement: "Signal %"

sblogoshs, 2023-08-03:
### The problem

**The original ESP-Home "WiFi Signal Sensor" configuration causes this error message.**

The previous versions also had this problem.

**_unit_of_measurement: "Signal %"_**

### Which version of ESPHome has the issue?

2023.7.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.0

### What platform are you using?

ESP8266

### Board

Mini D1

### Component causing the issue

WiFi Signal Sensor

### Example YAML snippet

```yaml
# Example configuration entry
sensor:
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 60s

# Example configuration entry with 2 sensors and filter
sensor:
  - platform: wifi_signal # Reports the WiFi signal strength/RSSI in dB
    name: "WiFi Signal dB"
    id: wifi_signal_db
    update_interval: 60s
    entity_category: "diagnostic"

  - platform: copy # Reports the WiFi signal strength in %
    source_id: wifi_signal_db
    name: "WiFi Signal Percent"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "Signal %"
    entity_category: "diagnostic"
```

### Anything in the logs that might be useful for us?

```txt
Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:659
Integration: Sensor (documentation, issues)
First occurred: 10:33:55 (2 occurrences)
Last logged: 11:36:34

Entity sensor.wifi_signal_percent_17 (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement 'Signal %' which is not a valid unit for the device class ('signal_strength') it is using; expected one of ['dB', 'dBm']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22
```

### Additional information

_All ESP's with WiFi signal sensor are affected._

DAVe3283, 2023-08-05:
You need to remove the default `device_class` (which is `signal_strength`) from the 2nd entry. Usually that can be done by setting it to an empty string, e.g.

```yaml
sensor:
  - platform: copy # Reports the WiFi signal strength in %
    source_id: wifi_signal_db
    name: "WiFi Signal Percent"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "Signal %"
    entity_category: "diagnostic"
    # This should fix the units error:
    device_class: ""
```

sblogoshs, 2023-08-05:
Thank you for this information but then the example should be on 
https://esphome.io/components/sensor/wifi_signal
also be adjusted, I did not know that I can use device_class: also empty.

bigbangus, 2023-11-11:
>     device_class: ""

This worked to suppress my home assistance core log warnings. thank you. what is the reporting mechanism to change the example in the ESPHome documentation?

ssieb, 2023-11-11:
Does changing it to `unit_of_measurement: "%"` work without changing the device class?
You can make a PR for the docs.

bigbangus, 2023-11-11:
> Does changing it to `unit_of_measurement: "%"` work without changing the device class? You can make a PR for the docs.

I don't think so, I tried that and it produces more errors about % not being the expected unit db, dbm for the class.

Ok I'll try to create a PR. Thank you.

bigbangus, 2023-11-15:
PR created: https://github.com/esphome/esphome-docs/pull/3379

closed 2023-11-15 by nagyrobi