esphome/issues #4767: esp32_rmt_led_strip breaks remote_transmitter ?

jason-curtis, 2023-08-04:
### The problem

I'm trying to use an M5stack Atom Echo to turn my projector on and off, and also use the status LED on the Echo. When I configure the `remote_transmitter` without the LED, it works fine. But when I add the `esp32_rmt_led_strip`, I get an error in the logs:

```
[E][remote_transmitter:026]: Configuring RMT driver failed: ESP_ERR_INVALID_STATE
```
When this happens my IR commands don't work. The LED is also "stuck" on. When I control the LED sometimes something subtle happens with the LED but it's hard to tell because it seems to be always on somewhere near 100%.

Speculation: is there some kind of conflict between these two drivers? It appears that the IR LED and the `SK6812` are on separate pins, but maybe they share an internal clock or something?

Any help is much appreciated!

### Which version of ESPHome has the issue?

2023.7.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.7.3

### What platform are you using?

ESP32-IDF

### Board

M5stack Atom Echo

### Component causing the issue

esp32_rmt_led_strip

### Example YAML snippet

```yaml
remote_transmitter:
  pin: GPIO12 # built-in
  # pin: GPIO26 # attached, also breaks
  carrier_duty_percent: 50%

# on/off control codes for my Optoma HD141X projector - these work as long as the `esp32_rmt_led_strip` is not configured.
switch:
  - platform: template
    name: Projector Power
    icon: mdi:projector
    assumed_state: true
    optimistic: true
    turn_on_action:
      - logger.log: "Transmitting power on signal"
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38kHz
          code: [8960, -4459, 566, -547, 565, -1658, 566, -547, 565, -548, 564, -1658, 566, -1658, 565, -548, 565, -570, 542, -1658, 565, -571, 542, -1658, 565, -1659, 565, -571, 541, -571, 541, -1659, 565, -1659, 565, -571, 541, -1659, 565, -571, 541, -571, 541, -571, 541, -571, 541, -571, 541, -571, 541, -1683, 541, -571, 541, -1683, 541, -1682, 542, -1682, 541, -1683, 541, -1683, 541, -1683, 541]
      - logger.log: "Transmitted power on signal"
    turn_off_action:
      - repeat:
          count: 2
          then:
          - logger.log: "Transmitting power off signal"
          - remote_transmitter.transmit_raw:
              carrier_frequency: 38kHz
              code: [8937, -4485, 564, -549, 564, -1660, 562, -550, 563, -550, 562, -1661, 538, -1686, 562, -572, 540, -572, 541, -1683, 540, -572, 540, -1684, 541, -1683, 539, -573, 516, -596, 516, -1708, 540, -1684, 540, -572, 540, -1683, 540, -1684, 540, -1685, 539, -573, 539, -1685, 539, -572, 516, -596, 540, -1684, 540, -572, 540, -572, 540, -573, 539, -1685, 539, -573, 539, -1685, 540, -1683, 540]
          - logger.log: "Transmitting power off signal part 2"
          - remote_transmitter.transmit_raw:
              carrier_frequency: 38kHz
              code: [8956, -2250, 540]
          - logger.log: "Transmitted power off signal"
          - delay: 1s

light:
  - name: Status Light
    platform: esp32_rmt_led_strip
    id: led
    restore_mode: ALWAYS_OFF
    pin: GPIO27
    default_transition_length: 0s
    chipset: SK6812
    num_leds: 1
    rgb_order: grb
    rmt_channel: 0
    effects:
      - pulse:
          transition_length: 250ms
          update_interval: 250ms
```

### Anything in the logs that might be useful for us?

```txt
[21:35:24][C][esp32_rmt_led_strip:171]: ESP32 RMT LED Strip:
[21:35:24][C][esp32_rmt_led_strip:172]:   Pin: 27
[21:35:24][C][esp32_rmt_led_strip:173]:   Channel: 0
[21:35:24][C][esp32_rmt_led_strip:198]:   RGB Order: GRB
[21:35:24][C][esp32_rmt_led_strip:199]:   Max refresh rate: 0
[21:35:24][C][esp32_rmt_led_strip:200]:   Number of LEDs: 1
[21:35:24][C][light:103]: Light 'Status Light'
[21:35:24][C][light:105]:   Default Transition Length: 0.0s
[21:35:24][C][light:106]:   Gamma Correct: 2.80
[21:35:25][C][template.switch:068]: Template Switch 'Projector Power'
[21:35:25][C][template.switch:070]:   Icon: 'mdi:projector'
[21:35:25][C][template.switch:073]:   Assumed State: YES
[21:35:25][C][template.switch:091]:   Restore Mode: always OFF
[21:35:25][C][template.switch:057]:   Optimistic: YES
[21:35:25][C][remote_transmitter:015]: Remote Transmitter...
[21:35:25][C][remote_transmitter:016]:   Channel: 0
[21:35:25][C][remote_transmitter:017]:   RMT memory blocks: 1
[21:35:25][C][remote_transmitter:018]:   Clock divider: 80
[21:35:25][C][remote_transmitter:019]:   Pin: GPIO12
[21:35:25][C][remote_transmitter:022]:     Carrier Duty: 50%
[21:35:25][E][remote_transmitter:026]: Configuring RMT driver failed: ESP_ERR_INVALID_STATE
```

### Additional information

My `esp32_rmt_led_strip` configuration is based on the [voice-assistant demo](https://github.com/esphome/firmware/blob/main/voice-assistant/m5stack-atom-echo.yaml#L127).

ssieb, 2023-08-04:
The logs show you what the problem is:
`[21:35:24][C][esp32_rmt_led_strip:173]:   Channel: 0`
`[21:35:25][C][remote_transmitter:016]:   Channel: 0`

Change the rmt channel for the led strip.

closed 2023-08-04 by ssieb

jason-curtis, 2023-08-04:
Thank you! I proposed a documentation update.